MCUdude/MicroCore #1: Add boards manager support

MCUdude, 2016-07-23:
The boards manager URL makes it easier to install the core in Arduino IDE and  in the Arduino Eclipse plugin. The core isn't ready for a v1.0.0 release yet, but we're getting closer. I know you're a busy man @per1234, but would you like to help me out with the boards manager URL on this core too? I promise this will be the last one 😉

per1234, 2016-07-23:
Sure, I'm happy to help with this and any other projects. I'm watching the repository so I'll get an email when you do the release.

MCUdude, 2016-07-25:
I can't find the PM functionality here on Github, so I guess I'll post it here instead:
According to the Github traffic page most visitors discover my core from the "Arduino on other Atmel chip" page. There have been added more and more links, and some are really obsolete (2011 and earlier). Who has the right to decide what to delete and what to keep on that site? Why list [this page](http://toasterbotics.blogspot.no/2011/08/attiny2313-with-arduino.html) when Dr Azzy have created an awesome core for it instead?

per1234, 2016-07-25:
>  most visitors discover my core from the "Arduino on other Atmel chip" page

Glad to hear! I do a lot of work on the Arduino Playground but sometimes it feels kind of dead. I like wikis better than forums as a way to share and organize knowledge because with forums it seems like the same questions get asked over and over again. There is a lot of information on the Arduino Forum but often you have sort through a lot of pages to find it.
> Who has the right to decide what to delete and what to keep on that site?

It's a publicly editable wiki so really anyone does. Of course anyone has the right to revert your edit also. I actually did remove some ESP8266 cores that had been deleted and removed the word *recommended* from the attiny core since I think ATTinyCore is much better when I added MiniCore to that page. I like Wikipedia's philosophy of "Be bold" and I do a lot of editing of the Playground. I usually only remove dead links, and will even try to replace them with Wayback Machine links when appropriate but I took a look at that ATtiny2313 link and I agree it no longer has any value so I removed it.

MCUdude, 2016-07-25:
>  I like Wikipedia's philosophy of "Be bold" and I do a lot of editing of the Playground. I usually only remove dead links, and will even try to replace them with Wayback Machine links when appropriate but I took a look at that ATtiny2313 link and I agree it no longer has any value so I removed it.

Great to hear! I suggest removing the "<i>ATtiny25/45/85/24/44/84 2313/4313 Alternative - works with 2k chips</i>" link as well, since it's basically replaced with Damellis' and Dr Azzy's core. Google Code is basically dead IIRC anyways.

How about moving the "most useful links" at the top, and the older and obsolete ones at the bottom? (I'm a perfectionist myself but don't want to step on someones toes 😉 )

MCUdude, 2016-10-01:
This issue is certainly solved! Thank you Pert!

closed 2016-10-01 by MCUdude