MCUdude/MicroCore #6: delayMicroseconds() needs to be rewritten

MCUdude, 2016-10-01:
There have been reported on the Arduino forum that the delayMicroseconds doesn't work/freezes the microcontroller if the argument is more than 199. [The issue](https://github.com/MCUdude/MicroCore/blob/72e15d24bf7b373bfbbebd17fe8d61ef41cf557d/avr/cores/microcore/wiring.c#L170) is the uS_new() function, which is some of the leftovers from core13. I'm no assembly guy, so maybe it should be based on _delay_us() from avr_libc instead?

MCUdude, 2016-10-01:
Actually, running a while loop like the delay() function is doing is actually occupying a lot of flash memory. The most efficient way to create a delay function wrapper for _delay_ms and _delay_us is by using the preprocessor:

``` c
#define delay(ms) _delay_ms(ms) // _delay_ms function wrapper
#define delayMicroseconds(us) _delay_us(us) // _delay_us function wrapper
```

nerdralph, 2016-10-02:
Actually, it's not that simple.  Since _delay_xx are compiler built-in inline-only functions, having delay() as a stand-alone function (with repeated calls to _delay_ms(1)) results in a smaller binary when there are a few calls to it.

MCUdude, 2016-10-02:
You're right, the more times i call _delay_ms or _delay_us, the larger the core grows.. So the most efficient efficient way to implement delay functions in the long run is still

``` c
void delay(uint16_t ms)
{
  while(--ms)
    _delay_ms(1);
}

void delayMicroseconds(uint16_t us)
{
  while(--us)
    _delay_us(1);
}
```

 you think?

nerdralph, 2016-10-02:
I'd leave delayMicroseconds as a macro, so that it is more accurate.  The function call overhead would be around 1us, which is insignificant for the delay() function.

p.s. for minimal code size, you should also be using LTO.
http://nerdralph.blogspot.ca/2014/04/gcc-link-time-optimization-can-fix-bad.html

MCUdude, 2016-10-02:
I already have the -flto flag as an option, but do you think it should be the default option instead?
[image]

MCUdude, 2016-10-04:
About the LTO; It's making the code a lot smaller if enabled on other cores too. Are there any downsides with the LTO, or can it/should it be enabled by default on my other cores as well?

closed 2016-10-18 by MCUdude