MCUdude/MicroCore #10: Error when burning Bootloader

Yucheng-L, 2017-01-02:
Hi,

I am very appreciated about you awesome work! I have been using Core13 but it is too simple in its supported functions.

However, there is a problem when I try to upload the Bootloader, it says "ser_open(): can't open device "unknown"", and the uploading fails. How can I fix this?

Thank you.

Yucheng-L, 2017-01-02:
Blow is the message I got. I am using Arduino UNO as programmmer.

Arduino: 1.8.0 (Windows 7), Board: "ATtiny13, 4.8 MHz internal osc., 1.8v, -Os LTO enabled (default)"

avrdude: Version 6.3, compiled on Dec 16 2016 at 13:33:19
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         Using Port                    : COM6
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         Setting bit clk period        : 32.0
         AVR Part                      : ATtiny13
         Chip Erase delay              : 4000 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65     5     4    0 no         64    4      0  4000  4000 0xff 0xff
           flash         65     6    32    0 yes      1024   32     32  4500  4500 0xff 0xff
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          2    0      0     0     0 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9007 (probably t13)
avrdude: erasing chip
avrdude: reading input file "0x3f"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3f:
avrdude: load data lock data from input file 0x3f:
avrdude: input file 0x3f contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 1 bytes of lock verified
avrdude: reading input file "0xfd"
avrdude: writing hfuse (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of hfuse written
avrdude: verifying hfuse memory against 0xfd:
avrdude: load data hfuse data from input file 0xfd:
avrdude: input file 0xfd contains 1 bytes
avrdude: reading on-chip hfuse data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 1 bytes of hfuse verified
avrdude: reading input file "0x39"
avrdude: writing lfuse (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lfuse written
avrdude: verifying lfuse memory against 0x39:
avrdude: load data lfuse data from input file 0x39:
avrdude: input file 0x39 contains 1 bytes
avrdude: reading on-chip lfuse data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 1 bytes of lfuse verified

avrdude done.  Thank you.

avrdude: ser_open(): can't open device "unknown": No such file or directory.

Error while burning bootloader.

MCUdude, 2017-01-02:
> I am very appreciated about you awesome work! I have been using Core13 but it is too simple in its supported functions.

Thanks! I've also used Core13 a little, but it never turned into something serious because the core/code support was so poor and unreliable. When creating MicroCore I discovered a lot of messy and inefficient code, which there's no room for on a 1kB device. The biggest dealbreaker for me was that 1ms actually lasted 1.3ms, because the millis() function was occupying all CPU power.

After you've gotten this core up and running you should experiment with the [core_settings.h](https://github.com/MCUdude/MicroCore/blob/master/avr/cores/microcore/core_settings.h) file, in order to make your code even smaller 😉 
<br>
<br>
> avrdude: ser_open(): can't open device "unknown": No such file or directory.

Not strange at all. The ATtiny13 doesn't have UART, and doesnt run a bootloader.
 according to your error message your're able to set the correct fuses, so the programmer is able to communicate with the microcontroller.

Don't disconnect your programmer from the ATtiny13 after you're burned the bootloader/sat the fuses.
What programmer are your using? In most cases it's just better to buy a cheap USBasp or USBtinyISP rather than to struggle with an Arduino as ISP.

Yucheng-L, 2017-01-02:
Thank you so much for your reply.

> In most cases it's just better to buy a cheap USBasp or USBtinyISP rather than to struggle with an Arduino as ISP.

Yes you are right, Arduino as ISP is TERRIBLE, I  set ATTiny13 working at 128KHz and now it
s dead and can no longer burn new Bootloader (slow mode no responding, normal mode read device ID 0x00), I have purchased a USBtinyISP and another two chips.

> "The biggest dealbreaker for me was that 1ms actually lasted 1.3ms, because the millis() function was occupying all CPU resources."

Really? But doesn't millis() use watchdog timer interrupt to count every 16ms while delay() occupies all CPU resources?  In my code I need a short delay (probably 10-20ms) so I use millis() to set a simple timer because there is another interrupt (comparator). I think he period will be varies if using delay().

MCUdude, 2017-01-02:
I'm very satisfied with [this USBtinyISP](http://www.ebay.com/itm/311566698886). It's tiny, works _all_ the time and has a micro USB connector, rather than the USB-A connector the USBasp has.

> Really? But doesn't millis() use watchdog timer interrupt to count every 16ms while delay() occupies all CPU resources? In my code I need a short delay (probably 10-20ms) so I use millis() to set a simple timer because there is another interrupt (comparator). I think the delay period will be varies if using delay().

The problem with Core13 was that millis() was generated by timer0, which is the only timer available om the ATtiny13. Timer0 is clocked by the internal oscillator, so the millis() function would have to compansate for this. The main problem is that timer0 is set to trigger an interrupt every time it overflows with no prescaler. If you're running the ATtiny13 at 9.6 MHz, this means the CPU will have to handle this event 37500 times EVERY SECOND! Instead, MiniCore uses the internal watchdog, so the CPU will have to hadle this interrupt 62.5 times every second independent of clock frequency. Much better!
The millis() fuction is as accurate as the WDT is, but the only down side is that the millis() function will count with steps of 16, because of the few interrupts every second.

Since the millis() function was frequency dependent with Core13, the function would have to handle every possible frequency. [This gets ugly really fast!](https://github.com/bartgee/at13/blob/master/attiny13/avr/cores/core13/wiring.c#L24). Compared to [the MicroCore code](https://github.com/MCUdude/MicroCore/blob/master/avr/cores/microcore/wiring.c#L36) its just unusable..

Yucheng-L, 2017-01-03:
That's pretty nice! Thank you for helping me solve the problem!

closed 2017-01-03 by Yucheng-L

haindvn, 2017-01-03:
Hi
Thank you very much for your hard-works on supporting ATTiny13, this is great, I can run some test codes on my ATTiny13 via UNO helper, plan to get an USBISP soon. I saw that we're discussing something about timer in this thread so I want just want to clarify something.
I tried to upload the Blink sketch with 1s delay to ATTiny13 (@ 9.6Mhz), however, I realized that the delay time is not 1s at all...it's much more slower than 1s....then I gradually changed the delay time to ~125ms then I see that it's 1s delay as I expected.

May I what I did wrong in this case? What formula can I use to calculate the correct time for delay() function for Attiny13 with the known freq.? i.e. I want to delay 1000ms, the function call delay(130) will do.

Thanks and rgds,

MCUdude, 2017-01-03:
The delay function is based on _delay_ms(), and is allways accurate, independent on clock frequency.
When I developed this core, an imporant feature was accurate timing. When writing the delay() function, I connected my signal genertor to the CLKI pin in the ATtiny13, set it to 9.6 MHz and started measuring pin toggling using a digital oscilloscope. with a _perfect_ clock like a frequency fenerator, fo got it to be accurate down to the microsecond. Still you won't experience this accuracy with the internal oscillator, as it may be as much as 10% off, depending on voltage, temperature and factory calibration.

If timing is completly off, you most likely have the wrong fuse settings. Try hitting "burn bootloader", and verify that the low fuse is sat to **0x3A**

haindvn, 2017-01-03:
Thank you very much for your recommendations, I have just tried to re-burn the boot loader and the problem is solved, I don't why the fuse settings are wrong with this chip because I already went through the boot loader process and thought the fuses are setup correctly, sorry for my false warning....the problem is gone and the delay function is working as it is :) it's so great with this core support.