MCUdude/MicroCore #11: Registers present in the attiny13a but not the attiny13 are not supported

SpenceKonde, 2017-01-19:
The ATTiny13A has a few registers that the ATTiny13 does not (BODSE is the one at issue here, or one of them at least). These defines are not available if you tell the compiler that it's building for a '13 instead of a '13a (the defines are present in avr\iotn13a.h, just not avr\iotn13.h).

See http://forum.arduino.cc/index.php?topic=449324.0 for the issue that prompted this. 

An option to choose which micro is being used is needed, I think.

MCUdude, 2017-01-19:
Thanks for the heads up!
In fact, the core is indeed building for the [ATtiny13 and not the 13A](https://github.com/MCUdude/MicroCore/blob/master/avr/boards.txt#L26). How about I just change this field to `attiny13a` instead? I think it would work just fine on the Attiny13, except for the BOD registers.

MCUdude, 2017-01-28:
Forgot to close this

closed 2017-01-28 by MCUdude