MCUdude/MicroCore #12: High consumption during sleep and not interrupting properly

Nystax, 2017-01-28:
Hi,
I want to build Nikon IR remote based on Attiny13A. Using your core it was easy to create basic working project. 
I want to power my device via battery, so I decided to put Attiny into sleep and interrupt from button, then send code and put into sleep again.
Unfortunetelly when I put Attiny into sleep it draw too much power and interrupt does not work.
But when I uploaded my sketch using smeezekitty's core sleep and interrupt worked perfectly, but not the actual IR code function. 
I'm new to programming mcu's, do you have any advices why I might have such problems?

This is my sleep function:
`void sleep() {
    GIMSK |= _BV(PCIE);                     // Enable Pin Change Interrupts
    PCMSK |= _BV(PCINT3);                   // Use PB3 as interrupt pin
    set_sleep_mode(SLEEP_MODE_PWR_DOWN); 
    cli();
    sleep_enable(); // Sets the Sleep Enable bit in the MCUCR Register (SE BIT)    
    sei();                                  // Enable interrupts
    sleep_cpu();                            // sleep
    cli();                                  // Disable interrupts
    PCMSK &= ~_BV(PCINT3);                  // Turn off PB3 as interrupt pin
    sleep_disable();                        // Clear SE bit
    }`

Setup:
`void setup()   {                
  pinMode(IRledPin, OUTPUT); 
  PORTB |= _BV(PB3);                    //pull-up
  ACSR  |= _BV(ACD);                    // Disable analog comparator                     
  ADCSRA &= ~(1<<ADEN);                  // Disable ADC
}`

Loop:
`void loop()                     
{
sleep();
SendNikonCode(); 
}`

MCUdude, 2017-01-28:
Thanks for using MicroCore. I hope you find it useful 😃 

The last commit I did was to improve the overall power consumption for the ATtiny13. If you manually download and install MicroCore, you an disable BOD, and then decrease the power consumption. [Arduino thread about this HERE](http://forum.arduino.cc/index.php?topic=449324.0).

There's no reason why interrupts shouldn't work. You're using register code, so it doesn't have anything with MicroCore to do.

About your sleep function:
``` c++
void sleep() 
{ 
  GIMSK |= _BV(PCIE); // Enable Pin Change Interrupts 
  PCMSK |= _BV(PCINT3); // Use PB3 as interrupt pin 
  set_sleep_mode(SLEEP_MODE_PWR_DOWN); 
  cli(); 
  sleep_enable(); // Sets the Sleep Enable bit in the MCUCR Register (SE BIT) 
  sei(); // Enable interrupts 
  sleep_cpu(); // sleep 
  cli(); // Disable interrupts 
  PCMSK &= ~_BV(PCINT3); // Turn off PB3 as interrupt pin 
  sleep_disable(); // Clear SE bit 
}
```

One of the last thing in you do in this function is disabling all global interrupts (cli). is this the problem?
You also might want to tweak around in the core_settings.h file to disable all the functions you don't need

Nystax, 2017-01-29:
Thank you quick response. l think I'm getting somewhere. Low power consumption (my multimeter can't even measure that low, below 0.1us) and button interrupts.
Of course, there is "but". After pushing the button attiny seems to be resetting 2 times. 
I've attatched normal led and added blinks of different lenghts in different parts of code.
When I push the button, the device goes to setup, then sleep function, then again setup and sleep.
It never goes beyond sleep.

MCUdude, 2017-01-29:
Does it work if you're using this template instead?

``` c++
int main(void)
{
  // Setup code goes here
}
while(1)
{
  // Loop code goes here
}
```

By using this rather than the setup and loop structure, you're excluding _all_ Arduino code, and the code will be like you used Atmel studio. If this works, We'll might be able to figure out the issue

Nystax, 2017-01-29:
It works exactly the same. 
To give you some data: I'm using Arduino to program Attiny with "Arduino as ISP" programmer.
I seem to have a problem now with buring bootloader (can't exactly say where this started to happen). 

This goes ok:
...avrdude.conf -v -pattiny13 -cstk500v1 -B32 -PCOM4 -b19200 -e -Ulock:w:0x3f:m -Uhfuse:w:0xff:m -Ulfuse:w:0x3A:m 

But I'm getting "ser_open(): can't open device "unknown": No such file or directory" after (strange it is splitted):
...avrdude.conf -v -pattiny13 -cstk500v1 Writing | ################################################## | 100% 0.01s
-B32 -Ulock:w:0x3e:m 

My code right now:
```#include <avr/interrupt.h>
#include <avr/sleep.h>

int IRledPin =  0;
int buttonPin = 3;
int diodepin = 1;

int main(void)                   
{
  // it IR pin as output
  pinMode(IRledPin, OUTPUT); 
  pinMode(diodepin, OUTPUT);
  // button pin pullup
  PORTB |= _BV(PB3);  
  // comparator off
  ACSR  |= _BV(ACD);
  // ADC off
  ADCSRA &= ~(1<<ADEN); 
  // long blink
  digitalWrite(diodepin, HIGH);
  delay(5000);    
  digitalWrite(diodepin, LOW);   
  delay(1000);

while(1){
  sleep();
  SendNikonCode();
}
}

void sleep() {
  // medium blink
  digitalWrite(diodepin, HIGH);
  delay(2000);    
  digitalWrite(diodepin, LOW);   
  delay(1000);
  // Enable Pin Change Interrupts
  GIMSK |= _BV(PCIE);
  // Use PB3 as interrupt pin                     
  PCMSK |= _BV(PCINT3);
  // set sleep mode              
  set_sleep_mode(SLEEP_MODE_PWR_DOWN);
  // disable interrupts
  cli();
  // enable sleep
  sleep_enable();
  // enable interrupts
  sei();
  // go to sleep
  sleep_cpu(); 
  // disable all interrupts to not mess with IR code
  //cli();
  GIMSK &= ~_BV(PCIE); 
  PCMSK &= ~_BV(PCINT3);
  // disable sleep
   sleep_disable();
  // short blink
  digitalWrite(diodepin, HIGH);
  delay(500);    
  digitalWrite(diodepin, LOW);   
  delay(500);
}

After pushing button it goes: 5s->2s->5s->2s, I would expect 0.5s->2s
Do you think it is more general question? Should I ask on Arduino forum?

MCUdude, 2017-01-29:
Please post your entire code,  including your IR code. You should also get yourself a proper programmer, such as the USBasp or the USBtinyISP. Preferably both. This will save you tons of problems in the future. A lot of issues have been solved by not using the Arduino as ISP. 

I'll upload the code to my own attiny13, and see if I experience the same problems as you

Nystax, 2017-01-29:
Thank you for advice, I will definetely do that.

```
#include <avr/interrupt.h>
#include <avr/sleep.h>
int IRledPin =  0;
int buttonPin = 3;
int diodepin = 1;

int main(void)                   
{
  // it IR pin as output
  pinMode(IRledPin, OUTPUT); 
  pinMode(diodepin, OUTPUT);
  // button pin pullup
  PORTB |= _BV(PB3);  
  // comparator off
  ACSR  |= _BV(ACD);
  // ADC off
  ADCSRA &= ~(1<<ADEN);   
  // long blink
  digitalWrite(diodepin, HIGH);
  delay(5000);    
  digitalWrite(diodepin, LOW);   
  delay(1000);

while(1){
  sleep();
  SendNikonCode();
}
}

void sleep() {
  // medium blink
  digitalWrite(diodepin, HIGH);
  delay(2000);    
  digitalWrite(diodepin, LOW);   
  delay(1000);
  // Enable Pin Change Interrupts
  GIMSK |= _BV(PCIE);
  // Use PB3 as interrupt pin                     
  PCMSK |= _BV(PCINT3);
  // set sleep mode              
  set_sleep_mode(SLEEP_MODE_PWR_DOWN);
  // disable interrupts
  cli();
  // enable sleep
  sleep_enable();
  // enable interrupts
  sei();
  // go to sleep
  sleep_cpu(); 
  // disable all interrupts to not mess with IR code
  //cli();
  GIMSK &= ~_BV(PCIE); 
  PCMSK &= ~_BV(PCINT3);
  // disable sleep
    sleep_disable();
  // short blink
  digitalWrite(diodepin, HIGH);
  delay(500);    
  digitalWrite(diodepin, LOW);   
  delay(500);
}

 
// This procedure sends a 38KHz pulse to the IRledPin 
// for a certain # of microseconds. We'll use this whenever we need to send codes
void pulseIR(long microsecs) {
  // we'll count down from the number of microseconds we are told to wait

  cli();  // this turns off any background interrupts
 
  while (microsecs > 0) {
    // 38 kHz is about 13 microseconds high and 13 microseconds low
   digitalWrite(IRledPin, HIGH);  // this takes about 3 microseconds to happen
   delayMicroseconds(10);         // hang out for 10 microseconds, you can also change this to 9 if its not working
   digitalWrite(IRledPin, LOW);   // this also takes about 3 microseconds
   delayMicroseconds(10);         // hang out for 10 microseconds, you can also change this to 9 if its not working
 
   // so 26 microseconds altogether
   microsecs -= 26;
  }
 
  sei();  // this turns them back on
}
 

void SendNikonCode() {
  // This is the code for my particular Nikon, for others use the tutorial
  // to 'grab' the proper code from the remote
  pulseIR(2080);
  delayMicroseconds(27);
  pulseIR(440);
  delayMicroseconds(1500);
  pulseIR(460);
  delayMicroseconds(3440);
  pulseIR(480);
 
 
  delay(65); // wait 65 milliseconds before sending it again
 
  pulseIR(2000);
  delay(27);
  pulseIR(440);
  delayMicroseconds(1500);
  pulseIR(460);
  delayMicroseconds(3440);
  pulseIR(480);
}
```
I've used the tutorial: http://www.ladyada.net/wiki/tutorials/learn/sensors/ir.html

MCUdude, 2017-01-30:
I've done some research with my own hardware, and I get the same results as you.
the `set_sleep_mode(SLEEP_MODE_PWR_DOWN);` function powers of the oscillator, but can be woken up by using "regular" interrupt, PCINT, or a watch dog reset. It turns out that the microcontroller resets itself after woken up from power down. I created a simple sketch just for testing the sleep functionality, and `set_sleep_mode(SLEEP_MODE_IDLE);`, which is the lightest sleep mode.

``` cpp
#include <avr/sleep.h>
#include <avr/sfr_defs.h>
#include <util/delay.h>
#include <avr/interrupt.h>

int buttonPin = 3;
int diodepin = 2;

void sleepNow()
{
    // Choose our preferred sleep mode:
    set_sleep_mode(SLEEP_MODE_IDLE);
 
    // Set sleep enable (SE) bit:
    sleep_enable();
 
    // Put the device to sleep:
    sleep_mode();
 
    // Upon waking up, sketch should contiye from this point.
    sleep_disable();
}

int main(void)
{
  DDRB |= _BV(diodepin);

  // Enable PCINT3
  PORTB |= _BV(PB3);
  GIMSK |= _BV(PCIE);
  PCMSK |= _BV(PCINT3);
  sei();
  
  // long blink
  PORTB |= _BV(diodepin);
  _delay_ms(5000);
  PORTB &= ~_BV(diodepin);
  _delay_ms(1000);
  
  while(1) 
  {
    PORTB |= _BV(diodepin);
    _delay_ms(2000);
    PORTB &= ~_BV(diodepin);
    _delay_ms(1000);
  
    sleepNow();
    // short blink
    PORTB |= _BV(diodepin);
    _delay_ms(500);
    PORTB &= ~_BV(diodepin);
    _delay_ms(500);
  }
}
```

I tested this code using MicroCore, [this fork of Smeezekitty's core13](https://github.com/bartgee/at13) and using "pure" avr-gcc and makefiles, and the MCU always reset itself after wakeup. Maybe the ATtiny13 isn't able to wake up without resetting?

nerdralph, 2017-01-30:
You need an empty ISR.  The default interrupt vectors jump to the reset
vector.

MCUdude, 2017-01-30:
Wow, thanks @nerdralph 😃 That did the trick!

@Nystax try adding:
``` cpp
// Empty PCINT ISR
ISR(PCINT0_vect)
{
  
}
```

to your Nikon IR sketch and try again 👍

Nystax, 2017-01-30:
Wow guys, thank you! It works. And I've learned so much.

closed 2017-01-30 by Nystax