MCUdude/MicroCore #13: Cannot find plugin 'liblto_plugin-0.dll'

oocx, 2017-03-19:
I'm trying to compile a simple blink example using Arduino 1.8.1 on Windows 10.

I get the following error:

`C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\mathi\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\mathi\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\mathi\Documents\Arduino\libraries -fqbn=MicroCore:avr:attiny13:clock=9M6,BOD=2v7,flag=Os_flto -ide-version=10801 -build-path C:\temp\arduino_build_923402 -warnings=none -prefs=build.warn_data_percentage=75 -verbose C:\Users\mathi\Documents\Arduino\blink\blink.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\mathi\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\mathi\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\mathi\Documents\Arduino\libraries -fqbn=MicroCore:avr:attiny13:clock=9M6,BOD=2v7,flag=Os_flto -ide-version=10801 -build-path C:\temp\arduino_build_923402 -warnings=none -prefs=build.warn_data_percentage=75 -verbose C:\Users\mathi\Documents\Arduino\blink\blink.ino
Using board 'attiny13' from platform in folder: C:\Users\mathi\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\1.0.1
Using core 'microcore' from platform in folder: C:\Users\mathi\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\1.0.1
Detecting libraries used...
"C:\Users\mathi\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=attiny13 -DF_CPU=9600000L -DARDUINO=10801 -DARDUINO_attiny -DARDUINO_ARCH_AVR -Wextra -flto  "-IC:\Users\mathi\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\1.0.1\cores\microcore" "C:\temp\arduino_build_923402\sketch\blink.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Users\mathi\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=attiny13 -DF_CPU=9600000L -DARDUINO=10801 -DARDUINO_attiny -DARDUINO_ARCH_AVR -Wextra -flto  "-IC:\Users\mathi\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\1.0.1\cores\microcore" "C:\temp\arduino_build_923402\sketch\blink.ino.cpp" -o "C:\temp\arduino_build_923402\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\Program Files (x86)\Arduino\tools-builder\ctags\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\temp\arduino_build_923402\preproc\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\Users\mathi\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=attiny13 -DF_CPU=9600000L -DARDUINO=10801 -DARDUINO_attiny -DARDUINO_ARCH_AVR -Wextra -flto  "-IC:\Users\mathi\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\1.0.1\cores\microcore" "C:\temp\arduino_build_923402\sketch\blink.ino.cpp" -o "C:\temp\arduino_build_923402\sketch\blink.ino.cpp.o"
Compiling libraries...
Compiling core...
Using previously compiled file: C:\temp\arduino_build_923402\core\WInterrupts.c.o
Using previously compiled file: C:\temp\arduino_build_923402\core\wiring.c.o
Using previously compiled file: C:\temp\arduino_build_923402\core\wiring_analog.c.o
Using previously compiled file: C:\temp\arduino_build_923402\core\wiring_digital.c.o
Using previously compiled file: C:\temp\arduino_build_923402\core\wiring_pulse.c.o
Using previously compiled file: C:\temp\arduino_build_923402\core\wiring_pwm.c.o
Using previously compiled file: C:\temp\arduino_build_923402\core\wiring_shift.c.o
Using previously compiled file: C:\temp\arduino_build_923402\core\Print.cpp.o
Using previously compiled file: C:\temp\arduino_build_923402\core\WMath.cpp.o
Using previously compiled file: C:\temp\arduino_build_923402\core\WString.cpp.o
Using previously compiled file: C:\temp\arduino_build_923402\core\main.cpp.o
"C:\Users\mathi\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-gcc-ar" rcs  "C:\temp\arduino_build_923402\core\core.a" "C:\temp\arduino_build_923402\core\WInterrupts.c.o"
C:\Users\mathi\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-gcc-ar: Cannot find plugin 'liblto_plugin-0.dll'

exit status 1
Error compiling for board ATtiny13.`

MCUdude, 2017-03-19:
The IDE is using avr-gcc 4.8.1 instead of 4.9.2. See [MightyCore issue #40](https://github.com/MCUdude/MightyCore/issues/40).

oocx, 2017-03-19:
Thanks for your fast response!

There was an older avr-gcc in "..AppData\Local\Arduino15\packages\arduino\tools\avr-gcc". I deleted that folder and can now compile without errors.

closed 2017-03-19 by oocx

MCUdude, 2017-03-19:
Great, thanks for using MicroCore! If you like it, please add a star ⭐️ 😉