MCUdude/MicroCore #15: This code worked with the previous version of micro core

caiopoit, 2017-04-20:
First, thanks for your job of creating a good core for ATTINY13.

I don't see the problem. Win10Home IDE 1.8.2

`void setup() {
  OSCCAL += 3; 
pinMode(3,INPUT);  
pinMode(0,OUTPUT);
pinMode(4,INPUT);  
pinMode(1,OUTPUT);
}

void loop() {

 if(pulseIn(3, HIGH,1000000)>1500){
  digitalWrite(0,HIGH);
 }
 else{
  digitalWrite(0,LOW);
 }
 
  if(pulseIn(4,HIGH,1000000)>1500){
  digitalWrite(1,HIGH);
 }
 else{
  digitalWrite(1,LOW);
 }
}`

_Build options changed, rebuilding all
Archiving built core (caching) in: C:\Users\CAIOPO~1\AppData\Local\Temp\arduino_cache_94127\core\core_MicroCore_avr_attiny13_clock_9M6,BOD_4v3,flag_Os_flto_3f822feb56820af16374d894820ba573.a
C:\Users\CAIOPO~1\AppData\Local\Temp\ccol7Afh.ltrans0.ltrans.o: In function `pulseIn.constprop.0':

ccol7Afh.ltrans0.o:(.text+0x44): undefined reference to `micros'

ccol7Afh.ltrans0.o:(.text+0x52): undefined reference to `micros'

ccol7Afh.ltrans0.o:(.text+0x72): undefined reference to `micros'

ccol7Afh.ltrans0.o:(.text+0x92): undefined reference to `micros'

ccol7Afh.ltrans0.o:(.text+0xa0): undefined reference to `micros'

C:\Users\CAIOPO~1\AppData\Local\Temp\ccol7Afh.ltrans0.ltrans.o:ccol7Afh.ltrans0.o:(.text+0xba): more undefined references to `micros' follow

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board ATtiny13._

MCUdude, 2017-04-20:
the `pulseIn` function depends on micros(). [Micros is diabled by default to save space](https://github.com/MCUdude/MicroCore/blob/master/avr/cores/microcore/core_settings.h#L31). You can edit the core_settings.h file to enable or disable core functionality

caiopoit, 2017-04-22:
It worked, thanks! Just one more thing, if i want to use a different frequency for the processor, do I have to burn bootloader? Because I've had some problems with that... I'm using an uno as programmer.

MCUdude, 2017-04-22:
Yes, you'll need to burn bootloader in order to set the fuses that determines the clock frequency.

closed 2017-04-22 by MCUdude