MCUdude/MicroCore #16: Attiny13 ADC only working on A3

12364789, 2017-04-30:
Wrong coding in wiring_analog.c
I fixed it this way, but there is probably a better way (I am newbie):
int16_t analogRead(uint8_t pin)
{
	uint8_t l,h,p;
	//FAULT ---> ADMUX = (ADMUX & _BV(REFS0)) | (pin & 0x03); // Setup ADC, preserve REFS0
	if(pin == 5)p=0;
	if(pin == 2)p=1;
	if(pin == 4)p=2;
	if(pin == 3)p=3;
	ADMUX = (ADMUX & _BV(REFS0)) | (p & 0x03);	
       ...

Now I could get it working on A2/PB4 as I first tried when it wouldn't work...

MCUdude, 2017-04-30:
You're absolutely right, thank you for reporting! Tuned out that the A0..A3 macros where wrong. Have tested it, and would be working just fine now 👍

12364789, 2017-04-30:
Yes it works! Tested on A2... Thank you!!

closed 2017-04-30 by 12364789