MCUdude/MicroCore #20: Cannot find plugin 'liblto_plugin-0.dll'

tekk, 2017-05-13:
I'm getting this error while compiling on WIndows 10, Arduino 1.8.2 (store version).

MCUdude, 2017-05-13:
Please turn on verbose compilation and upload in the IDE settings and post the output. I'm sure the problem is because the IDE is using an older version of avr-gcc, most likely 4.8.1.

tekk, 2017-05-14:
```Arduino: 1.8.2 (Windows Store 1.8.3.0) (Windows 10), Board: "ATtiny13, 9.6 MHz internal osc., 2.7v, -Os LTO enabled (default)"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.3.0_x64__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.3.0_x64__mdqgnx93n4wtt\hardware -hardware C:\Users\tekks\Documents\ArduinoData\packages -hardware D:\Dropbox\Moje\hardware -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.3.0_x64__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.3.0_x64__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\tekks\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.3.0_x64__mdqgnx93n4wtt\libraries -libraries D:\Dropbox\Moje\libraries -fqbn=MicroCore:avr:attiny13:clock=9M6,BOD=2v7,flag=Os_flto -ide-version=10802 -build-path C:\Users\tekks\AppData\Local\Temp\arduino_build_2736 -warnings=none -build-cache C:\Users\tekks\AppData\Local\Temp\arduino_cache_52680 -prefs=build.warn_data_percentage=75 -verbose C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.3.0_x64__mdqgnx93n4wtt\examples\01.Basics\Blink\Blink.ino
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.3.0_x64__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.3.0_x64__mdqgnx93n4wtt\hardware -hardware C:\Users\tekks\Documents\ArduinoData\packages -hardware D:\Dropbox\Moje\hardware -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.3.0_x64__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.3.0_x64__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\tekks\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.3.0_x64__mdqgnx93n4wtt\libraries -libraries D:\Dropbox\Moje\libraries -fqbn=MicroCore:avr:attiny13:clock=9M6,BOD=2v7,flag=Os_flto -ide-version=10802 -build-path C:\Users\tekks\AppData\Local\Temp\arduino_build_2736 -warnings=none -build-cache C:\Users\tekks\AppData\Local\Temp\arduino_cache_52680 -prefs=build.warn_data_percentage=75 -verbose C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.3.0_x64__mdqgnx93n4wtt\examples\01.Basics\Blink\Blink.ino
Using board 'attiny13' from platform in folder: C:\Users\tekks\Documents\ArduinoData\packages\MicroCore\hardware\avr\1.0.2
Using core 'microcore' from platform in folder: C:\Users\tekks\Documents\ArduinoData\packages\MicroCore\hardware\avr\1.0.2
Detecting libraries used...
"C:\Users\tekks\Documents\ArduinoData\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=attiny13a -DF_CPU=9600000L -DARDUINO=10802 -DARDUINO_attiny -DARDUINO_ARCH_AVR -Wextra -flto  "-IC:\Users\tekks\Documents\ArduinoData\packages\MicroCore\hardware\avr\1.0.2\cores\microcore" "C:\Users\tekks\AppData\Local\Temp\arduino_build_2736\sketch\Blink.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Users\tekks\Documents\ArduinoData\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=attiny13a -DF_CPU=9600000L -DARDUINO=10802 -DARDUINO_attiny -DARDUINO_ARCH_AVR -Wextra -flto  "-IC:\Users\tekks\Documents\ArduinoData\packages\MicroCore\hardware\avr\1.0.2\cores\microcore" "C:\Users\tekks\AppData\Local\Temp\arduino_build_2736\sketch\Blink.ino.cpp" -o "C:\Users\tekks\AppData\Local\Temp\arduino_build_2736\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.3.0_x64__mdqgnx93n4wtt\tools-builder\ctags\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\tekks\AppData\Local\Temp\arduino_build_2736\preproc\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\Users\tekks\Documents\ArduinoData\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=attiny13a -DF_CPU=9600000L -DARDUINO=10802 -DARDUINO_attiny -DARDUINO_ARCH_AVR -Wextra -flto  "-IC:\Users\tekks\Documents\ArduinoData\packages\MicroCore\hardware\avr\1.0.2\cores\microcore" "C:\Users\tekks\AppData\Local\Temp\arduino_build_2736\sketch\Blink.ino.cpp" -o "C:\Users\tekks\AppData\Local\Temp\arduino_build_2736\sketch\Blink.ino.cpp.o"
Compiling libraries...
Compiling core...
"C:\Users\tekks\Documents\ArduinoData\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=attiny13a -DF_CPU=9600000L -DARDUINO=10802 -DARDUINO_attiny -DARDUINO_ARCH_AVR -Wextra -flto  "-IC:\Users\tekks\Documents\ArduinoData\packages\MicroCore\hardware\avr\1.0.2\cores\microcore" "C:\Users\tekks\Documents\ArduinoData\packages\MicroCore\hardware\avr\1.0.2\cores\microcore\WInterrupts.c" -o "C:\Users\tekks\AppData\Local\Temp\arduino_build_2736\core\WInterrupts.c.o"
"C:\Users\tekks\Documents\ArduinoData\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=attiny13a -DF_CPU=9600000L -DARDUINO=10802 -DARDUINO_attiny -DARDUINO_ARCH_AVR -Wextra -flto  "-IC:\Users\tekks\Documents\ArduinoData\packages\MicroCore\hardware\a [... cut ...] \tekks\AppData\Local\Temp\arduino_build_2736\core\main.cpp.o"
"C:\Users\tekks\Documents\ArduinoData\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-gcc-ar" rcs  "C:\Users\tekks\AppData\Local\Temp\arduino_build_2736\core\core.a" "C:\Users\tekks\AppData\Local\Temp\arduino_build_2736\core\WInterrupts.c.o"
C:\Users\tekks\Documents\ArduinoData\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-gcc-ar: Cannot find plugin 'liblto_plugin-0.dll'

exit status 1
Error compiling for board ATtiny13.

tekk, 2017-05-14:
So I see it's using 4.8.1... Tested on Arduino 1.6.13 and it compiles fine. Haven't tried uploading yet.
Are there any plans for newer Arduino versions support?

MCUdude, 2017-05-14:
All IDE version from 1.6.3 and newer are supported. The problem is that the older avr-gcc 4.8.1 is still being used. Try opening the boards manager, and update the *Arduino AVR boards* to the latest version. Should be v1.6.18 by now. By doing this avr-gcc v4.9.2 will be downloaded, and that version supports LTO 👍

tekk, 2017-05-14:
Thank you for your support!

MCUdude, 2017-05-14:
Did it solve the problem?

tekk, 2017-05-15:
Unfortunately, not.
I've got _Arduino AVR boards_ version **1.6.18** installed in the IDE, but when I try to compile, verbose output shows that is uses the old `avr-gcc` **4.8.1**.
This is not an official ZIP/EXE installation of the Arduino IDE, but a Windows 10 Store "Metro" version of the app.

MCUdude, 2017-05-15:
You probably have both versions installed, and you probably need to delete 4.8.1
@per1234 du you know the compiler path on Windows?

BTW why use the Windows store version, and not the "real" one?

per1234, 2017-05-15:
With the regular IDE it's at:
```
C:\Users\{username}\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\{version}
```
but it looks like it might be in a different location when using the Windows Store version:
```
C:\Users\tekks\Documents\ArduinoData\packages\arduino\tools\avr-gcc
```

tekk, 2017-05-15:
I always use ZIP "portable" Arduino versions, don't install them. So this is only one I have.
And I've used the store version because it integrates better with Windows (i.e. you can pin it to taskbar and it properly shows icon, etc.)

tekk, 2017-05-17:
Recently I've read, that there is an option to change `avr-gcc` version when installing from Boards Manager.

MCUdude, 2019-12-23:
With the recent Arduino IDE versions (and boards manager versions) this isn't really an issue anymore, and/or sure you have a fairly new IDE version installed and a fairly new *Arduino AVR boards* boards manager package installed.

closed 2019-12-23 by MCUdude