MCUdude/MicroCore #21: Sorry - this program has been built without plugin support

tekk, 2017-05-13:
I'm getting this error `sorry - this program has been built without plugin support` while trying to compile on Raspberry Pi 3, Arduino version 1.6.12.

MCUdude, 2017-05-13:
Follow the same procedure as in #20

tekk, 2017-05-14:
Well this one's strange, I just ran `sudo apt-get autoremove`, which removed `arduino-core` (from default Raspbian repository, I think it's old Arduino 1.0.5), and some other packages I don't remember and now it compiles fine.

Thank you for this awesome project! 👍

closed 2017-05-14 by tekk

Nase00, 2019-07-19:
Sorry to necro an old issue, but I landed here while experiencing the same problem, and wanted to share the solution that worked for me in case anyone else lands here from Google.

https://forum.arduino.cc/index.php?topic=416897.msg2871731#msg2871731

> A bunch of people have had IDE 1.6.x problems, including IDE crashes and bizarre compile errors, when they had a previous 1.5.x/1.6.x installation.  At least some of those problems seem to be caused by out-of-date executables that the IDE saves outside the IDE installation directory.  Perhaps that is what is causing your particular problem.  In such cases it appears that the fix is to delete or rename the old "Arduino15" directory and re-install 1.6.x.
>
>On Mac OS X:
/Users/(username)/Library/Arduino15
>
>On Windows delete or rename both:
Arduino IDE 1.6.5r5 and previous: C:\Users\(username)\AppData\Roaming\Arduino15
Arduino IDE 1.6.6 and later:      C:\Users\(username)\AppData\Local\Arduino15
>
>On Linux:
/home/(username)/.arduino15    (a.k.a. ~/.arduino15)

I renamed the `Arduino15` to `Arduino15.bk`, restarted the IDE, and it worked. Not sure if this break my ESP etc. configurations, but I should be able to revert easily enough if need be.