MCUdude/MicroCore #22: "Slow" programmers not recognized

kirchnet, 2017-05-13:
When selection 128 kHz AVRdude (via the IDE) does not recognize the slow programmers
I use IDE version 1.8.2 and the latest MicroCore via JSON (under "Preferences")
Works fine at 600kHz but somehow the slow programmers don't work, no matter which one of them I select in the IDE
This is the error thrown by AVRdude in verbose output
> avrdude: Version  6.3, compiled on Jan 17 2017 at 12:00:53
>          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
>          Copyright (c) 2007-2014 Joerg Wunsch
> 
>          System wide configuration file is "C:\Users\kirchnet\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\1.0.2/avrdude.conf"
> 
> 
> avrdude: Can't find programmer id "{upload.protocol}"

MCUdude, 2017-05-13:
You're right, but i'm afraid there's not much I can do about it..

The slow programmers gets added to the IDE through MicroCore through a separate file. The problem is that the IDE cannot use custom programmers while uploading. Since the ATtiny13 depends on ISP programming, this _will_ become an issue.

The only "fix" I have for this is to remove the slow programmers and the 128 kHz clock option..

kirchnet, 2017-05-13:
Can I circumvent it by running AVRdude through the command line? If so, do you happen to know the command line options?

MCUdude, 2017-05-13:
Yes, you can. Start off by enabling verbose uploading in the IDE. Then you need to copy the path to avrdude, the path to the avrdude.conf file, and the hex file. You'll also need to replace *programmers* with your actual programmer

`avrdude -C /installation_path/MicroCore/avr/avrdude.conf -v -pattiny13 -c programmer -B32 -Uhfuse:w:0xfb:m -Ulfuse:w:0x3A:m -Uflash:w:/path_to/hexfile.hex:i `

kirchnet, 2017-05-14:
MCUdude, thank you for the tip. After some experimentation I found a way to make it work using the IDE and an Arduino acting as ISP. For the benefit of others here are the steps:
1. Open the sketch "ArduinoISP" that is under the File->Examples menu
2. Find the line **`#define SPI_CLOCK     (1000000/6)`** and replace  (1000000/6) by (128000/6) . In my case it is in line 53.
3. Find the line `_#define BAUDRATE	115200_ `and replace the 115200 by 19200 . In my case it was in line 145.
4. Upload this modified ArduinoISP sketch to the Arduino in the usual manner.
5. Wire the Attiny13 per the instructions in the ArduinoISP sketch. There are also plenty of websites that show the wiring with prettier graphics.
6. When uploading your sketch to the ATtiny13 select "Arduino as ISP" as the programmer. There are two others with similar names that wouldn't work for me. Make sure it's the one labeled exactly "Arduino as ISP"
7. Select "Sketch->Upload using Programmer" or Ctrl+Shft+U to upload
I was able to upload a simple Blink sketch as a test in this manner. IDE version 1.8.2