MCUdude/MicroCore #23: ISR(WDT_vect) error for ATtiny13

aamantov, 2017-05-20:
I am new with Arduino programming, mainly with ATtiny13, but am trying to use WDT as interrupt together with sleep in order to save power in a battery-based timer. The compiler in IDE 1.6.11 gives the error:
...
Compiling core...
Usando arquivo compilado anteriormente: C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\WInterrupts.c.o
Usando arquivo compilado anteriormente: C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\wiring.c.o
Usando arquivo compilado anteriormente: C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\wiring_analog.c.o
Usando arquivo compilado anteriormente: C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\wiring_digital.c.o
Usando arquivo compilado anteriormente: C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\wiring_pulse.c.o
Usando arquivo compilado anteriormente: C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\wiring_pwm.c.o
Usando arquivo compilado anteriormente: C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\wiring_shift.c.o
Usando arquivo compilado anteriormente: C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\Print.cpp.o
Usando arquivo compilado anteriormente: C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\WMath.cpp.o
Usando arquivo compilado anteriormente: C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\WString.cpp.o
Usando arquivo compilado anteriormente: C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\main.cpp.o
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs  "C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\core.a" "C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\WInterrupts.c.o"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs  "C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\core.a" "C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\wiring.c.o"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs  "C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\core.a" "C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\wiring_analog.c.o"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs  "C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\core.a" "C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\wiring_digital.c.o"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs  "C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\core.a" "C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\wiring_pulse.c.o"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs  "C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\core.a" "C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\wiring_pwm.c.o"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs  "C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\core.a" "C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\wiring_shift.c.o"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs  "C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\core.a" "C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\Print.cpp.o"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs  "C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\core.a" "C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\WMath.cpp.o"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs  "C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\core.a" "C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\WString.cpp.o"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs  "C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\core.a" "C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\core\main.cpp.o"
Linking everything together...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc" -w -Os -Wl,--gc-sections -mmcu=attiny13 -w -flto -o "C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp/Timer_NO_Contact_Light_Configuravel_10msON_Power_Down.ino.elf" "C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\sketch\Timer_NO_Contact_Light_Configuravel_10msON_Power_Down.ino.cpp.o" "C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp/core\core.a" "-LC:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp" -lm
wiring.c.o (symbol from plugin): In function `__vector_8':

(.text+0x0): multiple definition of `__vector_8'

C:\Users\Angelo\AppData\Local\Temp\build47f663238a2af287589b5a7c919d7973.tmp\sketch\Timer_NO_Contact_Light_Configuravel_10msON_Power_Down.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Erro compilando para a placa ATtiny13
================================
If I change the ISR vector name WDT_vect (highlighted in edition = reserved label ) to other name say WATCHDOG (non-highlighted in edition = ordinary label) the error disappears.
Any clue why this is happening?
Thanks.

MCUdude, 2017-05-20:
Yes, there's a good reason for that error, and there is a simple solution for it.

The ATtiny13 only has one timer, timer0. On "regular" arduinos like the UNO, this timer is used for generating `millis()` and `micros()`. On the ATtiny13 I'm using the watchdog for generating millis and micros, which frees timer0. The reason for the error you're seeing is that *WDT_vect* is already [defined here](https://github.com/MCUdude/MicroCore/blob/master/avr/cores/microcore/wiring.c#L27).

If you need to use the WDT, you can disable millis from the [core_settings.h file](https://github.com/MCUdude/MicroCore#core-settings)

aamantov, 2017-05-21:
Hans, Thank you for your immediate reply. So, I conclude that I have to give up using millis() and micros() in order to have WDT available to be programmed as 2 seconds (or more up to 8s I guess) interrupt for my application. I assume that I will have to implement a private delay since i need some and it also depends the millis() function. Am I correct? I wonder if I can implement a unique 10ms delay using a for or a while loop with an argument that takes the time I need. Obviously I will have to count the number of clock cycles it will take based on the clock I choose for the Attiny13.Other odd (for me) thing : I was searching the file core_settings.h you indicated to disable the function millis(). Neverthless the link : https://mcudude.github.io/MicroCore/package_MCUdude_MicroCore_index.json is present in Preferences > Additional Boards Manager URLs, I could not see the right files. What I have in C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino, among many files, is the wiring.c  file that is different of your file. Probably this was my mistake in previous attempts to install the Attiny13 core. It appears that the wiring.c file is the David Mellis one used in ATtiny85 former installation.Do you have an idea what happens, because the Tools > Boards > Microcore > ATtiny13 is present in IDE 1.6.11 that I have and I don´t know where it comes from.Thank you for your patience.Angelo

 

 
|  | Livre de vírus. www.avast.com.  |

per1234, 2017-05-21:
>What I have in C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino, among many files, is the wiring.c  file that is different of your file.

Those files are not used by MicroCore. It includes its own core. Those files are the Arduino AVR Boards `arduino:arduino` core. They are used by damellis/attiny but were not installed with that package, it just references the Arduino AVR Boards `arduino:arduino` core. David Mellis' name is on the copyright information of that file because he was one of the original members of the Arduino project.

Here's how you can easily find MicroCore's core_settings.h file on your computer:
1. Tools > Board > ATtiny13
2. File > Examples > AVR C code examples > Blink
3. Sketch > Show sketch folder - This will open the folder containing the example, probably C:\Users\Angelo\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\1.0.2\libraries\AVR_examples\examples\Blink
4. Navigate to C:\Users\Angelo\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\1.0.2\cores\microcore. In that folder you will find the file core_settings.h.

aamantov, 2017-05-22:
Hans,Yes! It compiled apparently without error! Thank you very much.
One little thing more: Will it work if I use the _delay_ms() from AVR Libc? In this case should I declare the F_CPU full or /8For instance: the ATtiny at 4.8MHz should I declare:
#define F_CPU 4800000UL  // 4.8 MHz#include <util/delay.h>
or 
#define F_CPU 600000UL  // 600KHz#include <util/delay.h>?
Thanks again.I confess that I was afraid to ask you silly questions as I did, but you treat them professionally. Congratulations.Angelo 

 

 

   - Tools > Board > ATtiny13

aamantov, 2017-05-22:
Thanks to Hans and per1234!

MCUdude, 2017-05-23:
you can use `_delay_ms()` in your code, no problem there. Just have in mind that it's an inline function, so the compiled code size will increase every time _delay_ms() is added. The `delay()` function [deals with this issue nicely](https://github.com/MCUdude/MicroCore/blob/d4c3756a44a1e831a859e8bc61a50444cc8e99ee/avr/cores/microcore/wiring.c#L122).

aamantov, 2017-05-23:
Thank you for the advise. I tought the delay() used millis() that I unselected in order to use the WDT_vect().I think all is done now. Thanks again.

closed 2017-05-23 by MCUdude