MCUdude/MicroCore #25: Code freezes Attiny

caiopoit, 2017-06-16:
Any idea why this code does not work with micro core even though it compiles?
https://forum.arduino.cc/index.php?topic=483961.0

`unsigned long previousMillis = 0;
#define interval 1000
/////////////////////////////////////////////////////////
void Blink();

void setup() {
  OSCCAL += 3;
  pinMode(3, INPUT);
  pinMode(0, OUTPUT);
  pinMode(4, INPUT);
  pinMode(1, OUTPUT);
}
/////////////////////////////////////////////////////////
void loop() {

  if (pulseIn(3, HIGH, 1000000) > 1500) {
    digitalWrite(0, HIGH);
  }
  else {
    digitalWrite(0, LOW);
  }

  if (pulseIn(4, HIGH, 1000000) > 1500) {
    Blink();
  }
  else {
    digitalWrite(1, LOW);
  }
}
//////////////////////////////////////////////////////////
void Blink() {

  if (millis() - previousMillis >= interval) {
    digitalWrite(1, HIGH);
    delay(70);
    digitalWrite(1, LOW);
    delay(200);
    digitalWrite(1, HIGH);
    delay(70);
    digitalWrite(1, LOW);
    previousMillis = millis();
  }
}`

MCUdude, 2017-06-17:
You probably need to enable micros in the core settings file. pulseIn depends on it

caiopoit, 2017-06-17:
It is enabled.

MCUdude, 2017-06-18:
I'm not able to do any testing in the near future. Try to isolate to issue as much as possible by removing everything unnecessary. Then find out where the code freezes, and which line that causes it.
Good luck 🙂

caiopoit, 2017-06-19:
The parts of the code work when they are splited, that is why it is strange.

caiopoit, 2017-07-14:
No one has an idea? It would be really helpful.

smeezekitty, 2017-07-22:
Is it actually freezing or is it getting stuck in a loop in pulseIn?

caiopoit, 2017-07-28:
How am I supposed to check those two options? I don't know wich one is actually occurring...

MCUdude, 2017-08-13:
I'm working on a new implementation of pulseIn which does depend on micros() to work. Does your code run if you replace pulseIn with pulseInNew?

``` c++
uint32_t pulseInNew(uint8_t pin, uint8_t state, uint32_t timeout)
{
	uint8_t stateMask = (state ? _BV(pin) : 0);
	
	uint32_t width = 0; // keep initialization out of time critical area
	
	// convert the timeout from microseconds to a number of times through
	// the initial loop; it takes 16 clock cycles per iteration.
	uint32_t numloops = 0;
	uint32_t maxloops = microsecondsToClockCycles(timeout) / 16;
	
	// wait for any previous pulse to end
	while (PINB & _BV(pin) == stateMask)
		if (numloops++ == maxloops)
			return 0;
	
	// wait for the pulse to start
	while (PINB & _BV(pin) != stateMask)
		if (numloops++ == maxloops)
			return 0;
	
	// wait for the pulse to stop
	while (PINB & _BV(pin) == stateMask) {
		if (numloops++ == maxloops)
			return 0;
		width++;
	}

	// convert the reading to microseconds. The loop has been determined
	// to be 20 clock cycles long and have about 16 clocks between the edge
	// and the start of the loop. There will be some error introduced by
	// the interrupt handlers.
	return clockCyclesToMicroseconds(width * 21 + 16); 
}
```

caiopoit, 2017-08-16:
I get this error:

> 
Build options changed, rebuilding all
C:\Users\Caio Poit\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5/bin/avr-gcc-ar: Cannot find plugin 'liblto_plugin-0.dll'

exit status 1
Error compiling for board ATtiny13.

My ICG is 4.8.1, how do I get the proper one?
And about the code above, I just updated the microcore, does it come with the newer version?

mcuee, 2023-05-12:
Old issue like this can probably be closed.

closed 2023-05-12 by MCUdude