MCUdude/MicroCore #26: can't use this library with attiny13

LeonRonin, 2017-07-12:
hi
i'm trying to make a IR remote for my dslr using this library:
https://github.com/outofjungle/NikonRemote
it compiles and works fine on my arduino uno but when I compile it for attiny13 it returns "error compiling for board attiny13"
do you know what is causing the error?

MCUdude, 2017-07-12:
Post your full error, and then we'll talk 😊

MCUdude, 2017-07-13:
you probably need to enable `micros()`. You can do so by changing the [core settings](https://github.com/MCUdude/MicroCore#core-settings).

LeonRonin, 2017-07-13:
this is the error message:
Arduino: 1.8.0 (Windows 10), Board: "ATtiny13, 9.6 MHz internal osc., 2.7v, -Os LTO enabled (default)"

C:\Users\Mohammad\AppData\Local\Temp\cchEWthY.ltrans0.ltrans.o: In function `NikonRemote::pulse_off(unsigned long) [clone .constprop.4]':

cchEWthY.ltrans0.o:(.text+0x22): undefined reference to `micros'

cchEWthY.ltrans0.o:(.text+0x2c): undefined reference to `micros'

C:\Users\Mohammad\AppData\Local\Temp\cchEWthY.ltrans0.ltrans.o: In function `NikonRemote::pulse_on(unsigned long) [clone .constprop.3]':

cchEWthY.ltrans0.o:(.text+0x54): undefined reference to `micros'

cchEWthY.ltrans0.o:(.text+0x62): undefined reference to `micros'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board ATtiny13.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

MCUdude, 2017-07-13:
Read my last post about micros 👍

LeonRonin, 2017-07-13:
just read it....and still trying to figure out how to edit that core_setting.h file. is it somewhere in my arduino folder??
thanks for the super fast respond thou...

MCUdude, 2017-07-13:
I'm on the computer right now, that's the fast response 😉

I guess you installed MicroCore using the boards manager URL. If so, the core_settings.h file is located in the Arduino program files folder. I believe you'll find **MicroCore**  at `C:\Program files (x86)\Arduino\hardware`

LeonRonin, 2017-07-13:
dude it just worked!!!! you need to be a little more specific on that "manual installation" part of readme text. people like me don't have much experience in this.
thank you for helping me with my project...much appreciate it!!!

closed 2017-07-13 by LeonRonin

allenchak, 2019-06-04:
@LeonRonin, I am using MacOS, and here is the file I found, /Users/allen/Library/Arduino15/packages/MicroCore/hardware/avr/1.0.3/cores/microcore/