MCUdude/MicroCore #29: Interesting bug on PWM function once I disabled SAFEMODE.

csnol, 2017-07-31:
Hi,Hans.
Thanks first for your great job!
Because flash size of Attiny13 ,I'v disabled SAFEMODE and MILLIS on my project . So, a trouble come in. 
 ` void analogWrite(uint8_t pin, uint8_t val)
{
  // SAFEMODE prevents you from inserting a pin number out of range
  #ifdef SAFEMODE
    DDRB |= _BV(pin & 0x02); // Set the correct pin as output
  #endif
  if(val == 0) // Handle off condition
    digitalWrite(pin, LOW);
  else if(val == 255) // Handle on condition
    digitalWrite(pin, HIGH);
  else // Otherwise setup the appropriate timer compare 
  { 
    if(pin == 1)  `

and

csnol, 2017-08-01:
I changed my code from analogWrite(0, 255)  to analogWrite(0, 250) , and from analogWrite(0, 0)  to analogWrite(0, 5) .
It ran OK on disabled SAFEMOD. it saved about 140 bytes, from 780 to 640 bytes.

MCUdude, 2017-08-01:
OK, so you're saying that your code freezes when you use `analogWrite(X, 0)` or `analogWrite(X, 255)` when SAFEMODE is disabled, right?

csnol, 2017-08-01:
Yes,  for save codes size, I don't wanna use turnOffPWM(pin) and reopen on them.

MCUdude, 2017-08-01:
Thanks for the info! With your input I was able to track down the bug. The ATTiny freezes because [the PWM output isn't disabled before running the digitalRead function](https://github.com/MCUdude/MicroCore/blob/master/avr/cores/microcore/wiring_pwm.c#L19-L23).

Try replacing line 19-21 with the code snippet below, and see if that helps:
```c++
if(val == 0) // Handle off condition
{
  turnOffPWM(pin);  
  digitalWrite(pin, LOW);
}
else if(val == 255) // Handle on condition
{
  turnOffPWM(pin);
  digitalWrite(pin, HIGH);
}
```

Anyways if you're short on space I recommend enabling LTO, which optimized the code even futher.
(ps: Let's keep the issue open until everything is resolved 😉)

csnol, 2017-08-01:
Sorry, 
These codes have been included in your "Wiring_digital.c" .
I don't think it useful.
>void digitalWrite(uint8_t pin, uint8_t val)
{
  // SAFEMODE prevents you from inserting a pin number out of range, and disables PWM if turned on
  #if defined(SAFEMODE)    //  #ifdef SAFEMODE
    if(pin > 5)
      return;
    if(pin < 2)
      turnOffPWM(pin); //If it's a PWM pin, make sure the PWM is off
  #endif

MCUdude, 2017-08-01:
No, I don't want the digitalWrite function to allways check of PWM is turned off when SAFEMODE is disabled. It's causing some overhead, and this is what I want to prevent as much as possible.

By placing the turnOffPWM function inside analogWrite, we reduce the number of times tyrnOffPWM is ran to a minimum.

I'm also working on a template based version version of **map**, which saves ablut 250b of memory.

Add the function below to your code, and replace *map* with *tmap*. I'll do some testing, and merge it into the main repo soon.

```c++
template <typename T, typename A, typename B, typename C, typename D> const T tmap(const T x, const A in_min, const B in_max, const C out_min, const D out_max)
{
  return (x - in_min) * (out_max - out_min) / (in_max - in_min) + out_min;
}
```

csnol, 2017-08-01:
Sorry, I missed the 'n' word in your codes '#ifndef' 
``` C++
  #ifndef SAFEMODE
    turnOffPWM(pin);  
  #endif 
```
thanks, I'll try again. and tmap(...)

csnol, 2017-08-01:
[image]
It's only 334 bytes !  without map,  just analogRead(1) divided by 4
``` C++
void setup() {
  pinMode(0, OUTPUT);    
  pinMode(1, OUTPUT);   
  pinMode(4, INPUT_PULLUP);
  analogWrite(0, 0);   
  analogWrite(1, 0);
}
void loop() {
  if (digitalRead(4)) {               
    analogWrite(0, analogRead(1)/4);
    analogWrite(1, 255 - analogRead(1)/4);
    delay(10);                      // wait for motor steady run.
  }
  else {
    analogWrite(0, 255);                  // Switch Motor to Run/Stop Mode.
    analogWrite(1, 0);
    delay(250);
    analogWrite(0, 0);
    analogWrite(1, 255);
    delay(250);
  }
}
```
It is OK! I've insert below codes to my libraries. Thanks Hans.
``` C++
if(val == 0) // Handle off condition
{
  #ifndef SAFEMODE
    turnOffPWM(pin);  
  #endif 
  digitalWrite(pin, LOW);
}
else if(val == 255) // Handle on condition
{
  #ifndef SAFEMODE
    turnOffPWM(pin);  
  #endif 
  digitalWrite(pin, HIGH);
}
```

MCUdude, 2017-08-03:
Again, thanks for reporting. The changes are now pushed to the master branch

closed 2017-08-03 by MCUdude