MCUdude/MicroCore #31: where can i find the core settings file?

consolacion, 2017-09-09:
I installed microcore via the boardsmanager but for the life of me i cant find the installed files, particularly the coresettings file.

The obvioisplace to go would be /{sketchbookfolder}/hardware  but i only find my attiny254585 cores there. Where is the Microcore?

per1234, 2017-09-09:
Boards Manager installs hardware packages to a different location. You can find the location by clicking the link on the line following **File > Preferences > More preferences can be edited directly in the file** and then navigating to the packages/MicroCore subfolder.

I'm not familiar with this "coresettings" file you speak of.

consolacion, 2017-09-09:
thanks for yr swift reply, in mean time i already found it in \AppData\Local\Arduino15\packages\MicroCore\hardware\avr\1.0.2\cores\microcore
the "core_settings.h" file is where several functions can be switched on or off

closed 2017-09-09 by consolacion