MCUdude/MicroCore #32: avrdude: Can't find programmer id "{upload.protocol}"

knobik, 2017-09-17:
Im using exacly this settings

[image]

on Arduino IDE 1.8.2

and in the end while trying to program the attiny13 im getting this error message.

avrdude: Can't find programmer id "{upload.protocol}"

MCUdude, 2017-09-17:
Are you using the latest version of MicroCore; v1.0.2 or a fresh manual install?

knobik, 2017-09-17:
Lastest version 1.0.2 installed from package manager

MCUdude, 2017-09-17:
It's an issue with the IDE. You can read more about it [here](https://github.com/MCUdude/MicroCore/issues/3). 
What you can do is choose **Arduino as ISP** instead. If you need a slower programmer you can edit the Arduino as ISP sketch directly.

MCUdude, 2017-09-21:
I'm closing this issue as we allready have issue #22 and issue #3 open

closed 2017-09-21 by MCUdude