MCUdude/MicroCore #33: bug in micros()

Robaquez, 2017-09-21:
I am not exactly sure what is happening but there is an issue with the way micros() are calculated. 

I am measuring width of PWM pulses sent by RC receiver. They are in the 1000-2000 us range, separated by about 16 ms. I don't need too high accuracy, and for most of the time lengths reported are good enough for my needs (requires some initial tinkering with OSCCAL, but it doesn't matter for the problem at hand). However, sometimes results of the measurements are longer than expected, and sometimes they are negative.

Here are examples of the numbers reported by my program running on the ATtiny13 with the MicroCore ("start" is a number reported by the first call to micros(), on the rising edge of the measure pulse, "end" is a number reported by the second call, on the falling edge, "pulse" is the pulse length calculated as a difference end-start):

a correct one:
start  521418632  end  521420214  pulse  1582

too long one (happened 5 times in 24k samples):
start  521436160  end  521439534  pulse  3374

negative one (happened 50 times in 24k samples):
start  526798027  end  526797824  pulse  -203

You will find that in each case when the result is wrong one of the younger bytes of either end or start equals 0 (and I can tell you it is no accident, all wrong results share this). I feel like it is somehow related to the moment the micros() function is called. Looks like sometimes it happens that the TCNT0 equals zero but the timer0_overlflow is not yet incremented. That means the number returned is short by 7*256=1792 us.

MCUdude, 2017-09-21:
Hi! Thanks for using MicroCore, and thanks for reporting the issue.
Are you using the latest version of MicroCore (v1.0.2 / fresh manual install)?
I wrote the micros() routine by applying a 9.6 MHz clock signal from my digital signal generator, and I thought I made is as good as possible with an uneven frequency like this. However, the issues you're seeing is clearly some kind of bug.

I don't have much time for support these days (hope I get more spare time next year), so I'm not able to help you out right now with real hardware. I'll of course reply to your messages 🙂 

Are you able to find out what the TCNT0 value actually is when the readings are wrong? Are you using the pulseIn functionfor calculating the pulse length?

Robaquez, 2017-09-21:
1.0.2 it is, downloaded and installed last week, works nicely and I am quite happy with it, thanks for your work on the project and sharing!

I can't use pulseIn, as the signal I am measuring has voltage that is sometimes too low and doesn't read as HIGH (nominally 3.3V vs 0.6*VCC, where VCC=5V, but strange things happen under a load). Hence my measurements are based on two analogRead()s:

  while (analogRead(PWM_ANALOG_LEG) < 300);
  unsigned long start = micros();
  while (analogRead(PWM_ANALOG_LEG) >= 300);
  unsigned long pulse = micros() - start;

I have not tested actual values of TCNT0, but I calculated them from the values returned by micros() function (after all it is just a matter of diving by 7 and isolating 8 lowest bits). For all measurement results outside of the expected range one of the TCNT0 values was always consistently 0. I can try to modify the micros() function to separately save the TCNT0 value, but I don't see how it could be different from the value extracted from the result returned by micros().

See if you can import these into a spreadsheet, timer_overflow/TCNT0 are calculated from the values returned by micros():

start	timer_overflow	TCNT0	end	timer_overflow	TCNT0	pulse length

415365545	231788	207	415367134	231789	178	1589
415384942	231799	162	415386531	231800	133	1589
415404262	231810	106	415405844	231811	76	1582
415423624	231821	56	415425213	231822	27	1589
415442958	231832	2	415444533	231832	227	1575
415462306	231842	206	415463895	231843	177	1589
415481619	231853	149	415483208	231854	120	1589
415500981	231864	99	415502570	231865	70	1589
415520315	231875	45	415521904	231876	16	1589
415537920	231885	0	415541308	231886	228	3388	<-
415559018	231896	198	415560607	231897	169	1589
415578408	231907	152	415579997	231908	123	1589
415597728	231918	96	415599310	231919	66	1582
415617090	231929	46	415618679	231930	17	1589
415636396	231939	244	415637978	231940	214	1582
415655758	231950	194	415657347	231951	165	1589
415675078	231961	138	415676653	231962	107	1575
415694440	231972	88	415696043	231973	61	1603
415713788	231983	36	415715384	231984	8	1596

418344787	233451	85	418346376	233452	56	1589
418364170	233462	38	418365766	233463	10	1596
418383497	233472	239	418385079	233473	209	1582
418402873	233483	191	418404462	233484	162	1589
418422193	233494	135	418423782	233495	106	1589
418441562	233505	86	418443158	233506	58	1596
418460882	233516	30	418460672	233516	0	-210	<-
418480230	233526	234	418481819	233527	205	1589
418499529	233537	175	418501111	233538	145	1582
418518898	233548	126	418520487	233549	97	1589
418538204	233559	68	418539786	233560	38	1582
418557552	233570	16	418559141	233570	243	1589
418576858	233580	214	418578440	233581	184	1582
418596206	233591	162	418597795	233592	133	1589

MCUdude, 2017-09-21:
I managed to create a markdown table of some of the relevant data. It's definitly related to TCNT0 beeing zero at the start or the end. BTW how did you print this? did you ran some kind of software serial library on the ATtiny13?

| micros() start | Timer0 overflow at start | TCNT0 at start | micros() end | Timer0 overflow end | TCNT0 end | Pulse length [us] |
|----------------|--------------------------|----------------|--------------|---------------------|-----------|-------------------|
| 415365545      | 231788                   | 207            | 415367134    | 231789              | 178       | 1589              |
| 415384942      | 231799                   | 162            | 415386531    | 231800              | 133       | 1589              |
| 415404262      | 231810                   | 106            | 415405844    | 231811              | 76        | 1582              |
| 415423624      | 231821                   | 56             | 415425213    | 231822              | 27        | 1589              |
| 415442958      | 231832                   | 2              | 415444533    | 231832              | 227       | 1575              |
| 415462306      | 231842                   | 206            | 415463895    | 231843              | 177       | 1589              |
| 415481619      | 231853                   | 149            | 415483208    | 231854              | 120       | 1589              |
| 415500981      | 231864                   | 99             | 415502570    | 231865              | 70        | 1589              |
| 415520315      | 231875                   | 45             | 415521904    | 231876              | 16        | 1589              |
| 415537920      | 231885                   | **0**          | 415541308    | 231886              | 228       | **3388**          |
| 415559018      | 231896                   | 198            | 415560607    | 231897              | 169       | 1589              |
| ..             | ..                       | ..             | ..           | ..                  | ..        | ..                |
| 418422193      | 233494                   | 135            | 418423782    | 233495              | 106       | 1589              |
| 418441562      | 233505                   | 86             | 418443158    | 233506              | 58        | 1596              |
| 418460882      | 233516                   | 30             | 418460672    | 233516              | **0**     | **-210**          |
| 418480230      | 233526                   | 234            | 418481819    | 233527              | 205       | 1589              |

Robaquez, 2017-09-21:
I wrote a simple communication system that uses two wires to send data from ATtiny13 to Arduino, and Arduino sends it (using just Serial) to my PC. A bit of scratching your left ear with a right hand - but it works :)

SpenceKonde, 2017-09-21:
Suspect the interrupt is firing while micros() is calculating the value. Is the micros() implementation hardened against that?

Phone sent hence terse

MCUdude, 2017-09-21:
> I wrote a simple communication system that uses two wires to send data from ATtiny13 to Arduino, and Arduino sends it (using just Serial) to my PC. A bit of scratching your left ear with a right hand - but it works :)

Awesome! You should definitely add it to Github. I'm sure a lot of MicroCore users would like to test it 🙂 

> Suspect the interrupt is firing while micros() is calculating the value. Is the micros() implementation hardened against that?

That was my first guess too. here's the micros() function for a 9.6MHz clock:

``` c
// This will cause an interrupt every 256/64 clock cycle
ISR(TIM0_OVF_vect)
{
  timer0_overflow++; // Increment counter by one
}

uint32_t micros()
{
  // Preserve old SREG vale
  uint8_t oldSREG = SREG;
  // Disable global interrupts
  cli();
  // Variable to hold the return value
  uint32_t x;
  
    // Each timer tick is 1/(9.6MHz/64) = 6.666us long. We multiply the timer0_overflow variable
    // by 256 (bitshift 8 times) and we add the current timer count TCNT0. Since each tick is 6.666us long,
    // we multiply by 7 at the end
    x = ((timer0_overflow << 8) + TCNT0) * 7;
  
  // Restore SREG
  SREG = oldSREG;
  return x;
}
```

Robaquez, 2017-09-21:
Just checked the problem can be easily reproduced by

``` c++
#include "util/delay_basic.h"

unsigned long m1, m2;

void setup() {
  pinMode(3, OUTPUT);
  m1 = 0;
}

void loop() {
  _delay_loop_2(1000);
  m2 = micros();
  if (m2 < m1) digitalWrite(3,HIGH);
  m1 = m2;
}
```

LED on 3 should never light up. It does, quite fast.

MCUdude, 2017-09-21:
Great, a bare minimum program to reproduce the problem 👍 

Let's try something ugly:
``` c
uint32_t micros()
{
  // Preserve old SREG vale
  uint8_t oldSREG = SREG;
  // Disable global interrupts
  cli();

  if(TCNT0 == 0)
    TCNT0 = 1;

  // Variable to hold the return value
  uint32_t x;
  
    // Each timer tick is 1/(9.6MHz/64) = 6.666us long. We multiply the timer0_overflow variable
    // by 256 (bitshift 8 times) and we add the current timer count TCNT0. Since each tick is 6.666us long,
    // we multiply by 7 at the end
    x = ((timer0_overflow << 8) + TCNT0) * 7;
  
  // Restore SREG
  SREG = oldSREG;
  return x;
}
```

SpenceKonde, 2017-09-21:
In the above posted micros() implementation, imagine that the overflow occurs while micros() is running. timer0_overflow would not have been incremented, but tcnt0 would be back at 0.

MCUdude, 2017-09-21:
> In the above posted micros() implementation, imagine that the overflow occurs while micros() is running. timer0_overflow would not have been incremented, but tcnt0 would be back at 0.

I see. So how did you deal with this with ATtinyCore?

MCUdude, 2017-09-21:
I've done som testing, and found this implementation to be the best one yet. It misses once in a while, but is certainly much better. It is even possible to get a bullet proof implementation of micros without actually stopping the timer?

``` c
uint32_t micros()
{
  uint32_t x;
  uint8_t t;  
  
  uint8_t oldSREG = SREG; // Preserve old SREG value 
  cli();                  // Disable global interrupts
  t = TCNT0;              // Store timer0 counter value
  x = timer0_overflow;    // Store timer0 overflow count
  SREG = oldSREG;         // Restore SREG
  
  #if F_CPU == 20000000L
    // Each timer tick is 1/(16MHz/64) = 3.2us long. We multiply the timer0_overflow variable
    // by 256 (bitshift 8 times) and we add the current timer count TCNT0. Since each tick is 3.2us long,
    // we multiply by 3 at the end
    return ((x << 8) + t) * 3;
  #elif F_CPU == 16000000L
    // Each timer tick is 1/(16MHz/64) = 4us long. We multiply the timer0_overflow variable
    // by 256 (bitshift 8 times) and we add the current timer count TCNT0. Since each tick is 4us long,
    // we multiply by 4 at the end
    return ((x << 8) + t) * 4;
  #elif F_CPU == 1200000L
    // Each timer tick is 1/(12MHz/64) = 5.333us long. We multiply the timer0_overflow variable
    // by 256 (bitshift 8 times) and we add the current timer count TCNT0. Since each tick is 5.333us long,
    // we multiply by 5 at the end
    return ((x << 8) + t) * 5;  
  #elif F_CPU == 9600000L
    // Each timer tick is 1/(9.6MHz/64) = 6.666us long. We multiply the timer0_overflow variable
    // by 256 (bitshift 8 times) and we add the current timer count TCNT0. Since each tick is 6.666us long,
    // we multiply by 7 at the end
    return ((x << 8) + t) * 7;
  #elif F_CPU == 80000000L
    // Each timer tick is 1/(8MHz/64) = 8us long. We multiply the timer0_overflow variable
    // by 256 (bitshift 8 times) and we add the current timer count TCNT0. Since each tick is 8us long,
    // we multiply by 8 at the end
    return ((x << 8) + t) * 8;
  #elif F_CPU == 4800000L
    // Each timer tick is 1/(4.8MHz/64) = 13.333us long. We multiply the timer0_overflow variable
    // by 256 (bitshift 8 times) and we add the current timer count TCNT0. Since each tick is 13.333us long,
    // we multiply by 13 at the end
    return ((x << 8) + t) * 13;
  #elif F_CPU == 1200000L
    // Each timer tick is 1/(1.2MHz/8) = 6.666us long. We multiply the timer0_overflow variable
    // by 256 (bitshift 8 times) and we add the current timer count TCNT0. Since each tick is 6.666us long,
    // we multiply by 7 at the end
    return ((x << 8) + t) * 7;
  #elif F_CPU == 1000000L
    // Each timer tick is 1/(1MHz/8) = 8us long. We multiply the timer0_overflow variable
    // by 256 (bitshift 8 times) and we add the current timer count TCNT0. Since each tick is 8us long,
    // we multiply by 8 at the end
    return ((x << 8) + t) * 8;  
  #elif F_CPU == 600000L
    // Each timer tick is 1/(600kHz/8) = 13.333us long. We multiply the timer0_overflow variable
    // by 256 (bitshift 8 times) and we add the current timer count TCNT0. Since each tick is 13.333us long,
    // we multiply by 13 at the end
    return ((x << 8) + t) * 13;    
 #endif

}
#endif // ENABLE_MICROS
```

Robaquez, 2017-09-21:
I think it may be still missing now and then (very rarely though), but I was doing several things at once and I am not entirely sure. I am going to leave it for a night. My code doesn't change the pin back to LOW for a purpose - this way the information about the bug is stored in the LED ON state forever.

Will check the original program tomorrow.

Robaquez, 2017-09-21:
Didn't take long - the LED is on :(

MCUdude, 2017-09-22:
You used the micros() code from my previous post, right?

``` c
#include "util/delay_basic.h"

unsigned long m1, m2;

void setup() {
  pinMode(2, OUTPUT);
  m1 = 0;
}

void loop() 
{
  //_delay_loop_2(1000);
  m2 = micros();
  if(m2 < m1)
    digitalWrite(2, HIGH);

  m1 = m2;
}
```

I left the code above running when I ate breakfast this morning. The on-board LED (connected to PB2) did not light up within the 30 minute before I had to leave for work.

I'm going to test this with the official Arduino repo as well (with an Arduino UNO). Not sure if this is a MicroCore spesific bug or not.

Robaquez, 2017-09-22:
To make things easier I am using this:

```c++
#include "util/delay_basic.h"

unsigned long m1, m2;

extern volatile unsigned int timer0_overflow;

uint32_t _micros()
{
  uint32_t x;
  uint8_t t;  
  
  uint8_t oldSREG = SREG; // Preserve old SREG value 
  t = TCNT0;              // Store timer0 counter value
  cli();                  // Disable global interrupts
  x = timer0_overflow;    // Store timer0 overflow count
  SREG = oldSREG;         // Restore SREG
  
    // Each timer tick is 1/(9.6MHz/64) = 6.666us long. We multiply the timer0_overflow variable
    // by 256 (bitshift 8 times) and we add the current timer count TCNT0. Since each tick is 6.666us long,
    // we multiply by 7 at the end
    return ((x << 8) + t) * 7;
}

void setup() {
  pinMode(3, OUTPUT);
  delay(10);
  m1 = 0;
}

void loop() {
  _delay_loop_2(1000);
  m2 = _micros();
  if (m2 < m1) digitalWrite(3,HIGH);
  m1 = m2;
}
```

MCUdude, 2017-09-22:
.. And you're able to make the LED to turn on?

Robaquez, 2017-09-22:
Yes, exactly this code turns the LED on. Takes minutes, up to tens of minutes.

Actually I am not sure if the above code is entirely OK, it is compiled for 9.6 MHz, but doesn't 9.6 MHz require changing CLKPR in the setup()? At least that is what I did in my original code, with

```c++
  CLKPR = 0x80;
  CLKPR = 0;   // presc 1  
```
(that, plus tinkering with OSCCAL allowed me to get the correct values from the micros() function).

Not that it matters for the problem.

MCUdude, 2017-09-22:
It's not necessary to change the CLKPR bits in the setup, the prescaler is 1 by default. You can try it yourself:

```c
void setup()
{
  //CLKPR = 0x80;
  //CLKPR = 0;   // presc 1  
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() 
{
  digitalWrite(3, HIGH); 
  delay(1000);
  digitalWrite(3, LOW); 
  delay(1000);
}
```

The LED should toggle every second even if you haven't touched the CLKPR setting.

Robaquez, 2017-09-22:
Did some more tests and found something strange: with the original MicroCore code it is much more likely to observe the problem when the main loop is using digitalRead() than when using analogRead().

New version is definitely much more reliable. Will leave it overnight test to see how reliable.

As to CLKPR: did some reading and testing, perhaps we are talking about differently configured chips. Factory default is that without modifying CLKPR chip doesn't run at 9.6 MHz, but at 1.2 MHz (8 times slower). Manual says:

> The CKDIV8 Fuse determines the initial value of the CLKPS bits.  (...) If CKDIV8 is programmed, CLKPS bits are reset to “0011”, giving a division factor of eight at start up. (...) The device is shipped with the CKDIV8 Fuse programmed.

So it is quite easy to reprogram the chip so that there is no need to touch the CLKPR (perhaps that's the case with the one you are using), but factory default is that is is necessary (and my ATtinys are in the factory configuration).

MCUdude, 2017-09-23:
> As to CLKPR: did some reading and testing, perhaps we are talking about differently configured chips. Factory default is that without modifying CLKPR chip doesn't run at 9.6 MHz, but at 1.2 MHz (8 times slower).

Yup, tiny13's ship with the CKDIV8 fuse sat. In order to change this you'll have to select 9.6 MHz in the Tools menu and hit Burn bootloader to set the correct fuses.

Robaquez, 2017-09-23:
OK, about 15 hours passed and the LED is still off, so it looks like the new micro() function works quite good with my old code.

In the past I have burned fusebits on ATmega328 from the command line using avrdude, so the idea is not completely alien to me ;) But I see changing the fusebit can save several bytes of code, may be important some day.

MCUdude, 2017-09-28:
Great that we figured this out! I'm working on some other improvements as well, but will do a boards manager release when all the loose ends are tied up 🙂

closed 2017-09-28 by MCUdude