MCUdude/MicroCore #34: Attiny 13 library in my project

Galeje, 2017-10-03:
Hi!
I am working on my own project which is called Cing (https://github.com/Galeje/Cing) and i wanted to use your library but i want to change one thing and that is name in arduino board bar(picture).
image.I know how to do it but i need your permison to do that.Can i do that?It will be so user of cing can better orient in boards because cing is a robot for beginers.

Thanks for respond.
Best regards Stanley member of GalejeNextGen team.

MCUdude, 2017-10-04:
Hi!
Of course you can! MicroCore is open source, so you're free to do whatever you want with it. Still I'd appreciate a little note in the readme, saying that the project builds upon MicroCore 😉

Galeje, 2017-10-08:
Thanks a lot. And of corse i will mention it in Readme

Stanley team member of Galeje

closed 2017-10-08 by Galeje