MCUdude/MicroCore #35: attachInterrupt() example needed

KochkinBochkin, 2017-10-13:
Hello! Please add the attachInterrupt(interrupt, function, mode) example. I tried many options, but I did not succeed with external interrupts with Attiny13. Thank you.

MCUdude, 2017-10-14:
AttachInterrupt works exactly the same as it does on any other Arduino. The code below will toggle an LED connected to pin 2 if the interrupt pin (pin 1) is connected to ground.

``` c
const uint8_t ledPin = 2;
const uint8_t interruptPin = 1;
volatile uint8_t state = LOW;

void setup() {
  pinMode(ledPin, OUTPUT);
  pinMode(interruptPin, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(interruptPin), blink, CHANGE);
}

void loop() {
  digitalWrite(ledPin, state);
}

void blink() {
  state = !state;
}
```

closed 2017-10-14 by MCUdude

KochkinBochkin, 2017-10-14:
Thank you very much!