MCUdude/MicroCore #37: Support for digitalPinToBitMask

thomaskilian, 2017-10-19:
I'm using a lib which utilizes `digitalPinToBitMask`. I once programmed an Attiny13 that used a lib (and it's working still) using this operation. Now I wanted to update it and tried compiling with this MicroCore. But obviously the operation is not implemented here. Is there any way to get this to work? Or am I forced to google/voodoo whatever I did once I got my code working?

MCUdude, 2017-10-19:
Sorry, I should probably implement it to keep the compability with other Arduino sketches as good as possible. Try adding:

```c
#define digitalPinToBitMask(p)  (((p) >= 0 && (p) <= 5) ? _BV(p) : -1)
```

thomaskilian, 2017-10-19:
`digitalPinToPort` would be the next one. (I'm out for some hours now...)

MCUdude, 2017-10-19:
Most of these macros are [All ready implemented](https://github.com/MCUdude/MicroCore/blob/d58d051039a1eb11b1bf6b842f0d2cdb8f0728fa/avr/cores/microcore/pins_arduino.h#L32). You just need to do a manual installation of MicroCore, because I haven't doe a release of it yet.

thomaskilian, 2017-10-19:
I manually included that header and it throws a couple of `error: redefinition of 'const uint8_t MOSI'
 static const uint8_t MOSI = 0;` errors. Probably I can ignore them?

thomaskilian, 2017-10-19:
Hmm. Commented the above and now it croaks `'digitalPinToBitMask' was not declared in this scope`

thomaskilian, 2017-10-19:
Ok. I included the `digitalPinToBitMask` and the `pins_arduino.h` manually, commented the duplicate definitions of `MISO` etc. and now it croaks `undefined reference to `micros'`

MCUdude, 2017-10-19:
You'll have to enable Micros() if your code depends on it. Read the **core settings** section in the README file for more info

thomaskilian, 2017-10-19:
That did it. Now I just have to figure out why my code uses up 400% of the available memory although I did not change much :-/

MCUdude, 2017-10-19:
Make sure LTO is enabled. That helps a lot on the compiled code size. I'll close this issue as soon as I've implemented the digitalPinToBitMask macro

thomaskilian, 2017-10-19:
Yeah, I'll look into that. But I guess my main issue is that I used a 45 and not a 13 - LOL. Thanks for the support and I'll find another application for the 13 soon :-) 
👍

closed 2017-10-21 by MCUdude