MCUdude/MicroCore #39: (.text+0x0): multiple definition of `__vector_8'

piotrkochan, 2017-12-03:
Hi I try to program Atmega13a to enable deep sleep mode every few seconds like in the article:
http://homecircuits.eu/blog/low-power-picopower-attiny13a/

So the code is:

```
#include <avr/interrupt.h>
#include <avr/sleep.h>
// Pin 4 for LED
int led = 4;

ISR(WDT_vect) {
  digitalWrite(led, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);               // wait for a second
  digitalWrite(led, LOW);    // turn the LED off by making the voltage LOW
  sleep_mode();
}

void setup() {
  // initialize the digital pin as an output.
  pinMode(led, OUTPUT);
  // prescale timer to 8s so we can measure current
  WDTCR |= (1<<WDP3 )|(0<<WDP2 )|(0<<WDP1)|(1<<WDP0); // 8s
  // Enable watchdog timer interrupts
  WDTCR |= (1<<WDTIE);
  sei(); // Enable global interrupts
  // Use the Power Down sleep mode
  set_sleep_mode(SLEEP_MODE_PWR_DOWN);
  for (;;) {
    sleep_mode();   // go to sleep and wait for interrupt...
  }
}
// the loop routine runs over and over again forever:
void loop() {
}
```

And compile error is:

```
....
Compiling libraries...
Compiling core...
Using precompiled core
Linking everything together...
/home/.../arduino-1.8.5/hardware/tools/avr/bin/avr-gcc" -w -Os -Wl,--gc-sections -mmcu=attiny13a -w -flto -o "/tmp/arduino_build_638757/Blink.ino.elf" "/tmp/arduino_build_638757/sketch/Blink.ino.cpp.o" "/tmp/arduino_build_638757/../arduino_cache_789397/core/core_MicroCore_avr_attiny13_clock_128k,BOD_2v7,flag_Os_flto_ceceed073e2d2fa20c28f2cdd89dc065.a" "-L/tmp/arduino_build_638757" -lm
wiring.c.o (symbol from plugin): In function `__vector_8':
(.text+0x0): multiple definition of `__vector_8'
/tmp/arduino_build_638757/sketch/Blink.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ATtiny13.
```

MCUdude, 2017-12-04:
Yep, that's correct. The WDT interrupt is used to [calculate millis()](https://github.com/MCUdude/MicroCore/blob/3282a2d1a49205f8a97e6d095a43f56d5fe29f4f/avr/cores/microcore/wiring.c#L27). You need to disable millis in order to use the WDT interrupt. You can du that my modifying the core_settings.h file

piotrkochan, 2017-12-04:
Ok Hans thank you, ill try it

piotrkochan, 2017-12-11:
It works, thanks

MCUdude, 2017-12-11:
Great!

closed 2017-12-11 by MCUdude