MCUdude/MicroCore #42: Add functions for pin change interrupt support

baruch, 2017-12-18:
I want to use the pin change interrupt rather than the INT0 single pin interrupt. Would it be acceptable to you if I added such new functions, possibly as an option like pwm/adc?

MCUdude, 2017-12-18:
I try to keep the core as "vanilla" as possible. The original Arduino core does not support PCINTs, so I'm not going to add it to the core. A separate PCINT library is fine though.

You probably know you can still use PCINTs even though they aren't implemented in the core.

baruch, 2017-12-18:
I know, I'm doing just that now, it's just that as opposed to the other features I need to actually read the documentation :-)

MCUdude, 2017-12-19:
I you have room for some extra code you may use the [PinChangeInterrupt library](https://github.com/NicoHood/PinChangeInterrupt). It works great with the ATtiny13!