MCUdude/MicroCore #44: Error while burning bootloader.Arduino.h: No such file or directory_compilation terminated.

kimaj, 2018-01-22:
Error while burning bootloader.
Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x000000
avrdude: Yikes!  Invalid device signature.
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

sketch\Blink_for_attiny13a.ino.cpp:1:21: fatal error: Arduino.h: No such file or directory

 #include <Arduino.h
 ^

compilation terminated.

exit status 1
Error compiling for board ATtiny13.

ide version 1.8.4
i renamed the microcore as:
F:\arduino-1.8.4 IDE\hardware\ATtiny13\avr\cores\ATtiny13

MCUdude, 2018-01-22:
First, please post your complete Arduino sketch. I'm pretty sure that's where the issue is.

You can either do a boards manager install (recommended), or a manual install. If you're doing a manual install you need to put the MicroCore folder into the right directory.

MCUdude, 2018-01-25:
Nothing?

closed 2018-02-23 by MCUdude

ASSOUMA97, 2018-04-11:
hi i have a problem compiling the attiny 13 with arduino uno i have tried several problem but there is still this error message that appears i need your help ... it's urgent! here's the program:
#include "Arduino.h"
#include <Arduino.h>

void setup() {
 
 pinMode(4, OUTPUT);
}
void loop() {
 digitalWrite(4, HIGH);   
 delay(1000);             
 digitalWrite(4, LOW);    
 delay(1000);             
} 
[image]
[image]

MCUdude, 2018-04-11:
Remove `#include "Arduino.h"` and `#include <Arduino.h>`

Arduino.h is included automatically

ASSOUMA97, 2018-04-13:
I raised it .. there is still the same mistake

ASSOUMA97, 2018-04-13:
I really need a solution because it is a project of the school! it's more
than a month that I'm trying to solve this problem .. there are no
resources that can help me

2018-04-13 10:56 GMT+01:00 Asmae Naim <asmaenaim97@gmail.com>:

per1234, 2018-04-13:
@ASSOUMA97 did you rename your core folder like @kimaj did? If so, then that error would be the expected result.

You need to leave the core folder named `microcore`. I can't imagine what would make someone think they should rename that folder.

Please do this:

1. File > Preferences > Show verbose output during: > compilation (check) > OK
2. Sketch > Verify/compile
3. After the compilation fails, you'll see a button on the right side of the orange bar "Copy error messages". Click that button.
4. Paste it to a reply here, using code fencing so that it will be correctly formatted (https://guides.github.com/features/mastering-markdown/).

ASSOUMA97, 2018-04-13:
I renamed nothing! I wrote the code after installing attiny13 then I compiled and the result of this compilation is this error message that I sent you

MCUdude, 2018-04-13:
Please follow per1234's instruction and paste your verbose output.

ASSOUMA97, 2018-04-13:
Arduino : 1.8.5 (Windows 8), Carte : "ATtiny13, 1.2 MHz internal osc., 4.3v, Enabled (default)"
------------------------------------------------------------------------------------------------------------

C:\Program Files\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\Arduino\hardware -hardware C:\Users\hp  hp\AppData\Local\Arduino15\packages -tools
 
C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -tools 

C:\Users\hp  hp\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\hp  hp\Documents\Arduino\libraries -fqbn=MicroCore:avr:attiny13:clock=1M2,BOD=4v3,lto=Os_flto -ide-version=10805 -build-path

 C:\Users\HPHP~1\AppData\Local\Temp\arduino_build_700928 -warnings=none -build-cache 

C:\Users\HPHP~1\AppData\Local\Temp\arduino_cache_33144 -prefs=build.warn_data_percentage=75 -verbose C:\Users\hp  hp\Desktop\test\test.ino

C:\Program Files\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files\Arduino\hardware -hardware C:\Users\hp  hp\AppData\Local\Arduino15\packages -tools 

C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -tools 

C:\Users\hp  hp\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\hp  hp\Documents\Arduino\libraries -fqbn=MicroCore:avr:attiny13:clock=1M2,BOD=4v3,lto=Os_flto -ide-version=10805 -build-path 

C:\Users\HPHP~1\AppData\Local\Temp\arduino_build_700928 -warnings=none -build-cache 

C:\Users\HPHP~1\AppData\Local\Temp\arduino_cache_33144 -prefs=build.warn_data_percentage=75 -verbose C:\Users\hp  hp\Desktop\test\test.ino
Using board 'attiny13' from platform in folder: C:\Users\hp  hp\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\1.0.3
Using core 'microcore' from platform in folder: C:\Users\hp  hp\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\1.0.3
Detecting libraries used...

"C:\Program Files\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=attiny13a -DF_CPU=1200000L -DARDUINO=10805 -DARDUINO_attiny -DARDUINO_ARCH_AVR -Wextra -flto  "-IC:\Users\hp  hp\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\1.0.3\cores\microcore" 

"C:\Users\HPHP~1\AppData\Local\Temp\arduino_build_700928\sketch\test.ino.cpp" -o "nul" -D COMPILER_LTO

"C:\Program Files\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=attiny13a -DF_CPU=1200000L -DARDUINO=10805 -DARDUINO_attiny -DARDUINO_ARCH_AVR -Wextra -flto  "-IC:\Users\hp  hp\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\1.0.3\cores\microcore" 

"C:\Users\HPHP~1\AppData\Local\Temp\arduino_build_700928\sketch\test.ino.cpp" -o 

"C:\Users\HPHP~1\AppData\Local\Temp\arduino_build_700928\preproc\ctags_target_for_gcc_minus_e.cpp" -D COMPILER_LTO

########################HERE IS THE MESSAGE THAT'S WRITTEN IN RED################

C:\Users\HPHP~1\AppData\Local\Temp\arduino_build_700928\sketch\test.ino.cpp:1:21: fatal error: Arduino.h: No such file or directory

 #include <Arduino.h>

                     ^

compilation terminated.

exit status 1
Erreur de compilation pour la carte ATtiny13

ASSOUMA97, 2018-04-16:
yes this file exist !

ASSOUMA97, 2018-04-16:
please I need a solution