MCUdude/MicroCore #46: Travis CI for MicroCore

MCUdude, 2018-02-21:
I think it may be a good idea to add Travis CI to MicroCore. Would you like to help me out @per1234?
MicroCore and MajorCore are the only "Arduino cores" left without CI 🙂

per1234, 2018-02-22:
OK, will do. I just fixed a serious bug in the script and I want to add a high priority improvement (https://github.com/per1234/arduino-ci-script/issues/16), then a new release. So I'll do that before the MicroCore PR so the subtree will contain the latest version of the script.

I recently added some formatting style compliance [checks ](https://github.com/per1234/arduino-ci-script/blob/0a77a961ab8c58c998d1595a654f98f9de302b78/.travis.yml#L46-L58)to my Travis CI builds:
- No trailing whitespace.
- Unix line endings (LF).
- Newline at end of file.
- No tabs (I use spaces). This could be modified to check that all indentation is done with tabs instead of spaces.
- No blank lines at start of file.
- No blank lines at end of file.

Kind of uptight but it is nice to have consistency, especially when multiple people are working on a project. I didn't add that stuff to arduino-ci-script because it's not Arduino specific and they're all pretty simple commands anyway. So if any of that is something you would find useful I can add it to your .travis.yml files.

MCUdude, 2018-02-22:
Does this means Travis will fail if it finds any tab or CR characters? I agree that using spaces and Unix line endings is a good idea.

per1234, 2018-02-22:
I have mine set up to immediately fail the build if any of those tests don't pass. It's also possible to simply add a warning to to the log but that's easier to ignore.

MCUdude, 2018-02-22:
I believe there exist a lot of code with tabs and CRs. Would be awesome to get rid of them all!

closed 2018-02-25 by MCUdude