MCUdude/MicroCore #49: Burn Bootloader question

beicnet, 2018-03-06:
Hi there,

I'm trying to burn the Bootloader at default frequency 9.6 MHz over Arduino as ISP, but at the end I got an error saying:

`avrdude: ser_open(): can't open device "unknown": The system cannot find the file specified.`

The sketch uploading through "Upload Using Programming" just working fine.

Arduino IDE 1.8.2 with MicroCore 1.0.3

Regards

beicnet, 2018-03-06:
... and here is the whole output:

```
avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\beic\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\1.0.3/avrdude.conf"

         Using Port                    : COM22
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         Setting bit clk period        : 32.0
         AVR Part                      : ATtiny13
         Chip Erase delay              : 4000 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65     5     4    0 no         64    4      0  4000  4000 0xff 0xff
           flash         65     6    32    0 yes      1024   32     32  4500  4500 0xff 0xff
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          2    0      0     0     0 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9007 (probably t13a)
avrdude: erasing chip
avrdude: reading input file "0x3f"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3f:
avrdude: load data lock data from input file 0x3f:
avrdude: input file 0x3f contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 1 bytes of lock verified
avrdude: reading input file "0xfb"
avrdude: writing hfuse (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of hfuse written
avrdude: verifying hfuse memory against 0xfb:
avrdude: load data hfuse data from input file 0xfb:
avrdude: input file 0xfb contains 1 bytes
avrdude: reading on-chip hfuse data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 1 bytes of hfuse verified
avrdude: reading input file "0x3A"
avrdude: writing lfuse (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lfuse written
avrdude: verifying lfuse memory against 0x3A:
avrdude: load data lfuse data from input file 0x3A:
avrdude: input file 0x3A contains 1 bytes
avrdude: reading on-chip lfuse data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of lfuse verified

avrdude: ser_open(): can't open device "unknown": The system cannot find the file specified.

Error while burning bootloader.
avrdude done.  Thank you.
```

MCUdude, 2018-03-06:
Thanks. I need to see the command itself too. this is just the avrdude output. I think you can enable this in the IDE settings

beicnet, 2018-03-06:
Verbose?

MCUdude, 2018-03-06:
[image]

beicnet, 2018-03-06:
@MCUdude I edited the output above with command line while trying to Burn the Bootloader.

MCUdude, 2018-03-06:
I believe you've mixed it up a little.

```
avrdude: verifying ...
avrdude: 1 bytes of lock verified
avrdude: reading input file "0xfb"
avrdude: writing hfuse (1 bytes):

C:\Users\beic\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/bin/avrdude -CC:\Users\beic\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\1.0.3/avrdude.conf -v -pattiny13 -cstk500v1 -B32 -Ulock:w:0x3e:m 

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

Error while burning bootloader.
Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of hfuse written
avrdude: verifying hfuse memory against 0xfb:
avrdude: load data hfuse data from input file 0xfb:
avrdude: input file 0xfb contains 1 bytes
avrdude: reading on-chip hfuse data:
```

Anyways it seems like the second command is missing the port parameter -P. I'm on vacation right now, but I'll look closer into it on Friday.

beicnet, 2018-03-06:
Ok, but how did I mixed it? I did not change anyting!

MCUdude, 2018-03-06:
No idea.. Did you use the programmer "Arduino as ISP (slow)" or "Arduino as ISP"? Please try both

beicnet, 2018-03-06:
I always using "Arduino as ISP", but I tried "Arduino as ISP (slow)", got error there too!

MCUdude, 2018-03-06:
Thanks! I've found a bug in the platform.txt file that's causing the problem. Will fix it when I get home 😊

beicnet, 2018-03-06:
:laughing: So, can you give me as temporary what to edit?! Kappa

beicnet, 2018-03-06:
I need to burn a bunch of chips today!

MCUdude, 2018-03-06:
Add 

{program.extra_params}

To line 123 in platform.txt. You can insert it right before -B32

beicnet, 2018-03-06:
Like this?

`tools.avrdude.erase.pattern="{cmd.path}" "-C{config.path}" -v -p{build.mcu} -c{protocol} {program.extra_params} -B32 {program.extra_params} -e -Ulock:w:{bootloader.unlock_bits}:m -Uhfuse:w:{bootloader.high_fuses}:m -Ulfuse:w:{bootloader.low_fuses}:m`

beicnet, 2018-03-06:
Ohhh, I found it, in my case it was line 127:

`tools.avrdude.bootloader.pattern="{cmd.path}" "-C{config.path}" {bootloader.verbose} -p{build.mcu} -c{protocol} -B32 {program.extra_params} -Ulock:w:{bootloader.lock_bits}:m`

Added what you said, and it's working now! :wink: 

Thank you!

closed 2018-03-09 by MCUdude

beicnet, 2018-03-09:
@MCUdude Can you please release a new Boards Manager version too?

MCUdude, 2018-03-09:
I usually do a boards manager release when there are major changes. There's been very few minor commits since the last version, so I don't see why that's necessary. Nobody really recommend using Arduino as ISP either, since is often leads to connection issues and errors. I prefer to use the USBasp or USBtinyISP. They are cheap, they're dedicated for the job and they works _every_ time.

If you want the latest changes a manual install is the way to go

beicnet, 2018-03-09:
@MCUdude  Okay then, thanks for everithing! :wink: 

I don't know, but I never had any issues with Arduino as ISP using on large scale of MCU's (ATTiny13, ATTiny2313, ATTiny85, ATTiny48, ATmega48, ATmega8 and ATmega328) on regular basis in the past maybe 4-5 years now.

I'm using this ISP Shield and it's really amazing, also made a few extension DIP board for it.
[image]

commit 2636232a (2018-03-09): Add missing field for programmer
This prevents Arduino as ISP to work properlt. Closes #49