MCUdude/MicroCore #50: Support AtTiny25/45/85

lazlyhu, 2018-03-08:
Hi there! 👋 

First of all I want to say thank you for you work, really useful and easy to move my solution from prototype to production without rewrite the code (or use other for prototyping, not Arduino). For home projects it's more then useful. 🙇 

My question is about I took a fast trip in your code, and look like there is no serious problem around use your solution to `AtTiny25/45/85` models too. As you said in the Readme, these 8 legs AVRs are really similar. What do you think about it? I know the solution named `ATTinyCore`, but looks like compile much more bigger hex, maybe I missed a flag somewhere?

For an example solution,

- with MicroCore (LTO Enabled):
`Sketch uses 946 bytes (92%) of program storage space. Maximum is 1024 bytes.`
- with ATTinyCore for AtTiny25 (LTO Enabled):
`Sketch uses 1500 bytes (73%) of program storage space. Maximum is 2048 bytes.`

Thank you for you support!

MCUdude, 2018-03-09:
Hi! Even though ATtiny13 and ATtiny25/45/85 may seem very similar, they have different peripherals and registers. MicroCore is very tightly written specific for the ATtiny13, and it would be a huge job to implement the ATtiny25/45/85 series. If you need efficiency for the ATtiny25/45/85 series your best shot is probably to install [sleemanj's attiny core](https://github.com/sleemanj/optiboot/blob/master/dists/README.md).

closed 2018-03-09 by MCUdude

lazlyhu, 2018-03-09:
@MCUdude Thank you, a did a fast test around it, the numbers looks pretty good. Best regards, Lazly