MCUdude/MicroCore #53: Arduino 1.8.5 Error compiling for board ATtiny13.

c-e-p-x-u-o, 2018-03-21:
I have successfully burned bootloader but can't compile simple blink example:
```
sorry - this program has been built without plugin support
exit status 1
Error compiling for board ATtiny13.
```

```
#define LED_BUILTIN 1

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  delay(200);
  digitalWrite(LED_BUILTIN, LOW);
  delay(100);
}
```

per1234, 2018-03-21:
I suspect the problem is that you're using a compiler version that doesn't support LTO but you have **Tools > Compiler LTO** set to "Enabled (default).

Try setting that option to "Disabled". If the problem goes away then you either need to leave it disabled (which is unfortunate because you need all the optimization you can get when working with such limited memory) or upgrade to a newer version of Arduino AVR Boards.

c-e-p-x-u-o, 2018-03-21:
Yes, you are right. Can it be autoconfigured to "Disabled" if compiled doesn't support it?

per1234, 2018-03-21:
No, the Arduino IDE doesn't offer that sort of powerful configuration option for hardware packages.

MCUdude does have LTO disabled by default in the other packages but I think the idea is that it's very necessary in MicroCore and most users are going to have a recent version of Arduino AVR Boards anyway so it's better to have the improved optimization as default at the cost of causing some confusion for the few people who haven't updated.

I think the best we could do is to make the error less confusing. To this end, my proposal is to add the following to MicroCore's Arduino.h:
```cpp
#define GCC_VERSION (__GNUC__ * 10000 + __GNUC_MINOR__ * 100 + __GNUC_PATCHLEVEL__)
#if defined(COMPILER_LTO) && GCC_VERSION < 40902
#error Your compiler does not support LTO. Please either upgrade Arduino AVR Boards or select Tools > Compiler LTO > Disabled.
#endif
```
Arduino AVR Boards was updated from avr-gcc 4.8.1 to 4.9.2 at the same time the LTO support was added so the GCC version should be a reasonable indicator of LTO support. I'm not aware of an LTO plugin specific macro.

This must be an `#error` rather than `#warning` in order to be shown before the compilation fails due to the missing LTO plugin.

What do you think of that idea @MCUdude?

c-e-p-x-u-o, 2018-03-21:
I'm using debian sid with Arduino IDE 1.8.5.
```
% apt policy gcc-avr               
gcc-avr:
  Installed: 1:5.4.0+Atmel3.6.0-1+b1
```
What should I update?

per1234, 2018-03-21:
Well the question is whether that is the version of avr-gcc that is actually being used. Usually the Arduino IDE will use the tools that are included with the hardware package rather than whatever is already installed on your system.

You could check this by:

1. File > Preferences > Show verbose output during: > compilation (check) > OK
2. Sketch > Verify/compile

After the compilation finishes, check the contents of the black console window at the bottom of the Arduino IDE window. You will need to scroll up to see it all. You should see the path to the compiler tools being used (e.g. avr-g++, avr-gcc).

If the problem is that your avr-gcc 5.4.0 is being used but was not built with the LTO plugin then the solution is going to be more complex than just updating Arduino AVR Boards.

My helpful error message will also fail in this case. As I indicated previously, the compiler version is not the best indicator since the same error will occur even with a new compiler that does not have the plugin.

Are you running the official Arduino IDE downloaded from https://www.arduino.cc/en/Main/Software or some modified nightmare from a package manager?

Which version is shown for Arduino AVR Boards when you open **Tools > Board > Boards Manager**?

MCUdude, 2018-03-21:
@per1234 I think it's a good idea to add the gcc error to Arduino.h. I'm kinda busy this week, but I'll merge it if you do a PR 👍

c-e-p-x-u-o, 2018-03-21:
4.8.1 you are right!

MCUdude, 2018-03-21:
For reference ec76770

MCUdude, 2018-03-26:
I'm closing this issue as I assume you figured this out

closed 2018-03-26 by MCUdude