MCUdude/MicroCore #54: FastLED compilation error

c-e-p-x-u-o, 2018-03-21:
```
In file included from /root/Arduino/libraries/FastLED/FastLED.h:51:0,
                 from /root/Arduino/RGB-HUE-RR/RGB-HUE-RR.ino:5:
/root/Arduino/libraries/FastLED/fastpin.h: In instantiation of 'class FastPin<1u>':
/root/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:96:49:   required from 'class ClocklessController<1u, 3, 3, 6, (EOrder)10u, 0, false, 10>'
/root/Arduino/libraries/FastLED/chipsets.h:460:7:   required from 'class WS2811Controller800Khz<1u, (EOrder)10u>'
/root/Arduino/libraries/FastLED/FastLED.h:111:52:   required from 'class WS2811<1u, (EOrder)10u>'
/root/Arduino/libraries/FastLED/FastLED.h:304:33:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with CHIPSET = WS2811; unsigned char DATA_PIN = 1u]'
/root/Arduino/RGB-HUE-RR/RGB-HUE-RR.ino:13:43:   required from here
/root/Arduino/libraries/FastLED/fastpin.h:207:2: error: static assertion failed: Invalid pin specified
  static_assert(validpin(), "Invalid pin specified");
  ^
exit status 1
Error compiling for board ATtiny13.
```

Code (it works with arduino atmega and Attiny85  ATTinyCore
```
// this is temporary workaround for ATTinyCore
extern "C" {
  volatile unsigned long timer0_millis = 0; // FAKE FAKE FAKE!
}

#include<FastLED.h>
#define NUM_LEDS 8

CRGBArray<NUM_LEDS> leds;

static uint8_t hue[NUM_LEDS];

void setup(){
  FastLED.addLeds<WS2811,1>(leds, NUM_LEDS);

  for(int i = 0; i < NUM_LEDS; i++){
    hue[i] = i;
  }
}

void loop(){ 
  for(int i = 0; i < NUM_LEDS; i++){   
    leds[i] = CHSV(hue[i],255,255);
    FastLED.delay(0);
    hue[i] ++;
  }
}
```

MCUdude, 2018-03-21:
I have no idea. I've never used the FastLED library, and I'm not even able to figure out what's actually causing the error. I suggest you create an issue over at the FastLED repo instead.

BTW are you even able to fit this into 1024 bytes?

c-e-p-x-u-o, 2018-03-21:
I don't know the size, as it's compilation time error.

focalintent, 2018-03-21:
The problem is FastLED does direct gpio register access at compile time, which means it needs explicit pin to port mappings defined for each platform variation, and no one has submitted one for the attiny13 yet.

MCUdude, 2018-03-21:
I don't think we'll ever see FastLED running on ATtiny13. When compiling the example you provided for ATtiny45, the compiled code takes 3334 bytes with LTO enabled

I don't see how the same example would ever fit into an ATtiny13.

closed 2018-03-23 by MCUdude

c-e-p-x-u-o, 2018-03-23:
thank you all for quick response and notes!