MCUdude/MicroCore #55: half-size millis ISR

nerdralph, 2018-03-23:
Although I rarely use Arduino/Wiring, I decided to take a quick look at how much code could be saved in the millis ISR.  The current version in C takes 72 bytes.  After a couple itterations I got it down to 36 bytes in assembler:

`#define tmp1 r16
.global WDT_vect
WDT_vect:
    push ZL
    in ZL, SREG
    push ZL                             ; save SREG
    push ZH
    push tmp1
    ldi ZL, lo8(wdt_interrupt_counter)  ; must be 8-byte aligned
    sub ZH, ZH                          ; also clears carry
add:
    ld tmp1, Z
    sbci tmp1, -1
    st Z+, tmp1
    andi ZL, 0x04
    breq add
    pop tmp1
    pop ZH
    pop ZL
    out SREG, ZL
    pop ZL
    reti
`

This is untested prototype code.  It requires the 4-byte counter to be 8-byte aligned; that can be guaranteed with a pragma.  Without the 8-byte alignment trick, the smallest version I could come up with was 42 bytes.

nerdralph, 2018-03-23:
I think it would be even better to change:
'    sbci tmp1, -1'
to
'    sbci tmp1, -16'

Then the millis() function could simply return the counter value without having to multiply by 16.

MCUdude, 2018-03-23:
Awesome! I have very little experience with assembler, but I understand it's very powerful and efficient.
How would this look like in inline assembly? I'd love to test this, but have now idea how..

nerdralph, 2018-03-23:
> How would this look like in inline assembly? I'd love to test this, but have now idea how...

When I do AVR development, I usually keep my asm in separate .S files and just link them in with the rest of the compiled .c & .cpp files.  The Arduino build system doesn't use standard makefiles, so that may make it more difficult.

I enjoy writing code much more than I do integrating it.  I might decide to dig a little deeper into how to easy it is to integrate asm into the core builds, but I'm more likely to get distracted by a more interesting coding problem.

MCUdude, 2018-03-23:
Thank you, I really appreciate your effort. Looking forward to see the outcome 🙂

beicnet, 2018-03-23:
Arduino Inline Assembly [Examples](https://ucexperiment.wordpress.com/2016/03/04/arduino-inline-assembly-tutorial-1/) and [eBook](https://leanpub.com/arduinoinlineassembly).

Just my two cents! :wink:

per1234, 2018-03-23:
You should be able to just drop the .S file in the core library folder. Here's an example of a .S file used in the Arduino AVR Boards package:
https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/wiring_pulse.S

MCUdude, 2018-03-23:
> You should be able to just drop the .S file in the core library folder. 

I'm struggling trying to include it in the source file. I've created *millis.S*, which contains this code:

``` asm
#include <avr/io.h>

#define tmp1 r16

.global WDT_vect
WDT_vect:
    push ZL
    in ZL, SREG
    push ZL                             ; save SREG
    push ZH
    push tmp1
    ldi ZL, lo8(wdt_interrupt_counter)  ; must be 8-byte aligned
    sub ZH, ZH                          ; also clears carry
add:
    ld tmp1, Z
    sbci tmp1, -1
    st Z+, tmp1
    andi ZL, 0x04
    breq add
    pop tmp1
    pop ZH
    pop ZL
    out SREG, ZL
    pop ZL
    reti
```

I've placed the file in the cores/microcore folder, and commented out the `ISR(WDT_vect)` located in wiring.h

This is the error that Arduino IDE outputs:
```
/Users/Hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/millis.S: Assembler messages:
/Users/Hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/millis.S:8: Error: operand out of range: 95
/Users/Hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/millis.S:23: Error: operand out of range: 95
exit status 1
Error compiling for board ATtiny13.
```

If I add `#include "Arduino.h"` to the assembler file the outputted error is endlessly long

nerdralph, 2018-03-23:
You'll need:
'#define __SFR_OFFSET 0'
before the #include <avr/io.h>

Here's an example of one of my avr assembler source files implementing an
ISR:
https://github.com/nerdralph/nerdralph/blob/master/avr/t32isr.S

nerdralph, 2018-03-23:
I've got it building with the millis.S.  It's not allocating the 4 bytes of global memory for the counter though.
```
Sketch uses 142 bytes (13%) of program storage space. Maximum is 1024 bytes.
Global variables use 0 bytes (0%) of dynamic memory, leaving 64 bytes for local variables. Maximum is 64 bytes.
```

nerdralph, 2018-03-23:
Actually this looks even better than I thought.  The compiler seems to optimize away the 4-bytes when millis is not used.  Here's my test sketch:
```
void loop() {
  // put your main code here, to run repeatedly:
  delay(500);
  digitalWrite(LEDPIN, HIGH);
  uint32_t count = millis();
  delay(count);
  digitalWrite(LEDPIN, LOW);
}
```
And the build output:
```
Sketch uses 168 bytes (16%) of program storage space. Maximum is 1024 bytes.
Global variables use 4 bytes (6%) of dynamic memory, leaving 60 bytes for local variables. Maximum is 64 bytes.
```

nerdralph, 2018-03-23:
And changing the ISR to count by 16 saves 14 bytes in the millis() function.
```
Sketch uses 154 bytes (15%) of program storage space. Maximum is 1024 bytes.
Global variables use 4 bytes (6%) of dynamic memory, leaving 60 bytes for local variables. Maximum is 64 bytes.
```
I dont' want to clone MicroCore, so I just pushed the code to my nerdralph repo:
https://github.com/nerdralph/nerdralph/blob/master/avr/millis.S

MCUdude, 2018-03-23:
Cool! Still can't quite figure out how to make the compiler include the assembler file though. I just commented out the watchdog counter variable (since it's defined in millis.S), and it seems like it can't find the variable:

```
/Users/Hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/wiring.c: In function 'millis':
/Users/Hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/wiring.c:38:10: error: 'wdt_interrupt_counter' undeclared (first use in this function)
   return wdt_interrupt_counter;
          ^
/Users/Hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/wiring.c:38:10: note: each undeclared identifier is reported only once for each function it appears in
/Users/Hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/wiring.c:39:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
exit status 1
Error compiling for board ATtiny13.
```

nerdralph, 2018-03-23:
I'm having a similar problem.  It's compiling millis.S to millis.S.o, and
adding it to core.a, but I think it is optimizing it away when linking with
LTO.  I'm pretty sure I'll be able to figure how to make sure the compiler
keeps it.

nerdralph, 2018-03-23:
Declaring wdt_interrupt_counter as extern in wiring.c, and defining it in millis.S forces the linker to pull in millis.S.o.  However wdt_interrupt_counter isn't getting initialized to zero before main.  I think I just have to put it in .bss to make that happen.  Will see.

MCUdude, 2018-03-24:
Great! It's 1AM where I live, so I'll have to test it in the morning. Will report back as soon as I can.

MCUdude, 2018-03-24:
Ive added `extern uint32_t wdt_interrupt_counter;` to Arduino.h and updated millis.S with the latest version you've uploaded. It compiles just fine but the code freezes pretty soon because it doesn't enter the WDT ISR. No idea why.

This is the code I'm trying to compile. The LED never turns on. Same result with and without LTO
``` c
#define LEDPIN 2

void setup()
{
  pinMode(LEDPIN, OUTPUT);
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(500);
  digitalWrite(LEDPIN, HIGH);
  uint32_t count = millis();
  delay(count);
  digitalWrite(LEDPIN, LOW);
}
```
With LTO:
```
Sketch uses 158 bytes (15%) of program storage space. Maximum is 1024 bytes.
Global variables use 4 bytes (6%) of dynamic memory, leaving 60 bytes for local variables. Maximum is 64 bytes.
```
Without LTO:
```
Sketch uses 324 bytes (31%) of program storage space. Maximum is 1024 bytes.
Global variables use 4 bytes (6%) of dynamic memory, leaving 60 bytes for local variables. Maximum is 64 bytes.
```

nerdralph, 2018-03-24:
Ok, I'll dig out a tiny13a and debug it later today.  I already noticed a
bug in the counter that means it counts incorrectly.

MCUdude, 2018-03-24:
Thanks. I've created a new branch where I've pushed the implementation.
[https://github.com/MCUdude/MicroCore/tree/millis_improvement](https://github.com/MCUdude/MicroCore/tree/millis_improvement)

nerdralph, 2018-03-24:
I also noticed a bug in the test.  It can delay for many seconds between turning on the LED and turning it off.  Instead of:
'uint32_t count = millis();'
limiting the delay to 1023ms would be better:
'uint32_t count = millis() & 0x3ff;'

That doesn't explain why you never saw the LED turn on in the first place though.

nerdralph, 2018-03-24:
The counter fix meant adding a few more instructions so only the low byte is incremented by more than 1.  According to the datasheet figures 19-65 and 19-66, the WD oscillator frequency is typically 110-111kHz, so I'm counting by 19 instead of 16.
I also noticed the following in the AVR Instruction Set manual:
Note that only the low byte of the Z-pointer is updated in devices with no more than 256 bytes data space. For such devices, the high byte of the pointer is not used by this instruction and can be used for
other purposes.

Therefore I don't need to save/restore ZH, which makes up for the few instructions I had to add to fix the counter bug.

MCUdude, 2018-03-24:
ok. I'm ready to test when you publish your code

nerdralph, 2018-03-24:
> ok. I'm ready to test when you publish your code

It's taking me longer than expected to test because of problems with avrdude.  I use v6.1 with my avr-gcc command-line development environment, and it works fine with the USBasp programmers I have.  Avrdude 6.3 that comes with Arduino 1.8.5 doesn't work with the exact same USBasp that works on the same computer with avrdude 6.1.  Rather than try to figure out the AVRdude issue, I'm just going to build with Arduino/MicroCore, and flash the .hex file with Avrdude 6.1 from the cli.

nerdralph, 2018-03-24:
OK, give it a try now as I just pushed a new millis.S.  I found a stupid bug that was putting the ISR into an infinite loop.
I've got a t13a blinking 500ms on, and a pseudo-random off time as expected.

MCUdude, 2018-03-24:
I'm experiencing weird behavior. The LED stays on for a about 30 seconds, then turns off for about 250ms, the on again. This repeats over and over again.

Running this code:
``` c
#define LEDPIN 2

void setup()
{
  pinMode(LEDPIN, OUTPUT);
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(500);
  digitalWrite(LEDPIN, HIGH);
  uint32_t count = millis();
  delay(count);
  digitalWrite(LEDPIN, LOW);
}
```

MCUdude, 2018-03-24:
I connected a 7-seg LED display I use when debugging the ATtiny13. When the WDT is running it just print random stuff on the LED display. However if I turn off interrupts it prints just fine. Seems like the WDT_vect routine messes up the overall behavior..

Here's the crude LED segment code:
``` c
// Prints numbers on the 74HC595 based LED display

static const uint8_t latchpin = 3;
static const uint8_t clockpin = 0;
static const uint8_t datapin = 1;

static const uint16_t powersof10[4] PROGMEM = {1, 10, 100, 1000};

// Must be stored in RAM
static const uint8_t segment[12] = 
{
  0b00111111, //0
  0b00000110, //1
  0b01011011, //2
  0b01001111, //3
  0b01100110, //4
  0b01101101, //5
  0b01111101, //6
  0b00000111, //7
  0b01111111, //8
  0b01101111, //9
  0b10000000, //Decimal 
  0b00000000  //Clear
};

uint32_t val = 0;
                      
void setup()
{
  pinMode(latchpin, OUTPUT);
  pinMode(clockpin, OUTPUT);
  pinMode(datapin, OUTPUT);
//  cli();
}

void loop()
{  
   val++;
  printSegment(val);
  delay(100);
}

void printSegment(uint16_t number)
{
  digitalWrite(latchpin, LOW);
  shiftOut(datapin, clockpin, MSBFIRST, segment[nthdigit(number, 0)]);
  shiftOut(datapin, clockpin, MSBFIRST, segment[nthdigit(number, 1)]);
  shiftOut(datapin, clockpin, MSBFIRST, segment[nthdigit(number, 2)]);
  shiftOut(datapin, clockpin, MSBFIRST, segment[nthdigit(number, 3)]);
  digitalWrite(latchpin, HIGH);
}

uint8_t nthdigit(int x, int n)
{
  return ((x / powersof10[n]) % 10);
}

```

nerdralph, 2018-03-24:
> I'm experiencing weird behavior. The LED stays on for a about 30 seconds, then turns off for 500ms, the on again. This repeats over and over again.
> 
> Running this code:
> 
> #define LEDPIN 2
> 
> void setup()
> {
>   pinMode(LEDPIN, OUTPUT);
> }
> 
> void loop() {
>   // put your main code here, to run repeatedly:
>   delay(500);
>   digitalWrite(LEDPIN, HIGH);
>   uint32_t count = millis();
>   delay(count);
>   digitalWrite(LEDPIN, LOW);
> }
If you look at my earlier comments, you'll see I said to use:
uint32_t count=millis() & 0x3ff;

nerdralph, 2018-03-24:
I am seeing some strange behavior as well, so there may yet be another bug in the ISR.
I added a debug line that toggles a pin in the ISR, and it is running as expected, at a frequency of ~52Hz (about every 19.3ms).
What I'm thinking of doing is porting my bit-bang assembler UART, so I can output some debug info.

MCUdude, 2018-03-24:
> If you look at my earlier comments, you'll see I said to use:
uint32_t count=millis() & 0x3ff;

I tried both. It was easier to spot the issue without `& 0x3ff`

nerdralph, 2018-03-24:
OK, I stepped away from the computer to get a snack, and then the bug was obvious.  Instead of stopping the counter loop 4 bytes after the address of wdt_interrupt_counter, I was topping at address 0x04 (i.e. after it wraps from 255->0).
Fixed version pushed, and so far the wierdness I was seeing is gone.

MCUdude, 2018-03-24:
I just tested it, and it does count now. However it seems like it doesn't count past 255 before starting over. I'm using a 74HC595 based 7seg display to print numbers, and I'm not able to print a number larger than 255.

MCUdude, 2018-03-24:
Here's the the loop I'm running to drive the display:

``` c
uint32_t currentMillis = 0;
uint32_t previousMillis = 0;

void loop()
{   
  currentMillis =  millis();
  previousMillis = max(currentMillis, previousMillis);
  printSegment(previousMillis);
}
```

[image]

nerdralph, 2018-03-24:
>I just tested it, and it does count now. However it seems like it doesn't count past 255 before starting over. I'm using a 74HC595 based 7seg display to print numbers, and I'm not able to print a number larger than 255.

I've been doing more testing as well and just noticed something similar (trying to implement a millis-based delay).  I think I already know the problem; I'm pretty sure cpi resets the carry flag from the sbci.

nerdralph, 2018-03-24:
Yup, cpi was resetting the carry flag so the counter wouldn't increment the next byte after rollover.  Fixed and pushed now.  There is still a potential bug with 32-bit rollover, so you might not want to push this to master yet.  The bug won't appear for 50 days (4.3 billion ms).  I have two ideas for a solution.  One takes an extra byte of RAM, the other takes 6-8 more bytes of code.  I'm going to sleep on it and see if I come up with a more optimal solution.

MCUdude, 2018-03-24:
It's finally counting up like it should! 
Quick question, I can save four bytes if I add *volatile* to `extern uint32_t wdt_interrupt_counter;`. Why does volatile reduce the code size, and should I use it when the variable is define in millis.S?

nerdralph, 2018-03-24:
It's not immediaty obvious to me why volatile reduces the code size.  I'll
try it and look at the disassembly of the compiled code.

MCUdude, 2018-03-25:
I'm not sure what we see as the most precious memory. The flash memory perhaps?

nerdralph, 2018-03-25:
On my testing I'm seeing more memory use with volatile.  When I use just the lower 16 bits of the counter, the compiler seems to get stupid when I make it volatile because it does 4 lds instructions instead of just 2.  If you want I can post a  gist of the example code.

nerdralph, 2018-03-25:
> I'm not sure what we see as the most precious memory.  The flash memory perhaps?

That's what I'm leaning towards.  And if RAM is more important, I have an idea to get back the 2 bytes of stack space that is wasted by the rcall to main from init.

MCUdude, 2018-03-25:
Great! Your latest commit works like a charm too. But why five bytes? 

Do you know how it would handle an overflow?

nerdralph, 2018-03-25:
> Do you know how it would handle an overflow?

With 4 bytes for the counter, it overflows after 50 days.  With 5 bytes, it will overflow after 35 years, causing it to corrupt the byte of memory after the 5 bytes reserved for the counter.  I don't think anyone can realistically say they expect to have an attiny13 running continuously for >35 years without a reset.

MCUdude, 2018-03-25:
>  I don't think anyone can realistically say they expect to have an attiny13 running continuously for >35 years without a reset.

Yep, that makes sense. I think we can "waste" one byte of ram in order to not having to deal with overflow.

nerdralph, 2018-03-30:
I noticed you switched the increment to 16 instead of 19 in the millis-improvement branch.  As I mentioned above, the increment of 19 is based on the documented typical frequency of the watchdog of 110-111kHz, or 9.05uS per clock.  With a prescaler of 2048, that's 18.53ms.

MCUdude, 2018-04-01:
Sorry for the late reply, I've been on vacation 😉 
I was doing some testing, and forgot to change it back to 19. Thanks for pointing that out.