MCUdude/MicroCore #56: small space savings tweaks

nerdralph, 2018-03-24:
According to the attiny13/13a manual section 4.3.1, the I bit in SREG is clear after reset.  Therefore the cli() in init() can be removed:
https://github.com/MCUdude/MicroCore/blob/master/avr/cores/microcore/wiring.c#L172

Also, from what I can tell from the manual, it is not necessary to reset the watchdog timer before enabling the watchdog interrupt.

Removing those two lines saves 4 bytes.

MCUdude, 2018-03-24:
Thanks, you're absolutely right! I'll push a commit after we're finished working with the millis ISR

nerdralph, 2018-03-24:
I'd also suggest checking SETUP_PWM in digitalWrite when SAFEMODE is on.  If SETUP_PWM is undefined, then you can skip turnOffPWM().

nerdralph, 2018-03-24:
You can save a few more bytes by changing the register settings in init() from
TCCR0B |= _BV(CS00) | _BV(CS01); 
to:
TCCR0B = _BV(CS00) | _BV(CS01); 
The assignment compiles to ldi + out, while the |= compiles to in, ori, + out.
Any time you are setting just one register bit, still use |= as that will compile to a single sbi instruction.

MCUdude, 2018-05-27:
Seems like all suggestions are now implemented 👍

closed 2018-05-27 by MCUdude