MCUdude/MicroCore #57: decrease F_CPU by 2-3%

nerdralph, 2018-03-25:
Based on my small sample of ATtiny13a chips (a few DIP, a few SOIC-8), the typical clock frequency is consistently lower than the nominal 9.6Mhz.  The datasheet figure 19-60 shows a typical frequency of ~9.5Mhz at 20-25C.  My sample of chips average around 9.4Mhz@5V and 9.3Mhz@3.3V.

I'm thinking of porting/testing my bit-bang assembler UART with MicroCore, but I don't want to do it if users are likely to run into timing problems due to incorrect assumptions about the typical oscillator frequency.

MCUdude, 2018-03-25:
Good idea! I'll do some testing as well. Do you have a quick code sample to output an F_CPU derived clock to one of the IO pins?

I'll try to tweak F_CPU a little and push it when we're finished the millis project

nerdralph, 2018-03-25:
> Do you have a quick code sample to output an F_CPU derived clock to one of the IO pins?
Just blink a LED using delay():
```
  delay(50);
  digitalWrite(LEDPIN, HIGH);
  delay(50);
  digitalWrite(LEDPIN, LOW);
```
Disable millis for better accuracy.  Set your multimeter  to frequency count mode, and the variance from 10Hz tells you the true clock rate.  I've found my low-end multimeters accurate to within 0.03% on frequency.
If you want to get particularly accurate, use an oscilloscope.  If you want to get *really* accurate and don't have a GPS with PPS output, use a NTP-synced pulse output.
http://n1.taur.dk/nft/nft.pdf

MCUdude, 2018-03-25:
Thanks! I have a decent digital oscilloscope (the famous Rigol DS1054z) I do my measurements on. Will look into this soon!

nerdralph, 2018-03-25:
> Thanks! I have a decent digital oscilloscope (the famous Rigol DS1054z) I do my measurements on.

Same here.  I used the 1kHz output tabs at the bottom right of the scope to verify the accuracy of my multi-meter's frequency counter.  I got 999.8 Hz.

MCUdude, 2018-03-25:
Woah! My Attiny13A which I bought cheaply from China years ago is way off! We're talking 93.8% of 9.6 MHz, which results in a 9.004800 MHz clock at 5V, and very much the same at 3.3V.

I've been using my accurate signal generator as a clock source when I've tuned the timing critical functions, so I've actually never noticed this

Jackjan4, 2019-02-26:
F_CPU however shouldn't be decreased since every Tiny13 is off by a different value. Instead user should use OSCCAL to tune their frequency for their own microcontroller.

MCUdude, 2019-12-12:
I figured since the internal oscillator of the ATtiny13 is so inaccurate it's better to provide some kind of OSCCAL sketch using @nerdralps awesome BasicSerial code to achieve this, by typing a known character in the terminal and let the T13 calculate the width of the pulses or something.

closed 2019-12-12 by MCUdude