MCUdude/MicroCore #58: A way to disable millis on specific project?

Renkin42, 2018-03-31:
Sorry if there's something obvious I've missed here. I was wondering if there is a way to change core options for an individual sketch rather than the entire core, in particularly disabling millis since, even though i don't need it for this project it will be annoying to turn back on if I need it in the future. I tried putting #undef ENABLE_MILLIS at the top of the sketch and making a copy of core_settings.h in the sketch folder, but that did nothing, sadly. Any ideas?

MCUdude, 2018-03-31:
Hi!
Sadly, it's not. You'll have to disable millis in the core settings file.
All the core files are compiled and linked before the sketch, so trying to #undef millis won't work either.  
What you can do is to disable millis in the core settings file, compile and store the compiled hex file somewhere.

nerdralph, 2018-03-31:
In th millis_improvement branch which uses my asm millis function, it seems to be automatically disabled when it is not used.  With LTO, the linker only uses object files for which it has undefined symbols.  When your sketch code doesn't reference wdt_interrupt_counter, millis.o doesn't get linked into the binary.

Renkin42, 2018-04-01:
I was afraid of that. Oh well, worth asking, I suppose. I might try that branch though, thanks for the tip, ralph. However I assume if I use the watchdog timer it will still cause a conflict? If so I can probably just work around that using Timer0, I'll just need to fiddle with my code a bit.

closed 2018-04-01 by Renkin42