MCUdude/MicroCore #60: INT0 not firing if millis are disabled?

Renkin42, 2018-04-11:
I was testing this sketch on my ATtiny to see if I could confirm the mode works, but annoyingly the interrupt didn't even seem to be firing, which was confounding since I had it at least firing before changing this code. I tried reordering the interrupt attach sequence and a few other things but it just did nothing. It occurred to me though that one thing that had changed was that I disabled millis. Sure enough when I re-enabled them the led turned off and back on when I pressed the button just as expected. Any ideas what the hell is going on here?

```
/* 
 * This sketch uses the ATtiny13 MicroCore by MCUdude
 * See https://github.com/MCUdude/MicroCore
 * To compile, ENABLE_MILLIS must be commented out in
 * core_settings.h. They can be re-enabled once
 * the code has been compiled.
 */
#include <avr/sleep.h>

#define DISP_PIN 3
#define DATA_PIN 0
#define CLK_PIN 2

#define DISP_TIME 4000 // Max = 4080
#define NUM_LOOPS 20
#define LOOP_TIME 1000 / NUM_LOOPS

volatile word onTime = 0;
volatile byte mode = 0;

void setup() {
  wdt_disable();
  analogReference(INTERNAL);
  pinMode(DATA_PIN, OUTPUT);
  pinMode(CLK_PIN, OUTPUT);
  pinMode(DISP_PIN, OUTPUT);
  digitalWrite(DISP_PIN, HIGH);
  set_sleep_mode(SLEEP_MODE_PWR_DOWN);
  sleep_enable();
  attachInterrupt(0, tempIsr, LOW);
  sleep_cpu();
}

void loop() {
  switch(mode) {
    case 1:
      digitalWrite(DISP_PIN, LOW);
      for(int i = 0; i < NUM_LOOPS; i++) {
        displayValue(0, false, false);
        delay(LOOP_TIME);
      }
      displayValue(0, false, false);
      waitMode();
      break;
    case 3:
      displayValue(0, false, false);
      waitMode();
      break;
    case 5:
      displayValue(0, false, false);
      waitMode();
      break;
    case 7:
      displayValue(0, false, false);
      waitMode();
      break;
    case 9:
      sleepMode();
      break;
    default:
      displayValue(0, false, false);
      if(onTime >= DISP_TIME) {
        sleepMode();
      }
      delay(1);
      onTime++;
      break;
  }
}

void tempIsr() {
  sleep_disable();
  detachInterrupt(0);
  mode++;
}

void displayValue(char value, boolean dec1, boolean dec2) {
  shiftOut(DATA_PIN, CLK_PIN, LSBFIRST, mode);
}

void waitMode() {
  mode++;
  onTime = 0;
  delay(1000);
  attachInterrupt(0, tempIsr, LOW);
}

void sleepMode() {
  mode = 0;
  digitalWrite(DISP_PIN, HIGH);
  delay(1000);
  sleep_enable();
  attachInterrupt(0, tempIsr, LOW);
  sleep_cpu();
}

```

MCUdude, 2018-04-11:
Good catch! Interrupts are not enabled by default to save a few bytes, but gets [automatically enabled](https://github.com/MCUdude/MicroCore/blob/5a4caa5e4b783f891ec8c8f1a25a8aac6e2324e0/avr/cores/microcore/wiring.c#L170) when millis is enabled.

Try adding `sei();` to your setup routine.

Renkin42, 2018-04-11:
Holy crap, that's it? Defeated by 1 line of code, lol. Sadly I just left for work, so I'll have to try it when I get home this afternoon.

MCUdude, 2018-04-11:
Yep, that's probably it 😉 However, interrupts should be automatically enabled when attachInterrupt is called, so I'll consider it a bug.

Renkin42, 2018-04-11:
OK, that worked. Unfortunately I have a new problem. The modes cycle just fine after programming, but after I disconnect and reconnect power, DISP_PIN goes flashes on and off and the shift register receives some nonsense data, and it will only stay on as long as the button is held, or until it "settles" back to mode 1, at which point everything works normally. Sometimes it's just a few cycles, sometimes it's several seconds. I've tried clamping mode below 9 but that doesn't seem to work.

MCUdude, 2018-04-11:
I just pushed a fix. However it's probably going to take a while before you see this fix in a boards manager release

commit d68eeb2d (2018-04-11): Turn on global interrupts when attachInterrupt is called
closes #60
closed 2018-04-11 by MCUdude