MCUdude/MicroCore #62: bad shiftOut code

nerdralph, 2018-05-26:
I looked at the shiftOut code, and it is pretty bad.  When clockPin, dataPin, and bitOrder are known at compile time, codesize is 22 instructions, and it takes a minimum of 21 cycles per bit.  A few years ago I explained in my blog why the default Arduino shiftOut code sucks:
http://nerdralph.blogspot.ca/2015/03/fastest-avr-software-spi-in-west.html

I modified my fast SPI C code to work with MicroCore:
```
#define SPIPORT PORTB
#define SPIPIN PINB

void spiNew(byte datapin, byte clockpin, byte mode, byte value){
  const byte mosipinmask = (1<<datapin);
  const byte clkpinmask = (1<<clockpin);
  byte i = 8;
  byte portbits = (SPIPORT & ~(mosipinmask | clkpinmask) );
  do{
    SPIPORT = portbits;      // clk and data low
    if ( mode == MSBFIRST){
      if(value & 0x80) SPIPIN = mosipinmask;
    }
    else{
      if(value & 0x01) SPIPIN = mosipinmask;
    }
    SPIPIN = clkpinmask;     // toggle clk
    if ( mode == MSBFIRST) value <<= 1;
    else value >>= 1;
  } while(--i);
  return;
}
```
My version compiles to 13 instructions, and takes 8 cycles per bit.
I can do a pull request, or you can copy/modify to your liking.

MCUdude, 2018-05-26:
Awesome! I'm going to test your code and add it to the master branch if it work like it should. 
Thanks for contributing Ralph!

MCUdude, 2018-05-27:
I modified your code a little. It works fine with a 7 segment LED display I use for testing. But shouldn't PORTB and PINB be controlled with some bitwise operations instead in order to not mess with the rest of PORTB? (See commented lines in code)

``` c
void shiftOut(uint8_t dataPin, uint8_t clockPin, uint8_t bitOrder, uint8_t value)
{
  
  const uint8_t datapinMask = _BV(dataPin);
  const uint8_t clkpinMask = _BV(clockPin);
  uint8_t i = 8;
  uint8_t portbits = (PORTB & ~(datapinMask | clkpinMask));
  
  #if defined(SAFEMODE)
    if(clockPin > 5 || dataPin > 5) // Return if pin number is too high
      return;
    if(clockPin < 2)
      turnOffPWM(clockPin); // If it's a PWM pin, make sure PWM is off
    else if(dataPin < 2)
      turnOffPWM(dataPin);  // If it's a PWM pin, make sure PWM is off
  #endif    
   
  do
  {
    PORTB = portbits; // Clock and data pin low   <---- wouldn't this affect the whole port, not just the clock and data pin??
    if(bitOrder == MSBFIRST)
    {
      if(value & 0x80)
        PINB = datapinMask;  // <---- wouldn't this affect the whole port, not just the clock and data pin??
    }
    else
    {
      if(value & 0x01) 
        PINB = datapinMask; // Toggle clock pin   <---- wouldn't this affect the whole port, not just the clock and data pin??
    }
    PINB = clkpinMask; // Toggle clock pin   <---- wouldn't this affect the whole port, not just the clock and data pin??
    if(bitOrder == MSBFIRST) 
      value <<= 1;
    else 
      value >>= 1;
  }
  while(--i);
}
```

nerdralph, 2018-05-27:
The portbits variable saves the state of the other pins so they are not
clobbered.

MCUdude, 2018-05-27:
Cool! I did a quick test and you're absolutely right.

closed 2018-05-27 by MCUdude

nerdralph, 2018-06-03:
I just realized a slight issue.  The current code leaves the clock pin high after shiftOut.  while the original code leaves it low.  The easy fix is to move
`   PORTB = portbits;`
to the end of the while loop.

MCUdude, 2018-06-03:
Thanks for pointing that out. Do you mean like this?

``` c
void shiftOut(uint8_t dataPin, uint8_t clockPin, uint8_t bitOrder, uint8_t value)
{
  
  const uint8_t datapinMask = _BV(dataPin);
  const uint8_t clkpinMask = _BV(clockPin);
  uint8_t i = 8;
  uint8_t portbits = (PORTB & ~(datapinMask | clkpinMask));
  
  #if defined(SAFEMODE)
    if(clockPin > 5 || dataPin > 5) // Return if pin number is too high
      return;
    if(clockPin < 2)
      turnOffPWM(clockPin); // If it's a PWM pin, make sure PWM is off
    if(dataPin < 2)
      turnOffPWM(dataPin);  // If it's a PWM pin, make sure PWM is off
  #endif    
   
  do
  {
    if(bitOrder == MSBFIRST)
    {
      if(value & 0x80)
        PINB = datapinMask;
    }
    else
    {
      if(value & 0x01) 
        PINB = datapinMask;
    }
    PINB = clkpinMask; // Toggle clock pin
    if(bitOrder == MSBFIRST) 
      value <<= 1;
    else 
      value >>= 1;
  }
  PORTB = portbits; // Clock and data pin low
  while(--i);
}
```

bratoff, 2019-11-04:
There is still an error in this code that could cause the first bit output to be incorrect or the first clock pulse to be missed, depending on the initial state of the data and clock pins.

If the state of the pin selected by dataPin at entry to shiftOut equals the state of the first bit to be shifted out, the wrong value will be sent.  For example, if dataPin were already high at entry and you attempted to do a MSB-first shift out of 0xA0, you would instead shift out 0x20.  This happens because PINB = datapinMask would set dataPin low rather than high (because writing to the PINB register toggles the masked pin).

Similarly, if the state of the pin selected by clockPin is high at entry to shiiftOut, the first clock pulse will be missed.

Both of these problems can be corrected during initialization.  Just replace
  uint8_t portbits = (PORTB & ~(datapinMask | clkpinMask));
with
  uint8_t portbits = (PORTB &= ~(datapinMask | clkpinMask));

This ensures that the two pins are initialized low and can only be toggled high.

A less desirable but equally effective solution would be to document that dataPin and clockPin must both be low when shiftOut is called.

nerdralph, 2019-11-04:
It's not an error.  The reset state of PORTB is zero, so there's no chance
of them being high unless someone intentionally sets it high.

bratoff, 2019-11-04:
But that's exactly the point - you don't know what the calling program has done prior to calling shiftOut.  You're good in the most likely case, but at the very least the limitation should be documented.

nerdralph, 2019-11-04:
Good coding practice dictates you leave the port in the state it started.
Hans can decide if he wants to document that.  I'd say no if it were up to
me since MicroCore is meant for people with more clue than a newbie
Ardruino user.  Trying to compensate for stupid things that can be done by
user not only takes up resources but also enables bad coding.

bratoff, 2019-11-04:
I think 45 years as a software professional makes me something other than a newbie, thank you.

You should never make assumptions about what the user will do.  In this particular case it's a trivial fix that adds 4 clock cycles not inside a loop.  If you're that worried about efficiency, put it inside the SAFEMODE wrapper.

nerdralph, 2019-11-04:
If you want an embedded library that works even when you do stupid things,
then you can't be a very good coder and should probably stick to the
regular Arduino core.
If you can show a real use case where a user would want to set the clock
pin high before shiftOut, I'll consider submitting a patch with safe mode
code to set the pin low.

bratoff, 2019-11-04:
I gave you a concrete example in which the data pin could be high at entry, which would definitely produce erroneous results.  There is nothing unusual (or newbie) about using the same data pin(s) with multiple clock pins to drive multiple devices.

Also, this is the second response in which you have cast personal aspersions rather than sticking to the technical issue at hand.  I can see from your profile that you're not a newbie either.  Is this how you would expect someone to respond to one of your constructive suggestions?

nerdralph, 2019-11-04:
The initial reference to newbies was not refering to you, but you decided
to take it that way.  My responses have been quite civil compared to what
you might get on something like the linux kernel mailing list.
I stick by my assertion that its stupid to set the clock pin high before
calling shiftOut.  I can think of no reason why an experienced, intelligent
embedded developer would want to do that.  It seems you can't either.

bratoff, 2019-11-04:
That's true if you if you ignore what I clearly stated twice.  Anyway, you can lead a horse to water but you can't make it drink, so I'm done.  Hans will read this and decide if it's worth doing anything.

nerdralph, 2019-11-04:
I ignored what you said because its not specific or realistif.  While it is
common to have multiple SPI slaves share the same data/clock with different
CS pins, my shiftOut code works pefectly in that case.
I doubt Hans will side with your emotional handwaving over what I think,
since he's relied on me for a number of complex optimizations to MicroCore.

bratoff, 2019-11-04:
If you've read anything emotional into what I have written, that's just your own ego interfering with your objectivity.  I have no personal stake in who sides with who for whatever reason. 

I have found this core useful for a number of small projects and I appreciate the work that both you and Hans (and whoever else) have put into it.  Because I appreciate that work, I offered a suggestion to improve one small part of it.  That's all.  You're welcome.  Have a nice life.

commit 77edefe7 (2018-05-27): Improve shiftOut
Thanks to @nerdralph we not got an even more optimizedshoftOut function. Closes #62