MCUdude/MicroCore #64: upload protocol missing for USBasp (slow)

nerdralph, 2018-05-28:
When I select USBasp (slow), and upload, avrdude fails because the upload protocol variable is not set:
`C:\Program Files (x86)\arduino-1.8.5\hardware\tools\avr/bin/avrdude -CC:\Users\parsons\AppData\Local\Arduino15\packages\MicroCore\hardware\avr\1.0.3/avrdude.conf -v -pattiny13 -c{upload.protocol} -Pusb -B32 -Uhfuse:w:0xff:m -Ulfuse:w:0x3A:m -Uflash:w:C:\Users\parsons\AppData\Local\Temp\arduino_build_559788/1638lite.ino.hex:i
`

nerdralph, 2018-06-02:
The fix is relatively easy.  Change
`usbasp.protocol=usbasp`
to:
`usbasp.upload.protocol=usbasp`
in programmers.txt.

On a somewhat related point, I found that avrdude6.3 included with arduino 1.8.5 won't work unless libusb is upgraded to libusbK.  Details here:
https://github.com/arduino/avrdude-build-script/issues/1

MCUdude, 2018-06-02:
This was [fixed four months ago](https://github.com/MCUdude/MicroCore/commit/ea73f0713d97bc7d14ba7c31086f46d00352dc72)...
I guess you're looking at the boards manager version? I haven't done a release in a while, but you can always do a manual install 👍

nerdralph, 2018-06-02:
Yup, I am using the boards manager version.  Perhaps time for a 1.0.4 release?

closed 2018-06-02 by nerdralph

MCUdude, 2018-06-03:
> Yup, I am using the boards manager version. Perhaps time for a 1.0.4 release?

it definitely is! I just want to get Tone to work properly first. I don't have much time at the moment, but I'll see what I can do..