MCUdude/MicroCore #65: tone generating wrong frequency

nerdralph, 2018-05-28:
just tried tone(4, 3520), and the generated frequency is way off.  I measured ~2750Hz with my multimeter.  I didn't hook up my scope, but the sound is distorted, like it is not a clean 50% duty cycle output.
With millis disabled the results were similar, with the measured frequency ~2800Hz.  This is using the 9.6Mhz (~9.2 on my test chip) internal RC oscillator.

nerdralph, 2018-05-28:
Just found this in the code:
`#define TONE_MAX_FREQ 2093 // 2093.0 = C6`

So generating a compile error when it is called with >2093 would be a good idea.
I tried 1760, and get an actual (distorted) frequency of ~1420Hz.

MCUdude, 2018-05-28:
Strange.. My scope tells me the frequency is pretty much half of the expected frequency

``` c
void setup() {
    tone(4, 1024);
}

void loop() {
 }
```

With a perfect 8 MHz external clock my scope measures 504Hz

MCUdude, 2018-05-28:
The code is borrowed from @sleemanj's [repo](https://github.com/sleemanj/optiboot). It acts exactly the same when using tone() with his Arduino core.

Seems like calculation of the OCR0A value is way off. I'll continue working on it.

MCUdude, 2018-05-28:
Try this one instead with your scope. Seems to work quite well here!

``` c

void newTone(const uint8_t pin, const uint16_t frequency, const uint32_t length)
{
  if(frequency == 0) 
    return noTone(pin);

  uint16_t  prescaleDivider = 1; // The divider increases by powers of 2
  uint8_t   prescaleBitMask = 1; // The bitmask only increments unitarily
  uint32_t  ticksPerMs      = F_CPU / 1000;
  
  
  // The numbers in each if statement are the lowest frequency that can be achieved
  // with the prescale inside the if statement
  //
  // eg: the frequency can be viewed as a number of transitions equal to the 
  //     frequency * ([LOW]-transition-[HIGH])
  //     (we ignore the extra transition between cycles, happens anyway, err, I think, maybe, not sure)
  //     the longer the time between transitions the lower the frequency
  //     the maximum length we can get is 255 ticks
  //     the minimum length we can get is 1 tick
  //
  //     therefore minimum is CPU_FREQ / Prescale / 255
  //               maximum is CPU_FREQ / Prescale / 1
  
  // We will add/remove 1 hz to account for rounding here
  
  #define TONE_MIN_FREQ_FOR_PRESCALE(PS) (((F_CPU / PS) / 255) + 1)
  #define TONE_MAX_FREQ_FOR_PRESCALE(PS) (((F_CPU / PS) / 1)   - 1)
  
  #define TONE_RANGE_ACTIVE_FOR_PRESCALE(PS) ( TONE_MAX_FREQ >= TONE_MIN_FREQ_FOR_PRESCALE(PS) && TONE_MIN_FREQ <= TONE_MAX_FREQ_FOR_PRESCALE(PS) )
    
  if(TONE_RANGE_ACTIVE_FOR_PRESCALE(1) && frequency >= TONE_MIN_FREQ_FOR_PRESCALE(1))     // (F_CPU / 1) / 255
  {
    prescaleDivider = 1;
    ticksPerMs      = F_CPU / 1000;
    prescaleBitMask = 1;
  }
  else if(TONE_RANGE_ACTIVE_FOR_PRESCALE(8) && frequency >= TONE_MIN_FREQ_FOR_PRESCALE(8))  // (F_CPU / 8) / 255
  {
    prescaleDivider = 8;
    ticksPerMs      = (F_CPU/8) / 1000;
    prescaleBitMask = 2;
  }      
  else if(TONE_RANGE_ACTIVE_FOR_PRESCALE(64) && frequency >= TONE_MIN_FREQ_FOR_PRESCALE(64))  // (F_CPU / 64) / 255
  {
    prescaleDivider = 64;
    ticksPerMs      = (F_CPU/64) / 1000;
    prescaleBitMask = 3;
  }  
  else if(TONE_RANGE_ACTIVE_FOR_PRESCALE(256) && frequency >= TONE_MIN_FREQ_FOR_PRESCALE(256))  // (F_CPU / 256) / 255
  {
    prescaleDivider = 256;
    ticksPerMs      = (F_CPU/256) / 1000;
    prescaleBitMask = 4;
  }      
  else if(TONE_RANGE_ACTIVE_FOR_PRESCALE(1024)  && frequency >= TONE_MIN_FREQ_FOR_PRESCALE(1024))  // (F_CPU / 1024) / 255
  {
    prescaleDivider = 1024;
    ticksPerMs      = (F_CPU/1024) / 1000;
    prescaleBitMask = 5;
  }
  
  toneRaw(pin, (((F_CPU / prescaleDivider) / frequency) / 2) - 1, length ? length * ticksPerMs : ~(0UL), prescaleBitMask);
}

```

MCUdude, 2018-07-08:
OK, I've done some more research, and I've come up with a function that works way better than the original one. It's not perfect though.
`tone(4, 130);` gives me 126.7 Hz at 9.6 MHz and 129.8 Hz at 8 MHz
`tone(4, 2093);` gives me 2142 at 9.6 MHz and 2155 Hz at 8 MHz

I've used my Rigol DS1054Z to measure the frequency and my Rigol DG1022 to output an exact clock reference for the ATtiny13.

It would be awesome if you'd give it a try and see if there's something that can be improved even further!

Here's the "new and improved" version:
```c
void tone(const uint8_t pin, const uint16_t frequency, const uint32_t length)
{
  if(frequency == 0) 
    return noTone(pin);

  uint16_t prescaleDivider = 1; // The divider increases by powers of 2
  uint8_t  prescaleBitMask = 1; // The bitmask only increments unitarily
  uint32_t ticksPerMs      = F_CPU / 1000;
  uint8_t  subtract        = 0;
  
  // The numbers in each if statement are the lowest frequency that can be achieved
  // with the prescale inside the if statement
  //
  // eg: the frequency can be viewed as a number of transitions equal to the 
  //     frequency * ([LOW]-transition-[HIGH])
  //     (we ignore the extra transition between cycles, happens anyway, err, I think, maybe, not sure)
  //     the longer the time between transitions the lower the frequency
  //     the maximum length we can get is 255 ticks
  //     the minimum length we can get is 1 tick
  //
  //     therefore minimum is CPU_FREQ / Prescale / 255
  //               maximum is CPU_FREQ / Prescale / 1
  
  // We will add/remove 1 hz to account for rounding here
  
  #define TONE_MIN_FREQ_FOR_PRESCALE(PS) (((F_CPU / PS) / 255) + 1)
  #define TONE_MAX_FREQ_FOR_PRESCALE(PS) (((F_CPU / PS) / 1)   - 1)
  
  #define TONE_RANGE_ACTIVE_FOR_PRESCALE(PS) ( TONE_MAX_FREQ >= TONE_MIN_FREQ_FOR_PRESCALE(PS) && TONE_MIN_FREQ <= TONE_MAX_FREQ_FOR_PRESCALE(PS) )
    
  if(TONE_RANGE_ACTIVE_FOR_PRESCALE(1))     // (F_CPU / 1) / 255
  {
    prescaleDivider = 1;
    subtract        = 0;
    ticksPerMs      = F_CPU / 1000;
    prescaleBitMask = 1;
  }
  else if(TONE_RANGE_ACTIVE_FOR_PRESCALE(8) && frequency >= TONE_MIN_FREQ_FOR_PRESCALE(8))  // (F_CPU / 8) / 255
  {
    prescaleDivider = 8;
    subtract        = 0;
    ticksPerMs      = (F_CPU/8) / 1000;
    prescaleBitMask = 2;
  }      
  else if(TONE_RANGE_ACTIVE_FOR_PRESCALE(64) && frequency >= TONE_MIN_FREQ_FOR_PRESCALE(64))  // (F_CPU / 64) / 255
  {
    prescaleDivider = 64;
    subtract        = 2;
    ticksPerMs      = (F_CPU/64) / 1000;
    prescaleBitMask = 3;
  }  
  else if(TONE_RANGE_ACTIVE_FOR_PRESCALE(256) && frequency >= TONE_MIN_FREQ_FOR_PRESCALE(256))  // (F_CPU / 256) / 255
  {
    prescaleDivider = 256;
    subtract        = 0;
    ticksPerMs      = (F_CPU/256) / 1000;
    prescaleBitMask = 4;
  }      
  else if(TONE_RANGE_ACTIVE_FOR_PRESCALE(1024))  // (F_CPU / 1024) / 255
  {
    prescaleDivider = 1024;
    subtract        = 0;
    ticksPerMs      = (F_CPU/1024) / 1000;
    prescaleBitMask = 5;
  }
  else
    return;
  
  toneRaw(pin, (((F_CPU / prescaleDivider) / frequency) / 2) - subtract, length ? length * ticksPerMs : ~(0UL), prescaleBitMask);
}

```

sleemanj, 2018-07-08:
Thinking aloud...

  * ignoring range limit, 1000 Hz at 9.6MHz should select 64 prescale according to my long forgotten code
  * At 64 prescale there are 9.6M/64 = 150k ticks per second, 150 ticks per mS
  * At 1000Hz, each cycle is 1mS, 150 ticks
  * If the midpoint is ` ((F_CPU / prescaleDivider) / frequency)` then that would make it 150, and so 300 ticks per cycle
  * ... yeah I must be off by double somehow... in my defence I only really intended this for playing Mario or whatever ;-)
 * double check with range limit taken into account
  * with existing (double freq) code then each prescale has these min/max @ 9.6MHz
    *   1: 37.6k -> 9.6M
    *   8: 4.7k -> 1.2M 
    *  64: 588 -> 150k 
    * 256: 147 -> 37.5k
    * 1024: 36 -> 9374
  * looking only at the default range enabled (130 -> 2093Hz), that would mean that only the 1024 PS is used
  * 9.6M/1024 = 9375 ticks/second, 9.375/mS
  * ` ((F_CPU / prescaleDivider) / frequency) == 9` so 18 ticks per cycle, yeah, definitely double
  * the prescale selection should be fixed as well as the doubled midpoint
  * each prescale frequency range needs to be double, this does not result in any actual change for the default frequency limit of 130 -> 2390Hz though

Summary:  looks like I goofed and didn't divide by 2, so the proposed division is correct for the primary fix, I think that `TONE_MIN_FREQ_FOR_PRESCALE` and `TONE_MAX_FREQ_FOR_PRESCALE` should be doubled also,,,

    #define TONE_MIN_FREQ_FOR_PRESCALE(PS) ((((F_CPU / PS) / 255) + 1)*2) 
    #define TONE_MAX_FREQ_FOR_PRESCALE(PS) ((((F_CPU / PS) / 1)   - 1)*2)

as for the tweaking adjustment (`subtract`), I think this is only adjusting for quantization issues and that it would be highly frequency dependant, so I don't know that selection of hard values there is worthwhile unless there is something else going on causing a constant "N ticks offset" in the 64 divisor range.

MCUdude, 2018-07-09:
@sleemanj is there any reason why it shouldn't automatically select the highest prescaler value possible, to gain extra resolution? (I think) 1000 Hz would be more accurate with a prescaler of 1024 rather than 64.

Other than this, is there anything that can be done in order to make tone() more accurate?

sleemanj, 2018-07-09:
> @sleemanj [1] is there any reason why it shouldn't automatically select the highest prescaler value possible, to gain extra resolution? (I think) 1000 Hz would be more accurate with a prescaler of 1024 rather than 64.

The range is limited so when you are using non-constant frequency your
code size doesn't blow up having to do all those tests every time you
call tone, those tests are inline so every time you call tone with
non-constant frequency the code to test them is duplicated.  On the t13
there isn't much space to spare so every saved byte counts.   

You can certainly change the min/max or make it configurable which will 
pick a better range dependant on frequency, but at the expense of code
space (for non constant frequency uses).   

Certainly possible my logic is screwy and it's limiting to sub optimal
prescaler(s) but that's the idea anyway. As I say,  made the code for
blurting out simple tunes and such, the majority of people using tone
are not going to need super accuracy. 

 

Links:
------
[1] https://github.com/sleemanj
[2]
https://github.com/MCUdude/MicroCore/issues/65#issuecomment-403579493
[3]
https://github.com/

closed 2018-07-13 by MCUdude

theduskin, 2023-10-31:
i found this:
http://www.avr-asm-tutorial.net/avr_en/micro_beginner/9_AudioGenerator/9_AudioGenerator.html#9_tonecontrol
saying at 9.3.2:
"It is already clear that the CTC mode of the timer has to be used here. "

to me, nothing is clear, because that's out of my scope. 
it seems normal mode is used by tone()?
so maybe its worth to have a look at it.

commit 3b27216a (2018-07-13): Fix tone()
The tone() frequency was way off. This makes it much closer to the actual value. It's not perfect, but does the job if you want to play the Super mario theme. Closes #65