MCUdude/MicroCore #66: Seems to be compiling at 1.2Mhz even though 9.6 selected

pman92, 2018-07-05:
I'm having problems with a code generator I'm trying to build on a attiny13a. 
I designed code on an arduino UNO and tested, and it worked fine. 
Since attempting to port to attiny, I have improved the code (and retested on uno) but still cannot get success on attiny13a. 
I'm using an arduino as ISP and a TL886 programmer to program/read the attiny. I've tried multiple chips with identical results. 

After lots of pulling my hair out I believe I've found whats going wrong. When I hit "burn bootloader" in arduino IDE (with settings 9.6mhz, 2.7v BOD, LTO enabled), fuses are burnt as follows:
Lfuse: 3A
Hfuse: FB
Lock: 3F
From the information I can find this is correct for 9.6mhz. 

Next, without disconnecting anything or changing any settings, I select "upload using programmer". avrdude reports upload success. However with an oscilloscope I can see the program is not running correctly (the output of the "data" pin for my code generator is garbage). 

Next, I remove the attiny IC (which is on a breadboard) and read it with my TL866. This confirms the fuses are set correctly (as above) for 9.6mhz. 

Here is the strange part. If I now set the CKDIV8 bit (making lfuse 2A), and reburn the chip without changing anything else, the code runs fine (see oscilloscope trace). Although I now assume its running at 1.2Mhz, and this is why the output is slightly slow (the code is too fast for 1.2mhz? ). 
I can achieve the same thing by burning bootloader with 1.2mhz selected, then changing to 9.6 and uploading.

I believe the problem is the code is being compiled for 1.2mhz when I upload it with arduino as ISP (even though 9.6 is selected in the menu).

I've attached a zip with my arduino sketch and the hex file arduino ide outputs when I select export compiled binary (with 9.6mhz selected), a word document showing everything arduino IDE shows when burning bootloader (at 9.6mhz) and compiling and uploading (again at 9.6mhz), a full binary read of the chip from my TL866 after it has been programmed, and an oscilloscope trace showing the output from the attiny afetr I have changed the fuse and set ckdiv8 (note its 12v and inverted to what the actual attiny output is because the 5v attiny pin is driving a transistor). 

If this doesn't seem to happen to anyone else, could someone compile at 9.6mhz and send me a copy of the hex/bin so I can burn it to my chips to test if that works 

[FOUND PROBLEM.zip](https://github.com/MCUdude/MicroCore/files/2165626/FOUND.PROBLEM.zip)

MCUdude, 2018-07-06:
Thanks for the very detailed explanation and isolation of the problem! It just makes it so much easier for me to help you out when all the non relevant stuff is gone.
I just looked through your code and uploaded the hex file you provided. My fuses are: *E:FF, H:FB, L:3A*

First the ATtiny13 was outputting garbage, but with a quick HW fix I got it working just fine. 

The problem is not your code, it's your circuity. The code waits for the pin to go high, and while it waits the pin is floating; picking up all sorts of noise from its surroundings.
What I did to fix this was to add a 10k resistor from ground to pin 3 (PB3). This makes sure the pin is held to ground when the button isn't pressed. 

If you don't want to add any external components you can use the internal pullup resistor on pin 3. This way you'll need to pull pin3 low in order for the T13 to output its data. See code below

``` c

#define DATA_RATE 1000 //microseconds between data bits
#define DATA_LENGTH 4 //length of data packet (in bytes)
#define PIN_OUTPUT 4 //Output pin to drive data line
#define PIN_INPUT 3 //Input pin pulled low to disable data
#define LOOPS_TO_DISABLE 200 //how many consecutive low input loops to disable data
const byte data[] = {0b10101011, 0b00101011, 0b00110010, 0b11001100}; //from first transmitted to last
byte currentBit = 8; //Start at MSB first
byte currentByte = 0;
unsigned long lastDataTime = 0;
byte inputState;
byte inputFrames = 0;

void setup() {
  // put your setup code here, to run once:
  pinMode(PIN_OUTPUT, OUTPUT);
  pinMode(PIN_INPUT, INPUT_PULLUP);
  lastDataTime = micros();
}

void loop() {
  // put your main code here, to run repeatedly:
  inputState = digitalRead(PIN_INPUT);
  if (inputState == LOW) { 
    //OK to transmit
    inputFrames = 0;
    if (micros() > (lastDataTime + DATA_RATE)) {
      //Time for next bit
      lastDataTime = micros();
      digitalWrite(PIN_OUTPUT, bitRead(data[currentByte], currentBit-1));      
      currentBit--;
      if (currentBit < 1) {
        //Reached the end of this byte
        currentBit = 8;
        currentByte++;
        if (currentByte >= DATA_LENGTH) {
          //reached the end of complete data packet
          currentByte = 0;
        }
      }
    }
   } else {
    inputFrames++;
    if(inputFrames >= LOOPS_TO_DISABLE){
      digitalWrite(PIN_OUTPUT, LOW);
      lastDataTime = micros();
      inputFrames = LOOPS_TO_DISABLE;
    }
  }
}
```

It's true that the array of 32 bytes should last for 32 ms, but in most cases this won't happen when using the internal oscillator. Mine is actually running at ~9.1 MHz, resulting a time of roughly 36ms instead of 32. You'll only get accurate timing if you use an external clock source. Sadly the T13 is not able to drive a crystal oscillator by itself, so you need a dedicated clock generator. You can build a generator like this yourself by using a 74HC14 and a few passive components:
[image]

Here are some scope screenshots just for reference. The first one is when digital pin 3 on the T13 is floating. The output signal doesn't match the "pattern" at all.
[image]

The second screenshot is when digital pin 3 is pulled low by a 10k resistor, while the button pulls the pin to 5V.
[image]

I'm sure this works without a resistor at 1.2 MHz because a lower clock frequency often leads to less noise to be picked up.

pman92, 2018-07-07:
Hi again. Thanks so much for your help.
I should of mentioned I have had a physical pull-up resistor connected to pin 3/PB3 during testing. That doesn't seem to be an issue. 
For me the code will only run with lfuse set to 2A (1.2mhz). With 3A (9.6mhz) set it runs sporadically and stops running completely at times. 
If the hex I uploaded runs perfectly at 9.6mhz for you, and runs the same (around 36ms for a cycle) at "1.2mhz" for me, my chips are faulty. I'm thinking it wouldn't really be 36ms for me if it actually was 1.2mhz if you have confirmed its around 36ms at 9.6mhz. 
Maybe my chips are running at 9.6mhz with the 1.2mhz fuse set, and with 9.6 fuses set they aren't running properly at all (faulty chips). I bought them cheap from china so they may be faulty fakes. 

I'll do some testing to confirm and let you know how I go.
Thanks again

pman92, 2018-07-08:
Ok so I've had another look and it seems my chips are failing at 9.6mhz. 

I can have them set to 1.2 or 4.8mhz, and the code runs as expected (at approx half or 1/8 the speed). 
As soon as I set the fuses to 9.6mhz everything falls apart, the code doesn't run as expected at all (it outputs rubbish and usually after a few seconds completely stops outputting all together). 

I think I will look into using an attiny45/85 for my application, because then I can get an accurate clock with just an external crystal. I don't have a lot of space for external components.

Thanks again for all your help. I

closed 2018-07-08 by pman92