MCUdude/MicroCore #68: {runtime.tools.avr-gcc.path}/bin/avr-g++: no such file or directory

gentunian, 2018-07-18:
Arduino version 1.8.5

The code compiles OK for others boards. Not sure why this path is not being recognized in `platform.txt`

```
# Default "compiler.path" is correct, change only if you want to overidde the initial value
compiler.path={runtime.tools.avr-gcc.path}/bin/
```

Distro is fedora 27 and avr gcc path is: 
```
/usr/lib64/ccache/avr-gcc
/usr/lib64/ccache/avr-g++
```

Setting to that path compiles but with a lot of errors, compiling [this code from sparkfun](https://github.com/sparkfun/MMA8452_Accelerometer/blob/master/Firmware/MMA842Q%20Sketches/MMA8452Q_AdvancedExample/MMA8452Q_AdvancedExample.ino):
<details>
<summary>Error output</summary>

```c
Build options changed, rebuilding all
In file included from /home/seba/Arduino/sketch_jul18a/sketch_jul18a.ino:40:0:
sketch/i2c.h: In function 'void i2cInit()':
i2c.h:102: error: 'TWCR' was not declared in this scope
  sbi(TWCR, TWEN); // Enable TWI
      ^
sketch/i2c.h:53:28: note: in definition of macro 'sbi'
 #define sbi(var, mask)   ((var) |= (uint8_t)(1 << mask))
                            ^~~
sketch/i2c.h:102:6: note: suggested alternative: 'EECR'
  sbi(TWCR, TWEN); // Enable TWI
      ^
sketch/i2c.h:53:28: note: in definition of macro 'sbi'
 #define sbi(var, mask)   ((var) |= (uint8_t)(1 << mask))
                            ^~~
i2c.h:102: error: 'TWEN' was not declared in this scope
  sbi(TWCR, TWEN); // Enable TWI
            ^
sketch/i2c.h:53:51: note: in definition of macro 'sbi'
 #define sbi(var, mask)   ((var) |= (uint8_t)(1 << mask))
                                                   ^~~~
sketch/i2c.h:102:12: note: suggested alternative: 'ADEN'
  sbi(TWCR, TWEN); // Enable TWI
            ^
sketch/i2c.h:53:51: note: in definition of macro 'sbi'
 #define sbi(var, mask)   ((var) |= (uint8_t)(1 << mask))
                                                   ^~~~
sketch/i2c.h: In function 'void i2cSetBitrate(short unsigned int)':
i2c.h:110: error: 'TWSR' was not declared in this scope
  cbi(TWSR, TWPS0);
      ^
sketch/i2c.h:54:28: note: in definition of macro 'cbi'
 #define cbi(var, mask)   ((var) &= (uint8_t)~(1 << mask))
                            ^~~
sketch/i2c.h:110:6: note: suggested alternative: 'ACSR'
  cbi(TWSR, TWPS0);
      ^
sketch/i2c.h:54:28: note: in definition of macro 'cbi'
 #define cbi(var, mask)   ((var) &= (uint8_t)~(1 << mask))
                            ^~~
i2c.h:110: error: 'TWPS0' was not declared in this scope
  cbi(TWSR, TWPS0);
            ^
sketch/i2c.h:54:52: note: in definition of macro 'cbi'
 #define cbi(var, mask)   ((var) &= (uint8_t)~(1 << mask))
                                                    ^~~~
sketch/i2c.h:110:12: note: suggested alternative: 'ADPS0'
  cbi(TWSR, TWPS0);
            ^
sketch/i2c.h:54:52: note: in definition of macro 'cbi'
[... 424 lines cut ...]
                            ^~~
/home/seba/Arduino/sketch_jul18a/sketch_jul18a.ino:346:7: note: suggested alternative: 'EECR'
   cbi(TWCR, TWEN);  // Disable TWI
       ^
sketch/i2c.h:54:28: note: in definition of macro 'cbi'
 #define cbi(var, mask)   ((var) &= (uint8_t)~(1 << mask))
                            ^~~
sketch_jul18a:346: error: 'TWEN' was not declared in this scope
   cbi(TWCR, TWEN);  // Disable TWI
             ^
sketch/i2c.h:54:52: note: in definition of macro 'cbi'
 #define cbi(var, mask)   ((var) &= (uint8_t)~(1 << mask))
                                                    ^~~~
/home/seba/Arduino/sketch_jul18a/sketch_jul18a.ino:346:13: note: suggested alternative: 'ADEN'
   cbi(TWCR, TWEN);  // Disable TWI
             ^
sketch/i2c.h:54:52: note: in definition of macro 'cbi'
 #define cbi(var, mask)   ((var) &= (uint8_t)~(1 << mask))
                                                    ^~~~
exit status 1
'TWCR' was not declared in this scope
```
</details>

Two things here, how to avoid editing the `platform.txt` so it would work out of the box and I'm not sure why changing the board I'm getting those errors. Any help? :)

MCUdude, 2018-07-18:
Seems like the compiler can't find avr-libc. I don't have much experience with Arduino IDE on linux, but maybe @per1234 may be able to help?

per1234, 2018-07-18:
With the official Arduino IDE, when you install MicroCore via Boards Manager, any necessary parts of the toolchain will also be installed. This error means something went wrong during that process (error on the part of the Arduino IDE, not MicroCore).

To fix this, you need to take some drastic action. Be aware this is somewhat advanced and will remove any boards you have installed via Boards Manager:

1. Click the link on the line following File > Preferences > More preferences can be edited directly in the file. This will open the `Arduino15` (or similar name depending on OS) folder.
2. Delete all files and folders under the Arduino15 folder except for preferences.txt. Please be very careful when deleting things from your computer. When in doubt, back up!
3. Restart the Arduino IDE.
4. Reinstall MicroCore.
5. Verify you can now use MicroCore before installing any other boards via Boards Manager.

The Arduino IDE does not use your existing installation of avr-gcc. It installs its own version.

Make sure to use the official Arduino IDE downloaded from:
https://www.arduino.cc/en/Main/Software
If you're using some version of the IDE from a package manager that's been modified in unspecified ways by some random then it's anyone's guess what's wrong.

gentunian, 2018-07-19:
@per1234 removing everything except `preferences.txt` and installing it again didn't work. Installation process does not throw any error.

Compiling error:
```
Build options changed, rebuilding all
fork/exec {runtime.tools.avr-gcc.path}/bin/avr-g++: no such file or directory
Error compiling for board ATtiny13.
```

clytle374, 2019-10-07:
Sorry to resurrect a dead thread, but I'm pretty sure I'm in the exact same spot as per1234 was.  I've deleted the contents of the directory and it didn't help as well. 

I'm on Gentoo Linux and not using the Arduino toolchain.  Yes it is an option, but I'd like to avoid it on my system.  The IDE is from Arduino.  

I can build for the default boards without any trouble.  As soon as I try any from the mightycore list I get the same error.

> Arduino: 1.8.7 (Linux), Board: "ATmega644, Yes (UART0), Standard pinout, 644P / 644PA, BOD 2.7V, Disabled, External 20 MHz"
> 
> /usr/share/arduino/arduino-builder -dump-prefs -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/cory/.arduino15/packages -tools /usr/share/arduino/hardware/tools/avr -tools /home/cory/.arduino15/packages -libraries /home/cory/Arduino/libraries -fqbn=MightyCore:avr:644:bootloader=uart0,pinout=standard,variant=modelP,BOD=2v7,LTO=Os,clock=20MHz_external -ide-version=10807 -build-path /tmp/arduino_build_352307 -warnings=none -build-cache /tmp/arduino_cache_931693 -prefs=build.warn_data_percentage=75 -verbose /home/cory/data/DCloadDIY/testZero/testZero.ino
> /usr/share/arduino/arduino-builder -compile -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/cory/.arduino15/packages -tools /usr/share/arduino/hardware/tools/avr -tools /home/cory/.arduino15/packages -libraries /home/cory/Arduino/libraries -fqbn=MightyCore:avr:644:bootloader=uart0,pinout=standard,variant=modelP,BOD=2v7,LTO=Os,clock=20MHz_external -ide-version=10807 -build-path /tmp/arduino_build_352307 -warnings=none -build-cache /tmp/arduino_cache_931693 -prefs=build.warn_data_percentage=75 -verbose /home/cory/data/DCloadDIY/testZero/testZero.ino
> Using board '644' from platform in folder: /home/cory/.arduino15/packages/MightyCore/hardware/avr/2.0.3
> Using core 'MCUdude_corefiles' from platform in folder: /home/cory/.arduino15/packages/MightyCore/hardware/avr/2.0.3
> Detecting libraries used...
> "/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega644p -DF_CPU=20000000L -DARDUINO=10807 -DARDUINO_AVR_ATmega644 -DARDUINO_ARCH_AVR   "-I/home/cory/.arduino15/packages/MightyCore/hardware/avr/2.0.3/cores/MCUdude_corefiles" "-I/home/cory/.arduino15/packages/MightyCore/hardware/avr/2.0.3/variants/standard" "/tmp/arduino_build_352307/sketch/testZero.ino.cpp" -o "/dev/null"
> Generating function prototypes...
> "/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega644p -DF_CPU=20000000L -DARDUINO=10807 -DARDUINO_AVR_ATmega644 -DARDUINO_ARCH_AVR   "-I/home/cory/.arduino15/packages/MightyCore/hardware/avr/2.0.3/cores/MCUdude_corefiles" "-I/home/cory/.arduino15/packages/MightyCore/hardware/avr/2.0.3/variants/standard" "/tmp/arduino_build_352307/sketch/testZero.ino.cpp" -o "/tmp/arduino_build_352307/preproc/ctags_target_for_gcc_minus_e.cpp"
> fork/exec /bin/avr-g++: no such file or directory
> Error compiling for board ATmega644.

Is the compiler being called differently when using the mightycore boards?  If so where is this being configured?  Any other suggestions?
Thanks
Cory

bigbn, 2020-07-25:
@gentunian try this:

Tools > Board > Boards Manager
Wait for downloads to finish
Click on "Arduino AVR Boards"
From the dropdown menu select "1.6.20"
Click "Install"
Wait for installation to finish
Click "Close"
Try compiling again

according to this https://forum.arduino.cc/index.php?topic=547545.0

This steps has resolved this issue for me (archlinux, fresh arduino IDE install)

gentunian, 2020-07-26:
thanks for the update, I didn't test it because I've dropped the project for now. Will keep this as a reference!

znoxx, 2020-07-29:
@bigbn, thanks for tip, worked for Manjaro

closed 2020-09-08 by MCUdude