MCUdude/MicroCore #69: my atmega328 (uno)PWM code dont work with tiny

majidmp97, 2018-09-19:
hello
my cod work in atmega328 uno but i try to work with your micro core for tiny 13 but i have error register and don't work
`#include "TimerOne.h"               // include TimerOne.h
//**************************temp*****************
#include <avr/wdt.h>
const int analogInPin = A0;  // Analog input pin that the potentiometer is attached to
int sensorValue = 0;        // value read from the pot
//*******************************************
int dutycycle = 0;
int dutycycle1 = 0;               // Initailize duty cylce variable as integer data type
int dutycycle2 = 0;
int f = 0;
//int button2 = 0;
//int button1 = 0;
int ledR = 5;
int ledG = 6;
int ledB = 7;
int conter = 0;

void setup()               // Setup function
{
  wdt_disable();
  wdt_enable(WDTO_4S);
  //**************************temp*****************
  Serial.begin(115200);
 
  pinMode (9, OUTPUT);             // set pin 9 as an output pin for pwm
  pinMode (10, OUTPUT);            // set pin 10 as an output pin for pwm
  pinMode (13, OUTPUT);            // set pin 10 as an output relay
 
  pinMode(5, OUTPUT);//reed
  pinMode(6, OUTPUT);//green
  pinMode(7, OUTPUT);//blue
  digitalWrite(ledR, HIGH);
  digitalWrite(ledG, HIGH);
  digitalWrite(ledB, HIGH);

  Timer1.initialize(20000);  //20000           // Initailize timer1 time period as 20 milli second (50 Hz frequency)
  //Timer1.attachInterrupt(loop);      // battery_voltage_measurement function will be executed every 20 milli second using timer 1 overflow interrupt
  TCCR1A = (TCCR1A & 0x0F) | 0xB0 ;             // set pin 10 inverted of pin 9
  wdt_reset();

f = 18634;
  dutycycle = (0) ;  // 0-100% = duty cycle
  dutycycle1 = map(dutycycle, 0, 100, 0, 1024);
  dutycycle2 = map(dutycycle, 0, 100, 1024, 0);
  dutycycle1 = map(dutycycle, 0, 100, 0, 1024);
  dutycycle2 = map(dutycycle, 0, 100, 1024, 0);
  Timer1.pwm(9, dutycycle1, 1000000 / f * 1);        // Timer1.pwm function takes argument as (pin no. , dutycycle , time period)
  Timer1.pwm(10, dutycycle2, 1000000 / f * 1);
  wdt_reset();
 
}

//////////////////////////////////////////////////////
void loop()              // loop function starts
{
 wdt_reset();
  f = 632;//1135
  dutycycle = (11) ;  // 0-100% = duty cycle 1 a sw 3-4 hzspark
  dutycycle1 = map(dutycycle, 0, 100, 0, 1024);
  dutycycle2 = map(dutycycle, 0, 100, 1024, 0);
  Timer1.pwm(9, dutycycle1, 1000000 / f * 1);        // Timer1.pwm function takes argument as (pin no. , dutycycle , time period)
  Timer1.pwm(10, dutycycle2, 1000000 / f * 1);

  delay(1000);
  wdt_reset();

  }

`

beicnet, 2018-09-19:
Your code will never fit in ATtiny13 anyway, also it's not gonna work because of your Pin numbering.

MCUdude, 2018-09-19:
* ATtiny13 doesn't have a hardware serial port. Serial.print will never work.
* ATtiny13 only has one timer; timer0. Timer1 will never work
* ATtiny13 only has 5 IO pins you can use.

You can't just expect any program to run on the ATtiny13. In most cases it needs to be written spesifically for the hardware. Remember it only has 1k of flash and 64B of RAM. If you need help porting your code please use the Arduino forum instead.

closed 2018-09-19 by MCUdude