MCUdude/MicroCore #71: Suggestion: Add "Preserve EEPROM through erase" menu

bratoff, 2018-11-28:
The released core always programs "preserve eeprom", which is not always desirable.  Please see attached boards.txt for a suggested way to make this another menu option:

[boards.txt](https://github.com/MCUdude/MicroCore/files/2622863/boards.txt)

Note - this boards.txt also modifies the BOD and LTO menu texts so they display more meaningfully on the status line.

maxint-rd, 2020-07-18:
When I found out that the EEPROM of my ATtiny13A was erased every time I flashed it, I quickly found this post. I tested it and it works great for me.

Now I can make sketches that use the EEPROM without losing its contents after programming the MCU.
(Badly needed as I'm currently working on an I2C-peripheral that allows the I2C address to be changed and stored to survive resets).

+1 in support of incorporating this option in the otherwise fabulous MicroCore.

MCUdude, 2020-07-18:
It was added in a411871, but I haven't done a new release yet. If you need it right now, you can do a manual install

maxint-rd, 2020-07-18:
Hi @MCUdude , Thank you. I've already done that.

Thanks btw for providing your cores. I'm happily enjoying them in my experiments with various Atmel chips. Great work!

closed 2020-11-21 by MCUdude