MCUdude/MicroCore #78: If statement with two conditions

opinto1, 2019-04-08:
Hello,
I have a project where I need to use a If statement with two conditions.Something like this:
`if (number > 1486 && number < 1496)`

But when I upload the code to the attiny13a and try it the project dosen't works so I have tried a if statement only with one condition and it have worked well

Is possible to solve this problem in the next MicroCore Update?  Because without a if statement with two conditions my project is useless

MCUdude, 2019-12-12:
Sorry for the incredibly late reply. Can you post your entire code? if conditions have worked since day 1, so I'm sure it all comes down to your code.

MCUdude, 2019-12-22:
Please re-open if you believe you still facing the same issue and I'll help you out. Currently, I believe there's an error in your code.

closed 2019-12-22 by MCUdude

opinto1, 2020-01-07:
Hello sorry for the late responce but my code is this one

`int RXin = 2; //Recetor
int AUXout = 1; // LED

unsigned long number;
//int signal = 0;

void setup()
{

  pinMode(RXin, INPUT);
  pinMode(AUXout, OUTPUT);

}

void loop()
{
  number = pulseIn(RXin, HIGH);

   if (number <= 983)
   {
    digitalWrite(AUXout, LOW);
   }

  else if (number > 1486 && number < 1496)
   {
     digitalWrite(AUXout, HIGH);
   }

  else if (number >= 1999 && number <= 2005)
   {
     digitalWrite(AUXout, HIGH);
     delay(1000);
     digitalWrite(AUXout, LOW);
     delay(1000);
   }
}`

MCUdude, 2020-01-11:
It turns out that `shiftIn()` is very inaccurate, so the expected number is way out compared to what you would expect it to be. I used a pulse generator to test this.

I'll see what I can do about this.