MCUdude/MicroCore #79: Something wrong with variable memory allocation

harrylisby, 2019-05-30:
Something seems odd with data types for this implementation, today I wrote a simple program ~80 lines of code for my ATTINY13A. And suddenly I can't compile my code anymore because I'm over the maximum flash memory inside the microcontroller.

I made some checks and found out that variables are using way more bytes than what actually should be used. I tried with int32_t, long, int8_t and int. All of them seem to be using WAY more than they should.

Here a few screenshots proving the issue, all I did was changing the variable type in line 44  and compiling the software again:

[image]

[image]

[image]

MCUdude, 2019-05-30:
Please post your code so I can check it myself.

harrylisby, 2019-05-30:
`#define IR_OUT   1   //IR LED output2
#define RLAY_OUT 0   //Relay ON/OFF output
#define ANLG_IN  A1
#define SENS_ADJ A2
#define TIME_ADJ A3

#define ON_TIME 10000   //On time for output pin (in microseconds)
#define RUN_SPD 500     //Per-cycle time (in microseconds)
#define ADJ_SPD 2000

#define PULSES 3

uint16_t sensitivity;
uint16_t nonMapSensitivity;
bool once=true;
uint8_t randTime;
uint8_t counter;

int32_t mainTime,timer,mainTimeLast,adjTimeLast,offDelay,nonMapTimer;

void setup() {

  DDRB = _BV(0) | _BV(1);

  // PORTB |= _BV(1);
  // _delay_ms(1000);
  // PORTB &= ~(_BV(1));

}

void loop() {
  mainTime=millis();

  if(once){
    randTime = (analogRead(ANLG_IN)%10)+1;
    once=false;
  }

  if((mainTime-adjTimeLast)>ADJ_SPD){ //Executes on every ADJ_SPD ms
    adjTimeLast = mainTime;
    nonMapSensitivity = analogRead(SENS_ADJ);
    sensitivity = nonMapSensitivity/5;//map(nonMapSensitivity,0,1023,0,1023);
    nonMapTimer = analogRead(TIME_ADJ);
    timer = map(nonMapTimer,0,1023,ON_TIME,0);
  }

  if((mainTime-mainTimeLast)>RUN_SPD){ //Executes on every RUN_SPD ms
    mainTimeLast=mainTime;
    for (int i = 0; i < PULSES; i++) {
      PORTB |= _BV(1);
      delay(1);
      uint16_t preRead = analogRead(ANLG_IN);
      PORTB &= ~(_BV(1));
      delay(randTime); //_delay_ms(randTime);
      uint16_t afterRead = abs(preRead-analogRead(ANLG_IN));

      if(afterRead>sensitivity){
        counter++;
        if(counter>PULSES)counter=0;
      }else{
        counter=0;
      }
    }

    if (counter == PULSES) {
      PORTB |= _BV(0);
      offDelay=mainTime+timer;
      counter=0;
    }else if(mainTime>offDelay){
      PORTB &= ~(_BV(0));
    }
  }
}`

MCUdude, 2019-05-30:
I understand what you mean, but that's just how the compiler works. I tried compiling the same code for Arduino UNO and is also have the same kind of memory increment when using int32_t instead of uint8_t.

harrylisby, 2019-05-30:
Does it need to be fixed, or is that intented? Or do you think there someone we can ask?

This is quite bad, more with this type of embedded systems that have very little flash space.

MCUdude, 2019-05-30:
> Does it need to be fixed, or is that intended? Or do you think there someone we can ask?

I don't know really. I'm compiling with I've tried compiling with avr-gcc 5.4.0 and 7.3.0. Seeing about the same memory usage on both. There's nothing I can do about this really. If you want to know more I'd probably ask a question over at the [avrfreaks forum](https://www.avrfreaks.net). Some of the guys over there are really hardcore!

harrylisby, 2019-05-30:
Alright, many thanks. Will be posting my question there and hope for some good news!

nerdralph, 2019-06-02:
" All of them seem to be using WAY more than they should."
A long uses 4 bytes.  When you declare 6 of them, that's 6 * 4 = 24bytes of
RAM used.  You have a few other variables in your program like sensitivity
and counter.  Add them all up, and they are using EXACTLY the amount of RAM
that they should.

MCUdude, 2019-06-02:
I think he reacted on the flash memory usage. It's more than doubled

nerdralph, 2019-06-03:
That's to be expected as well.  Since the AVR is an 8-bit MCU, it takes
four instructions to load four bytes from RAM into four registers when
using a long or uint32.  It also reduces the number of registers the
compiler can use internally, requiring even more code to save and restore
registers before and after function calls.

closed 2019-06-05 by MCUdude