MCUdude/MicroCore #80: aTTiny13A millis() no funciona MicroCore

DominicanMS, 2019-06-02:
All right, I have a week trying to solve a problem with Millis and with the delay function works as it should, now I want it to work with the multiservices of aTiny13A, I am using MicroCore and simulated in Proteus, physically and in Arduino one works

Dear colleagues in your time and reading, I am too green in Arduino.
MILLIS ()

I'm using google translator i do not speak much english

`const int ledPin =  0;      // the number of the LED pin
 
// Variables will change:
int ledState = LOW;             // ledState used to set the LED
long previousMillis = 0;        // will store last time LED was updated
  unsigned long currentMillis;
long interval = 1000;           // interval at which to blink (milliseconds)
 
void setup() {
  // set the digital pin as output:
  pinMode(ledPin, OUTPUT);      
}
 
void loop()
{

  currentMillis = millis();
 
  if(currentMillis - previousMillis > interval) {
    // save the last time you blinked the LED 
    previousMillis = currentMillis;   
 
    // if the LED is off turn it on and vice-versa:
    if (ledState == LOW)
      ledState = HIGH;
    else
      ledState = LOW;
 
    // set the LED with the ledState of the variable:
    digitalWrite(ledPin, ledState);
  }
}`

nerdralph, 2019-06-30:
I recognize one of my bitbang serial libraries.
byte c is unitialized, so TxByte(c) will print whatever unpredictable value c has.
I'm guessing you cut/paste some of my example code without paying too much attention to it.

I'd remove those two lines, and add a delay(500) to the loop to give plenty of time between loop iterations.  The value returned from millis() should go up by more than 500 each loop.

DominicanMS, 2019-07-04:
nerdralph greeting, the problem is not the serial communication but millis add the delay (1000) and keep sending 0 with micro () ago to flash a led as an example always and it works well and the values ​​sent by the serial port work correctly, I copied that example code.

if they have argun example that functions correctly with millis () I would appreciate it

I just managed to do it right by putting at the beginning of the void loop piniendo (time ++) replacing a millis ()

Discurpe is that I do not have much experience and I always have my project researching on the internet and I always try to find the most flexible way and take up less space

nnttoo, 2019-07-22:
I tried a few times, it seems that Millis really doesn't work, then I tried BOD disabled and it worked well

pastprimitive, 2019-10-14:
So I desperately need to get Millis() working with LTO enabled. I can confirm that Millis() does not work when LTO is enabled. Works as expected when disabled. Anyone make any progress on resolving why that might be? I'm going to poke around and see what I can figure out. Hoping to get some leads.

nerdralph, 2019-10-14:
I recall old versions of avr-gcc failing to link in millis.S when LTO was enabled.  Which Arduino IDE version are you using?
I just double-checked the compiled code with 1.8.9 under Windoze, which uses gcc-5.4.0.  Everything looks good (vector 8 points to the millis ISR) with and without LTO enabled.  I didn't test it on a physical t13 yet.

nerdralph, 2019-10-14:
I found one of my t13 chips without the DWEN fuse set, and tested a millis-based blink sketch.  It works find with and without LTO enabled.
`
#define LEDPIN 2

#include <avr/sleep.h>

unsigned millis16() { return millis() & 0xffff; }

void sleep_ms(unsigned ms) {
  unsigned snapshot = millis16();
  // SLEEP_MODE_IDLE is default
  //set_sleep_mode(SLEEP_MODE_PWR_DOWN);
  do {
    //sleep_enable();
    MCUCR = 1 << SE;
    sleep_cpu();
  } while (millis16() - snapshot <= ms);
}

void setup() {
  pinMode(LEDPIN, OUTPUT);
  //pinMode(4, OUTPUT);  // millis ISR debug
}

void loop() {
  sleep_ms(125);
  digitalWrite(LEDPIN, HIGH);

  sleep_ms(125);
  digitalWrite(LEDPIN, LOW);
}`

pastprimitive, 2019-10-14:
I actually just tried returning the boot loader to make sure my fuse settings were correct and that seemed to fix whatever was going on. Thank you so much for helping me trouble shoot it.

MCUdude, 2019-12-12:
Sorry for not being actively helping here. But thanks to Ralph it the problem looks to be solved!

closed 2019-12-12 by MCUdude