MCUdude/MicroCore #84: Fix Travis

MCUdude, 2019-09-02:
Travis fails not that I've added a dedicated millis/micros menu option. Could you look into it @per1234?

MCUdude, 2019-09-11:
Sorry to bother you @per1234. There's no hurry, it's just that (Travis) CI is definitely not my strength.

MCUdude, 2019-12-10:
Turned out it wasn't very difficult to fix this. Just added a few lines!

closed 2019-12-10 by MCUdude

MCUdude, 2020-06-11:
@per1234 I haven't heard from you in a while. Hope you're doing OK!

I have a Travis related question. The last month the CI has just timed out due to no output from the build command. Do you know why?

per1234, 2020-06-12:
Hi @MCUdude. Sorry I've been MIA. I haven't been doing much with my own CI script (though there have been quite a few releases since the version MicroCore is using), but I've been doing a lot of work lately on creating an official CI system for compilation testing of Arduino projects. The new system is targeted to the new GitHub Actions CI service rather than Travis CI:
https://github.com/arduino/actions/tree/master/libraries
There are still more things I want to do, but it's at the point where I feel it's a reasonable alternative to my personal CI script. GitHub Actions has some advantages over Travis CI. One big advantage of the new CI system is it uses Arduino CLI instead of Arduino IDE (which my script uses, since there was no Arduino CLI when I wrote it). Arduino CLI is much smaller than the Arduino IDE, which makes the CI builds faster.

---
The problem with MicroCore's CI build is caused by this breaking change to boards.txt:
https://github.com/MCUdude/MicroCore/commit/45a58675b5a42166aa82b63e1738f300fea82790
It changed the FQBN from `MicroCore:avr:attiny13` to `MicroCore:avr:13`, but the Travis CI configuration is still using the original FQBN. As you discovered, changing the FQBN can cause some confusion for anyone who is using MicroCore from a CLI (whether that's the Arduino IDE CLI or Arduino CLI), so if you're set on this change I recommend a major version bump on the next release and a note about it in the release notes.

This issue was exacerbated by the Arduino IDE's annoying behavior of trying to open the GUI when something goes wrong, which Travis CI's headless system doesn't support, so you just get the stalled build. With the recent IDE versions, it actually just does a nice failure with exit status 3 (Invalid argument for commandline option), which provides at least a clue as to what went wrong.

So the solution is to either change all the `MicroCore:avr:attiny13` in .travis.yml to `MicroCore:avr:13`, or else change boards.txt to use the original board ID of `attiny13`.

MCUdude, 2020-06-12:
Thanks for your reply! I may look into Github Actions later, or perhaps the official CI script sometime. But for now, I'll just patch this issue up and call it a day.

The reason why I changed from `attiny13` to `13` was to keep it uniform with my other cores. It should perhaps be documented, even though I'm not sure how may MicroCore users use the Arduino CLI. Personally, I prefer using PlatformIO when I'm working on a larger project with dependencies since the Platformio.ini file can store all hardware and library-related settings. I've also been involved in some of the newer PlatformIO AVR infrastructures, and I created a script that lets you set the fuses and burn the correct bootloader based on the parameters found in platformio.ini, that's now a part of the PIO atmelavr package. This, combined with the Intellisense feature VScode gives you makes my cores pretty darn great if I'm to say it myself! 😄 

MegaCoreX is also getting mature, but it could really need a CI (same with megaTinyCore). But that will have to be another time. I'm too busy with parenting at the moment!

MCUdude, 2020-08-02:
@per1234 Travis CI is somehow broken again. It seems like it'strying to insall the windows store version of the latest Arduino IDE. Any idea how to fix this?