MCUdude/MicroCore #85: Pinout

Ehsan-Shahnazi, 2019-09-09:
hi
when i use:
#define pin A1
...
  pinMode(pin, OUTPUT);

In accordance with the Pinout map, pin 7 Must be output, but pin 6 is output.
and Also for #define pin A2 accordance with the Pinout map, pin 3 Must be output, but pin 7 is output.
but #define pin A3 is true and pin 2 is output.

i write code ion arduino ide then use exported compiled binary and then uplode .hex file to ATtiny13 with tnm5000 programmer.

tnx

MCUdude, 2019-09-09:
First, you can't and should not redefine A0-A3. They are reserved, and defined [here](https://github.com/MCUdude/MicroCore/blob/95243232f489dfbe9e982aa751cc368ba6ce53e4/avr/variants/attiny13/pins_arduino.h#L72-L75).

All pins can be set to either input or output. I don't understand your question, not can I see a question mark.

Ehsan-Shahnazi, 2019-09-09:
how can i define "pin" to A1 Pin?
What code do I write to attribute the phrase pin to pin A1?

Ehsan-Shahnazi, 2019-09-09:
i want run this code to ATtiny13.
[https://github.com/offcircuit/SDHT/pull/1#issuecomment-519702705]
after write to attiny 13,Outputs do not work.
but in Arduino uno work correctly.

Ehsan-Shahnazi, 2019-09-09:
Please specify what code should I write instead of these codes?
```

**#define pin A2
#define PinPotTemp A1
#define PinPotHum A3
#define OutTemp 2
#define OutHum 3**
#define DHT_CYCLES microsecondsToClockCycles(200)
#define TOLERANCE_CELSIUS 10
#define TOLERANCE_HUMIDITY 10
```

MCUdude, 2019-09-09:
I don't know what your hardware looks like, but if you're doing digital operation on pin A2, you should use ` int pin = 4` instead.

nerdralph, 2019-09-09:
It's pretty bad code you're trying to use.  I'm surprised Hans didn't just
close the issue, as MicroCore isn't 100% compatible with all code that
happens to run on the Arduino.  Read up on good C/C++ coding methods,
particularly alternatives to mis-use of #defines (i.e. enums and
constants).  Also read up on good coding style, particularly good naming
conventions.  The name 'pin' is non-descrptive and very generic (an
therefore more likely to conflict with libraries). A name like
Sensor_IO_Pin is a lot better.

MCUdude, 2019-09-09:
>  I'm surprised Hans didn't just
close the issue, as MicroCore isn't 100% compatible with all code that
happens to run on the Arduino.

That's true. The ATtiny13 is very limited compared to other AVRs, so MicroCore is absolutely not 100% compatible with existing Arduino code. You can't really expect any Library to work, and I'm pretty sure the DHT11/DHT22 won't work without custom code written specifically for the ATtiny13.

In other words; the ATtiny13 is so limited you'll have to be more than a beginner to successfully build a decent project around it. You'll have to know of its shortcomings and how to write space efficient code for it.

Ehsan-Shahnazi, 2019-09-09:
in this code dont use any library...
[]https://github.com/offcircuit/SDHT/pull/1#issuecomment-518931077
and The code size is about 900 bytes.

MCUdude, 2019-09-09:
The ONLY pins that use analogRead is PinPotTemp and PinPotHum. That means that these are the only pins that should be defined as analog too. This means that your pin definitions should look like this instead:

```c
#define pin 4         // PB4
#define PinPotTemp A1 // PB2
#define PinPotHum A3  // PB3
#define OutTemp 0     // PB0
#define OutHum 1      // PB1
```

Just for reference (and formatting), here's the code.
```c++
#define pin A2
#define PinPotTemp A1
#define PinPotHum A3
#define OutTemp 2
#define OutHum 3

#define DHT_CYCLES microsecondsToClockCycles(200)
#define TOLERANCE_CELSIUS 10
#define TOLERANCE_HUMIDITY 10

void setup() {
pinMode(OutTemp, OUTPUT);
pinMode(OutHum, OUTPUT);
}

void loop() {
uint8_t data[5] = {0, 0, 0, 0, 0};

pinMode(pin, OUTPUT);
digitalWrite(pin, LOW);
pulseIn(pin, HIGH, 1000);
digitalWrite(pin, HIGH);
pinMode(pin, INPUT);

if (pulseIn(pin, HIGH, 300)) {

for (uint8_t i = 0; i < 40; i++)
  data[i / 8] += data[i / 8] + (pulse(LOW) > pulse(HIGH));

if (data[4] == uint8_t(data[0] + data[1] + data[2] + data[3])) {

  int16_t celsius = (uint16_t((data[2] & 0x7F) << 8) | data[3]) * ((data[2] & 0x80) ? -1 : 1);
  uint16_t humidity = (uint16_t(data[0] << 8) | data[1]);

  int16_t level = map(analogRead(PinPotTemp), 0, 1023, -400, 800);

  if ((level + TOLERANCE_CELSIUS) < celsius)
    digitalWrite(OutTemp, HIGH);
  else if ((level - TOLERANCE_CELSIUS) > celsius)
    digitalWrite(OutTemp, LOW);

  level = map(analogRead(PinPotHum), 0, 1023, 0, 1000);

  if ((level + TOLERANCE_HUMIDITY) < humidity)
    digitalWrite(OutHum, HIGH);
  else if ((level - TOLERANCE_HUMIDITY) > humidity)
    digitalWrite(OutHum, LOW);
}
}
delay(2000);
}

uint16_t pulse( bool state)
{
uint16_t cycle = DHT_CYCLES;
while ((digitalRead(pin) == state) && cycle--);
return cycle;
}
```

Ehsan-Shahnazi, 2019-09-09:
> The ONLY pins that use analogRead is PinPotTemp and PinPotHum. That means that these are the only pins that should be defined as analog too. This means that your pin definitions should look like [...]

hi... thanks !!
but dont work...
Where do you think the problem is?

MCUdude, 2019-09-10:
No idea. If you have trouble getting your code running you're better off asking on the [Arduino forum](https://forum.arduino.cc/). This place is for issues with MicriCore itself, which this issue clearly isn't.

closed 2019-09-10 by MCUdude

Ehsan-Shahnazi, 2019-09-14:
hello
I apologize for the many messages.
i tested many program...just led blink sketch worked correctly on ATtiny13...!!
i write this program for run lm35 dz on ATtiny13. on uno worked correctly but on ATtiny13 dont work.
can you test it and Help me in detail.

`#define PinSens A3         // PB3 = 2
#define PinPtnTemp A2 // PB2 = 7
#define PinOutTemp 1    // PB0 = 6

#define TOLERANCE_CELSIUS 1

// uint16_t temp;
// int tempPin = A3; // PB3

void setup() {
  pinMode(PinOutTemp, OUTPUT);
// Serial.begin(9600);
}

void loop() {
uint16_t temp = analogRead(PinSens);
 // read analog volt from sensor and save to variable temp
 temp = temp * 0.48828125;
 // convert the analog volt to its temperature equivalent

   int16_t level = map(analogRead(PinPtnTemp), 0, 1023, 2, 150);

   if ((level + TOLERANCE_CELSIUS) < temp)
    digitalWrite(PinOutTemp, HIGH);
  else if ((level - TOLERANCE_CELSIUS) > temp)
    digitalWrite(PinOutTemp, LOW);

/* Serial.print("   levelTemp => ");
 Serial.println(level);

 Serial.print("TEMPERATURE = ");
 Serial.print(temp); // display temperature value
 Serial.print("*C");
 Serial.println();
*/ 
 delay(1000); // update sensor reading each one second
}`

MCUdude, 2019-09-14:
I'm on my way out, so I don't have time to look at the code at the moment. But if you want you can look at some code I wrote a while ago for the ATtiny13. It's a fan controller that uses an LM35DZ to read the temperature of a heatsink, and controls a PC fan using PWM.

``` c++
/*************************************
| Altec Lansing 9444B fan controller |
| Created by MCUdude                 |
| MCU:    ATtiny13                   |
| F_CPU:  9.6 MHz                    |
| BOD:    2.7V                       |
| LTO:    Enabled                    |
| Millis: Disabed                    |
 ************************************/

// Physical connections
const uint8_t fanSpeedPin = 0;
const uint8_t tachPin = 1;
const uint8_t tempPin = A1;
const uint8_t fanPowerPin = 3;
const uint8_t ledPin = 4;

// Variables
uint8_t currentTemperature = 0;
uint8_t fanSpeed = 0;

// Calculate temperature
uint16_t readTemperature()
{
  // We should actually divive by 2.048 in order
  // for the calculation to be exact, but 2 is
  // good enough anyway
  uint16_t temp = analogRead(tempPin)/2;
  return temp;
}

void setup() 
{
  // Set inputs and outputs
  pinMode(fanSpeedPin, OUTPUT);
  pinMode(tachPin, INPUT);
  pinMode(fanPowerPin, OUTPUT);
  pinMode(ledPin, OUTPUT);

  // Enable fan power
  digitalWrite(fanPowerPin, HIGH);

  // Enable WDT
  wdt_enable(WDTO_1S);
  WDTCR = _BV(WDTIE);
  sei();
}

void loop() {
  // Read temperature
  currentTemperature = readTemperature();

  // Stop fan completly if temp < 40degC
  if(currentTemperature < 40)
    digitalWrite(fanPowerPin, LOW);
  
  // Run fan at lowest speed if temp < 50degC
  else if(currentTemperature < 50)
  {
    digitalWrite(fanPowerPin, HIGH);
    fanSpeed = 0;
  }

  // Temp is < 70degC
  else if(currentTemperature < 70)
    fanSpeed = map(currentTemperature, 50, 70, 1, 100);
    
  // Temp is > 70degC
  else if(currentTemperature < 100)
    fanSpeed = map(currentTemperature, 70, 100, 101, 255);

  // Temp is > 100degC
  else
    fanSpeed = 255;

  analogWrite(fanSpeedPin, fanSpeed);

  delay(100);
}

ISR(WDT_vect)
{
  // Toggle LED
  digitalWrite(ledPin, !digitalRead(ledPin));

  // Adjust blink speed according to temperature
  if(currentTemperature < 40)
    wdt_enable(WDTO_8S);
  else if(currentTemperature < 50)
    wdt_enable(WDTO_2S);
  else if(currentTemperature < 70)
    wdt_enable(WDTO_1S);
  else if(currentTemperature < 100)
    wdt_enable(WDTO_500MS);
  else if(currentTemperature < 130)
    wdt_enable(WDTO_120MS);
  else
    wdt_enable(WDTO_60MS);

  // Enable WDTIE again
  WDTCR |= _BV(WDTIE);
}

```