MCUdude/MicroCore #89: Implement a sketch for tuning the internal oscillator

MCUdude, 2019-12-15:
This has been discussed in #88, but I thought it might be easier to follow the discussion if I created a separate issue for the OSCCAL part of the serial implementation.

@nerdralph have already a working tuning sketch where the ATtiny13 received a character and tunes its own OSCCAL register to get the timing correct.

nerdralph, 2019-12-20:
Here's a working prototype.  It needs bbuart (BBUart.h + BBUart.S) installed as an Arduino library.
https://gist.github.com/nerdralph/4f2439af784fb08ab06e9fbcd378d98f

I think the process is somewhat cumbersome.  You need a programmer (i.e. USBasp) and serial adapter.  At least on my system, USBasp doesn't coexist well with other USB devices on a hub.  That means unplugging the USBasp after programming, then plugging in a USB-ttl uart dongle to run tuning program.

I think the tuning algorithm is quite robust.  I'm open to suggestions on improving usability.

MCUdude, 2019-12-20:
Thanks! However, I don't receive anything from the T13 when sending `x` form the serial monitor. Any idea what may be wrong? I've removes all HalfDuplexSerial related files too, just to get it to compile.

It would be great if it would seamlessly integrate with the `Serial` (HalfDuplexSerial) implementation I've been working on. If I don't remove the files the compiler is complaining big time.

nerdralph, 2019-12-20:
I was going to ask if you changed UART_RX to match your setup, but then you added that detail.
Have you tried restarting the Arduino IDE before testing?  I was using 1.8.9 on Windoze, and find it's serial port detection and use to be buggy.  When it wasn't working with the Serial monitor, I'd test it with putty to see if the problem was the code or the Arduino serial monitor.

As for the integration with HalfDuplexSerial, I didn't notice you have the branch pushed to github.  I'll pull that and make any mods necessary to the tuning sketch.

I might be a few days before I get around to it as I want to finish some changes to the core and push them to my fork first; I don't want to juggle a couple different versions of the core.  Just before you added the issue for HalfDuplexSerial, I had almost finished transparent millis() support.  When you call millis() in your sketch, millis.S gets linked in, and when you don't call it, it gets left out.  I also optimized the millis() function using inline asm.  The last thing I need to do is move the millis init code (WDTCR & sei) out of wiring.c into millis.S.

MCUdude, 2019-12-20:
> Have you tried restarting the Arduino IDE before testing? I was using 1.8.9 on Windoze, and find it's serial port detection and use to be buggy. When it wasn't working with the Serial monitor, I'd test it with putty to see if the problem was the code or the Arduino serial monitor.

I didn't restart Arduino IDE, but I can confirm that data is received by the T13 since the RX LED is flashing. I can also confirm that it's not transmitting since the debug LED isn't toggling, not does the TX LED flash.

> I might be a few days before I get around to it as I want to finish some changes to the core and push them to my fork first; I don't want to juggle a couple of different versions of the core. Just before you added the issue for HalfDuplexSerial, I had almost finished transparent millis() support. When you call millis() in your sketch, millis.S get linked in, and when you don't call it, it gets left out. I also optimized the millis() function using inline asm. The last thing I need to do is move the millis init code (WDTCR & sei) out of wiring.c into millis.S.

No worries, take your time! I'm currently away for the Christmas holiday, but I brought with me my (mac) computer, my AVR DIP-8 development board you've seen, and a USBtinyISP programmer. I don't have too much time to play around with this the following days, but at least I have the gear to do it.

BTW would `attachInterrupt` be fast enough to use instead of PCINT? I mean, using this gives code that's "easier to read", and since the Rx pin is on PB1 we have the INT0 vector free anyways.

And how about different baud rates? Can the code use the default defined baud rate and use this to for timing calculations?

I'm interested to see what millis() improvements you're able to come up with! The only bad thing about the current implementation is the speed and accuracy of the WDT timer. It's not very accurate and not very fast. Maybe (in the future) there's possible to use a (super) optimized millis implementation using timer0? Not exactly top priority though.

MCUdude, 2019-12-20:
I managed to get it working, and I've even been able to run it using HalfDuplexSerial. Amazing work!!
I've tried "Arduinoized" it a little, with success. The baud rate does not matter either; it runs well on slow and fast baud rates.  BTW I did figure out why the PCINT ISR wasn't firing. It turned out that I had to move it up; earlier in the setup().

```cpp
const uint8_t uart_rx_pin = 1;
uint8_t oldOSCCAL;

// Converts 4-bit nibble to ascii hex
uint8_t nibbletohex(uint8_t value)
{
  value &= 0x0F;
  if (value > 9)
    value += 'A' - ':';
  return value + '0';
}

// Function to run when Rx interrupt occurs
void rxInterrupt()
{
  uint8_t current = TCNT0;
  // Start timer when pin transitions low
  if ((PINB & _BV(uart_rx_pin)) == 0)
    TCCR0B = _BV(CS00);
  else
  {
    // End of interval, reset counter
    TCCR0B = 0;
    TCNT0 = 0;
    // 'x' begins with 3 zeros + start bit = 4
    uint8_t target = 4 * F_CPU / BAUD_RATE;
    int8_t delta = target - current;
    if (delta > 8) 
      OSCCAL++;
    if (delta < -8)
      OSCCAL--;
    asm("lpm"); // 3 cycle delay
    // Print calculation
    Serial.print(F("Delta: 0x"));
    Serial.write(nibbletohex(delta >> 4));
    Serial.write(nibbletohex(delta));
    Serial.print(F("  New OSCCAL: 0x"));
    Serial.write(nibbletohex(OSCCAL >> 4));
    Serial.write(nibbletohex(OSCCAL));
    Serial.print(F("  Old OSCCAL: 0x"));
    Serial.write(nibbletohex(oldOSCCAL >> 4));
    Serial.write(nibbletohex(oldOSCCAL));
    Serial.write('\n');
  }
  // Clear interrupt flag in case another triggered
  GIFR = _BV(INTF0);
}

void setup() 
{
  // Store old OSCCAL value for reference
  oldOSCCAL = OSCCAL;
  
  // Prepare for sleep mode
  MCUCR = _BV(SE);

  // Enable Rx pin pullup
  digitalWrite(uart_rx_pin,HIGH);
  
  // Setup RX interrupt
  attachInterrupt(0, rxInterrupt, CHANGE);
  
  delay(1000);

  // Print default message
  Serial.write('x');
  Serial.write('\n');
  
  // Wait for tuning character to ensure not reading noise
  // before entering tuning mode
  wait_x:
  uint8_t counter = 0;
  while (digitalRead(uart_rx_pin));
  do 
  {
    counter++;
  } while (digitalRead(uart_rx_pin) == 0);

  // Low period should be 4 bit-times
  uint8_t margin = BIT_CYCLES/8;
  if (counter - (uint8_t)BIT_CYCLES > margin)
    goto wait_x;
  else
  {
    if ((uint8_t)BIT_CYCLES - counter > margin)
      goto wait_x;
  }

  delay(1); // Skip remaining bits in frame
}

void loop() 
{
  // Nothing here really...
  asm("sleep");
}

```

MCUdude, 2019-12-21:
I've also pushed an updated HalfDuplexSerial library that's compatible with the sketch above and some Serial examples borrowed from @sleemanj's repo.

nerdralph, 2019-12-21:
That's great.  I am a bit concerned about attaching the interrupt before checking for the first 'x'.  Before I added that, sometimes the noise when connecting the uart dongle would be misread as a tuning character, throwing the timing off even more, making the tx output garbage.

nerdralph, 2019-12-21:
It would also be best to choose millis disabled (until I push my millis auto-linking patch).  If the millis ISR happens to be running when INT0 fires, it will delay starting the timer.
The millis ISR could be changed to interruptible, but that's a discussion for another day.

nerdralph, 2019-12-21:
Another suggestion is to reduce the delta range for modifying OSCCAL.  I picked +-8 based on my testing with 9.6Mhz and 38400 baud, which would tune to within +-1% of the uart timing from the host.  For 9.6Mhz and 115.2k, I'd change the range to +-3.

MCUdude, 2019-12-21:
I have a few other questions as well. What does the delta value actually represent? And why is the microcontroller put into sleep mode?

MCUdude, 2019-12-21:
Is there something wrong with the calculation? I've previously measured the correct OSCCAL value for the 4.8 MHz oscillator to be 0x69. However, when I calibrate with the sketch about I get this output when repeatedly pressing [x] [enter]:

```
⸮⸮⸮⸮⸮⸮w⸮r@`⸮'⸮⸮2⸮⸮⸮⸮w⸮@`⸮m⸮⸮⸮⸮⸮⸮3⸮r@`⸮hb@@⸮⸮⸮⸮⸮3⸮@`⸮i⸮⸮⸮⸮⸮⸮v@`⸮c⸮@@⸮⸮⸮@⸮⸮⸮r@`⸮m⸮⸮⸮⸮⸮⸮z@`⸮c⸮@@⸮⸮⸮@⸮⸮⸮z@`⸮m⸮⸮⸮⸮⸮⸮z@`⸮ce@@⸮⸮⸮@⸮⸮⸮z@`⸮e⸮⸮⸮⸮⸮⸮z@`⸮b⸮@@⸮⸮⸮@⸮⸮⸮z@`⸮f`⸮⸮⸮⸮⸮z@p⸮cs@@⸮⸮⸮@⸮⸮⸮z@`⸮va⸮⸮⸮⸮⸮z`p⸮r⸮``⸮⸮⸮@⸮⸮⸮z`p⸮6r
Delta: 0x2x  Ne⸮ ⸮⸮l: 0⸮63
⸮elta: 0x23  New cal: 0x64
Delta: 0x20  New cal: 0x65
Delta: 0x18  New cal: 0x66
Delta: 0x13  New cal: 0x67
Delta: 0x0C  New cal: 0x68
Delta: 0x0E  New cal: 0x69
Delta: 0x09  New cal: 0x6A
Delta: 0x03  New cal: 0x6A
Delta: 0x03  New cal: 0x6A
Delta: 0x01  New cal: 0x6A
Delta: 0x01  New cal: 0x6A
Delta: 0x⸮1  New cal: 0x6A
Delta: 0x02  New cal: 0x6A
Delta: 0x02  New cal: 0x6A
Delta: 0x02  New cal: 0x6A
Delta: 0x01  New cal: 0x6A
```

As you can see, one in a while I get some '?' characters. I do not get this when the OSCCAL value is 0x69. For reference, I've disabled millis and micros.

The baudrate I'm using for 4.8 MHz is 57600, and I'm using a delta of +-8.

nerdralph, 2019-12-21:
Another suggestion is to reduce the delta range for modifying OSCCAL.  I picked +-8 based on my testing with 9.6Mhz and 38400 baud, which would tune to within +-1% of the uart timing from the host.  For 9.6Mhz and 115.2k, I'd change the range to +-3.

> 
> 
> I have a few other questions as well. What does the delta value actually represent? Can a delta of 3/-3 be used bor both 115200 and 38400?
> And why is the microcontroller put into sleep mode?

The delta represents the difference between the expected timing and the measured timing.  'x' is 0x78, or 0b0111 1000.  Uart transmission is LSB first, with a 0 start bit, so the first four bits are 0, followed by 4 1 bits.  The ISR times the first four 0 bits.  The variable target is the calculated timer value for the given bitrate.  Mabye 'expected' would be a better name for it.  When delta is only 1 cycle, changing OSCCAL will likely result in a larger error, since changing OSCCAL by 1 changes the frequency by about 1%.  At 38400 and 9.6Mhz, I calculated that when delta is +-8, the timing is within +- 0.8% of optimal.  At 115.2, to get within +-0.8%, you'd need to use a delta of 3.  Using a delta of 3 at 38.4 is OK, it just means it will try to find an OSCCAL value that is within +-0.3% of optimal.  Since the t13's RC oscillator isn't that precise, the tuner program will usually end up alternating between two OSCCAL values that get closest to that 0.3% range.

The reason for sleep is to eliminate interrupt latency jitter.  It's not necessary as the jitter would be 1 cycle without it.  But it's a simple thing to do, so I go for the 0 jitter.  The reason it works is because wake from idle sleep is guaranteed to take 4 cycles.  When not in sleep, the ISR handling starts after the current instruction completes. Since most AVR instructions are 1 or 2 cycles, when not in sleep the ISR latency will vary between 1 and 2 cycles depending on the current instruction.

nerdralph, 2019-12-21:
> Is there something wrong with the calculation? I've previously measured the correct OSCCAL value for the 4.8 MHz oscillator to be 0x69. 

It looks like it's miscalculating the optimal timing by about 9 or 10 cycles.  I don't see anything obvious in your modifications that would throw off the timing.  I'll try testing your version, using PB0/PB1 and INT0, to see what's going on.  I wonder if attachInterrupt() is throwing off the timing a bit vs. a direct ISR.  I'll be able to tell for sure by looking at the output assembler code.

nerdralph, 2019-12-21:
Looking at the asm, there was a 4-cycle latency between when the timer gets started and when TCNT0 is sampled.  Moving 'current = TCNT0' into the else block reduces that to 0.
https://gist.github.com/nerdralph/4f2439af784fb08ab06e9fbcd378d98f

Your results suggest another 5 or 6 cycles of error somewhere, so I'll keep investigating.

MCUdude, 2019-12-21:
I just updated the sketch above with minor tweaks. It would be nice if the calibrated OSCCAL value could be stored in EEPROM address 0 after the value has stabilized followed by a message like this:
`OSCCAL calibrated. Value stored in EEPROM addr. 0`. I'll try your fix and report back!

MCUdude, 2019-12-21:
At 4.8 MHz and 57600 baud I now get this output:

```
⸮⸮⸮⸮⸮⸮r@`⸮gb@@⸮⸮⸮@⸮⸮⸮v@`⸮m⸮⸮⸮⸮⸮⸮z@`⸮cg@@⸮⸮⸮@⸮⸮⸮r@`⸮m⸮⸮⸮⸮⸮⸮z@p⸮ca@@⸮⸮⸮@⸮⸮⸮z@`⸮u⸮⸮⸮⸮⸮⸮z@`⸮b⸮@@⸮⸮⸮@⸮⸮⸮z@`⸮n`⸮⸮⸮⸮⸮z@p⸮r⸮@@⸮⸮⸮@⸮⸮⸮z@`⸮vq
⸮⸮⸮⸮⸮z@p⸮r⸮@`⸮⸮⸮@⸮⸮⸮z@p⸮vr
D⸮⸮t⸮: 0⸮26` ⸮⸮⸮@⸮⸮⸮z 0⸮63
Delta: 0x20  Ne⸮ ⸮al: 0⸮64
Delta: 0x1B  New cal: 0x65
Delta: 0x15  New cal: 0x66
Delta: 0x0F  New cal: 0x67
Delta: 0x0A  New cal: 0x68
Delta: 0x0B  New cal: 0x69
Delta: 0x05  New cal: 0x69
Delta: 0x05  New cal: 0x69
Delta: 0x07  New cal: 0x69
Delta: 0x06  New cal: 0x69
Delta: 0x05  New cal: 0x69
Delta: 0x06  New cal: 0x69
Delta: 0x05  New cal: 0x69
Delta: 0x06  New cal: 0x69
Delta: 0x07  New cal: 0x69
Delta: 0x05  New cal: 0x69
Delta: 0x07  New cal: 0x69
Delta: 0x06  New cal: 0x69
Delta: 0x07  New cal: 0x69
Delta: 0x07  New cal: 0x69
Delta: 0x07  New cal: 0x69
Delta: 0x08  New cal: 0x69
Delta: 0x06  New cal: 0x69
Delta: 0x05  New cal: 0x69
Delta: 0x06  New cal: 0x69
Delta: 0x08  New cal: 0x69
Delta: 0x08  New cal: 0x69
Delta: 0x07  New cal: 0x69
Delta: 0x06  New cal: 0x69
Delta: 0x07  New cal: 0x69
Delta: 0x07  New cal: 0x69
Delta: 0x07  New cal: 0x69
Delta: 0x05  New cal: 0x69
Delta: 0x06  New cal: 0x69
Delta: 0x07  New cal: 0x69
Delta: 0x06  New cal: 0x69
```

It stays at 0x69, but is close to "rolling over" to 0x6A.

nerdralph, 2019-12-21:
> 
> 
> I just updated the sketch above with minor tweaks. It would be nice if the calibrated OSCCAL value could be stored in EEPROM address 0 after the value has stabilized followed by a message like this:
> `OSCCAL calibrated. Value stored in EEPROM addr. 0`. I'll try your fix and report back!

One thing at a time...

nerdralph, 2019-12-21:
I'm now wondering how accurate the CH330N is with it's UART timing.  I use the Pl2303HX.  They were the cheapest thing I could find several years ago, and as I got to play with them, I found their specs are better than anything 2-3x the cost.  Their output timing is within 0.1%, even up to 6mbps (twice the max speed of the FT232L).
http://nerdralph.blogspot.com/2014/05/high-speed-uart-pumping-out-data-at.html

MCUdude, 2019-12-21:
> One thing at a time...
 Yes, I know. I'm just very enthusiastic about this! When all this is ready, it will be a significant milestone for MicroCore!

> I'm now wondering how accurate the CH330N is with it's UART timing. I use the PL2303HX. They were the cheapest thing I could find several years ago, and as I got to play with them, I found their [...]

You're right. The CH330N (and other crystal-less CH3*'s) are only specified to be 1% accurate. Did you look through the assembler output and concluded that there's nothing wrong with the code anymore, timing wise?

nerdralph, 2019-12-21:
> You're right. The CH330N (and other crystal-less CH3*'s) are only specified to be 1% accurate. Did you look through the assembler output and concluded that there's nothing wrong with the code [...]

Have you measured the actual timing with your scope?  If not, I can do it when I receive your board with the CH330N.

I didn't find anything wrong with the code _per se_, however I realized that since the goal of the tuner is to optimize the timing of my soft uart, not to actually tune as close as possible to the chosen F_CPU.  It's a subtle difference, but I think I can modify the tuning code to pick the best OSCCAL based on the known error offset of my uart code.

As I mentioned before, my uart is accurate within +-1.5 cycles because the delay loop has a resolution of 3 cycles.  At 57.6k and 4.8Mhz, one bit should take 83.33cycles.  My uart will output one bit every 82cycles, making it fast by 1.63%.  So the optimal OSCCAL would be when the t13 is running 1.63% slower than the way I currently calculate.  Interestingly, that would be a delta of 5.33 cycles for the tuner.

MCUdude, 2019-12-21:
> Have you measured the actual timing with your scope? If not, I can do it when I receive your board with the CH330N.

No, I haven't. I'm currently away and have minimal equipment available. Would be interesting to see what error you'll measure your chip to be. The error here will certainly affect the calculated value, and is pretty much unavoidable.

> I didn't find anything wrong with the code per se, however I realized that the goal of the tuner is to optimize the timing of my soft uart, not to actually tune as close as possible to the chosen F_CPU. It's a subtle difference, but I think I can modify the tuning code to pick the best OSCCAL based on the known error offset of my uart code.

Would this make a practical difference to the actual calculated value, or will it only be a fraction of the calculated delta? After I moved `current = TCNT0;` into `else` it seems to be "better" at calculating the correct value, as long as I keep the delta fairly high (+-8).

nerdralph, 2019-12-21:
> Would this make a practical difference to the actual calculated value, or will it only be a fraction of the calculated delta? After I moved `current = TCNT0;` into `else` it seems to be "better" at [...]

It'll make slightly more difference than moving the TCNT0 sampling did.

nerdralph, 2019-12-23:
I figured out why the interrupt wasn't starting for you with INT0.  The counter code in setup was using digitalRead() which, when used on PB0 or PB1, disables PWM before reading the pin.  That threw off the timing of the counter loop, so a received 'x' would never be recognized.
I updated my version to use direct port access, and I adjusted the interrupt to tune towards the best OSCCAL for the actual number of cycles TxByte() takes.  I also updated the delta range to +-4.  In my testing with 115.2@9.6Mhz, a delta of 8 was to far to get the best OSCCAL (off by 1).

I didn't get around to pulling the softSerial branch, so you'll need to copy the appropriate changes from my gist.
https://gist.github.com/nerdralph/4f2439af784fb08ab06e9fbcd378d98f

MCUdude, 2019-12-23:
Awesome, there we have it! Instead of getting 0x6A (which is too high and I'm getting `?`'s) I'm now getting a stable 0x67 that seems to be spot on. I think I'm ready to merge the branch into master!
Again, thank you for your amazing work!

closed 2019-12-23 by MCUdude

nerdralph, 2019-12-23:
I noticed in the docs you say, "select No line ending".  This should not be necessary, as the ISR will not finish until well after the LF (or CR/LF) is received.  It will trigger the INT0 flag, which gets reset at the end of the ISR.

MCUdude, 2019-12-23:
Hmm. I turned on CR LF line endings and the calculated OSCCAL value I got 0x69. With no line endings, I got 0x67 (4.8 MHz, 57600 baud). IMO it doesn't hurt to specify *No line ending* even though it works with.

PR #92 (2019-12-23): Add Serial/UART support and oscillator calibration
We finally have proper "Arduino-like" serial communication. Another issue with the ATtiny13 is that the internal oscillator is known to be way out of tune, especially those sourced from China. This PR also includes a nice oscillator calibration sketch that uses a serial interface for tuning.
Closes #88 and closes #89.