MCUdude/MicroCore #93: Improve serial receive handling

MCUdude, 2019-12-23:
@nerdralph mentioned in #88 that the UART Rx routine could be further improved to free up more time. Will it be ISR based and/or timer-based? Will this affect timing derived from timer 0?

Maybe it's also possible to implement a 1-byte buffer so that Serial.available can be used? However, I'm not sure how well this would go together with the current implementation, mostly done by @sleemanj. Anyways, it's very interesting!

nerdralph, 2020-05-08:
It took a little while, but here it is:
https://github.com/nerdralph/picoUART/releases/tag/v1.0.0
I tested on a t13 and a t85.

In the process of writing it, I've learned a lot about AVR interrupt latencies.
http://nerdralph.blogspot.com/2020/04/measuring-avr-interrupt-latency.html

MCUdude, 2020-05-09:
Thanks! I will give it a try on my test board and add it to MicroCore if everything seems to work fine.

Hans

MCUdude, 2020-05-09:
I’ve now done most of the work, and receive and transmit works perfect. The only think missing now is the OscillatorCalibration sketch. Do you have any idea what to replace TXDELAYCOUNT with?

Hans

nerdralph, 2020-05-09:
> 
> I’ve now done most of the work, and receive and transmit works perfect. The only think missing now is the OscillatorCalibration sketch. Do you have any idea what to replace TXDELAYCOUNT with? Hans
>

The bit delay code is the biggest change with picoUART vs. my earlier bitbang UART.  I'll try to take a few minutes over the next day to see if it is a simple recalculation.  My top priority project right now is an interrupt-driven PWM-assisted transmit routine.  I had a prototype version working last year, and lately I've been optimizing the code and modifying it to work on not just the t13 but also the t85 and t84 as well (and maybe even the t88).

nerdralph, 2020-05-09:
I took a quick look at the code (no actual testing), and for line 81 I think 
'uint8_t expected = 4 * BIT_CYCLES;'
should do the trick.
https://github.com/MCUdude/MicroCore/blob/master/avr/libraries/Serial_exampes/examples/OscillatorCalibration/OscillatorCalibration.ino#L81

I also think I noticed a bug in the noise skipping code that you ran into before,  It might be a few days before I get around to fixing and testing it though.

MCUdude, 2020-05-10:
Are you sure you don’t mean PUBIT_CYCLES? Because BIT_CYCLES are not defined in picoUART v1.0.0
I tried to replace it with 4 * PUBIT_CYCLES, but I’m now getting errors related to PUBIT_CYCLES other places, since PUBIT_CYCLES is a flood, while BIT_CYCLES was an integer. It would be great if you had time to look into it one day! This is the last thing to be figured out before I could push the entire update.

> I also think I noticed a bug in the noise skipping code that you ran into before, It might be a few days before I get around to fixing and testing it though.

Noise skipping code? In the OscillatorCalibration sketch?

Hans

nerdralph, 2020-05-10:
> Are you sure you don’t mean PUBIT_CYCLES? Because BIT_CYCLES are not defined in picoUART v1.0.0 I tried to replace it with 4 * PUBIT_CYCLES, but I’m now getting errors related to PUBIT_CYCLES other [...]

Yeah, I changed it to PUBIT_CYCLES to avoid name collision with the BIT_CYCLES that was used in my earlier UART code.  Trying to do floating point math with macros was messy, which is why I switched to inline functions.  The quickest way I'll get an OSCCAL sketch working is to modify my original version.  The modifications you did tied it to MicroCore, while my version was generic so it would work with other tiny cores (ATtinyCore, and even my picoCore).  So what I'll probably do is include an updated version of my OSCCAL sketch as one of the picoUART examples.  It won't be dependent on any specific core, so you or @sleemanj, or anyone else can use it.

> I also think I noticed a bug in the noise skipping code that you ran into before, It might be a few days before I get around to fixing and testing it though.
> Noise skipping code? In the OscillatorCalibration sketch? Hans
>

Yes.  In my version I tried to detect something that looked like an 'x' (with a wide timing margin) before enabling the timing interrupt.  You had commented that it didn't work for until you enabled the interrupt near the beginning of the setup() function.  That's fine if the USB-ttl adapter is connected at boot time like with your tiny dev board, but when connecting an external USB-ttl adapter after the AVR has started, there is usually a lot of noise for several tens of ms before the line is at a stable high (idle) voltage.

MCUdude, 2020-05-10:
> The quickest way I'll get an OSCCAL sketch working is to modify my original version. The modifications you did tied it to MicroCore, while my version was generic so it would work with other tiny cores (ATtinyCore, and even my picoCore).

Sorry, but I can't seem to find this version. Do you have a link?

nerdralph, 2020-05-10:
> 
> 
> > The quickest way I'll get an OSCCAL sketch working is to modify my original version. The modifications you did tied it to MicroCore, while my version was generic so it would work with other tiny cores (ATtinyCore, and even my picoCore).
> 
> Sorry, but I can't seem to find this version. Do you have a link? And is this version compatible with picoUART v1.0.0?

I'm referring to the one I made for my old BBuart code.
https://gist.github.com/nerdralph/4f2439af784fb08ab06e9fbcd378d98f

It will require some basic changes for picoUART.  Since it should only take me an hour or so, I'll make it a priority.  Though with it being Mother's Day, I might not get to it until tomorrow.

MCUdude, 2020-05-10:
Thanks! Mother's Day here in Norway is in February.

Yes, I remember that code now. However, it still has TXDELAYCOUNT and BIT_CYCLES, so as you pointed out it's not directly compatible. I'll stay tuned!

nerdralph, 2020-05-11:
I've got the tuner sketch working with picoUART 1.0, but I'm going to make some improvements before I release it as part of picoUART 1.1.
I'm going to have it report when no change from the factory OSCCAL value is necessary.
I'm also making portability/configurability changes.  I already changed it from using INT0 to using PCINT, so it is not limited to using the INT0 pin for receive.  On the t85 INT0 is PB2, vs PB1 on the t13.

MCUdude, 2020-05-11:
> I've got the tuner sketch working with picoUART 1.0, but I'm going to make some improvements before I release it as part of picoUART 1.1.

Great, hope you will release it soon so I can add it to MicroCore! Are there other changes to 1.1 vs 1.0 that I should know of?

I do agree that it's better to use PCINT than INT0, because, as you've pointed out gives the user greater flexibility to switch to a different receive pin.

nerdralph, 2020-05-12:
I've just pushed the new tuner example.
https://github.com/nerdralph/picoUART/tree/master/examples/ClockTune
I still have to do some housekeeping work like updating the changelog and README before I tag a v1.1.0 release, but I consider the ClockTune to be release ready.
Although I think it's simpler than the earlier osccal tuner, I'll probably write a blog post about the 1.1 release including instructions on using ClockTune.
I remembered @RalphBacon commenting how he experimented with clock tuning only to find the factory OSCCAL value was good.  So ClockTune reports "OSCCAL OK" after the first 'x' if the clock very close.

MCUdude, 2020-05-13:
Thanks. I had to do a little tweaking to get it to work though. At first, it printed the welcome message, but never printed anything when I sent 'x' characters. by moving the PCINT ISR setup a little earlier in setup() (before the goto part), it suddenly worked.

nerdralph, 2020-05-13:
The ISR is enabled after the busy-loop check in order to avoid having the interrupt trigger on noise.   It'll also mess up the initial check to see if calibration is needed or not.
If you get the "Hit x to test." and then "OSCCAL OK", the ISR never runs because the factory OSCCAL is satisfactory.
You had the same problem in my earlier version, so I'd like to figure out what's going on.  If you just see the "Hit x to test." and nothing more, it's possible my code to roughly recognize the first 'x' is too narrow.  I'm going to add a diagnostic when it discards a character as noise.

MCUdude, 2020-05-13:
> If you just see the "Hit x to test." and nothing more, it's possible my code to roughly recognize the first 'x' is too narrow.

That's exactly what's happening. However, the oscillator is so off that the message isn't readable:

```
  5C`Hit x t⸮`te⸮⸮.
 OSC⸮⸮⸮: 0x 5⸮ - slow 
OSCCAL: 0x 5D - slow 
OSCCAL: 0x 5E - slow 
OSCCAL: 0x 5F - slow 
OSCCAL: 0x 60 - slow 
OSCCAL: 0x 61 - slow 
OSCCAL: 0x 62 - good 
```

nerdralph, 2020-05-13:
I just pushed the updated code.  I widened the range of what it initially accepts as an x from +-12.5% to +-25%.
If you have properly-tuned chip, you can test the noise elimination by sending a 'o' instead of an 'x'.  You'll get a "noise skipped" message.

MCUdude, 2020-05-13:
I moved the interrupt initialization to the end of setup() and added your latest commit:
Here's the output when repeatedly sending x's:
```
5⸮ ⸮⸮t ⸮ ⸮o ⸮⸮⸮tn
 ⸮o⸮se s⸮⸮p⸮ed
 ⸮o⸮⸮e`s⸮ip⸮e⸮
 ⸮oi⸮e ⸮ki⸮p⸮d
 ⸮oi⸮e sk⸮pp⸮⸮
 No⸮s⸮ ⸮k⸮⸮p⸮d
 ⸮o⸮se`s⸮⸮p⸮ed
 ⸮⸮⸮⸮⸮ ⸮k⸮p⸮e⸮
 ⸮o⸮s⸮`s⸮ip⸮e⸮
 ⸮o⸮s⸮ ⸮k⸮p⸮e⸮
```

nerdralph, 2020-05-13:
OK, try /2 instead of /4 on line 87.
https://github.com/nerdralph/picoUART/blob/master/examples/ClockTune/ClockTune.ino#L87

MCUdude, 2020-05-13:
I'm getting the same output. The only thing I've modified is replacing `prints_P(STR(` with `Serial.print(F(`. However, Serial.print uses the same putx function as prints, so this shouldn't make a difference

nerdralph, 2020-05-13:
OK, then it must be the math on the margin calculations, because there' no way the timing is off by more than 50%.  Maybe a uint8_t overflow corner case or something like that.
Or maybe it's the abs() function; I think you redefine it in MicroCore, but I don't in picoCore which is what I've been testing with.  Can you try __builtin_abs() on line 88?
https://github.com/nerdralph/picoUART/blob/master/examples/ClockTune/ClockTune.ino#L88

MCUdude, 2020-05-13:
Guess what, that did the trick!

The reason why I've re-defined abs in MicroCore is that Arduino did it in the code. Any idea why they did?

nerdralph, 2020-05-13:
> 
> 
> Guess what, that did the trick!
> 
> The reason why I've re-defined abs in MicroCore is that Arduino did it in the code. Any idea why they did?

The Arduino core does lots of stupid things; many of the contributors weren't experts in embedded software.  Macro functions should only be used when you *REALLY* know what you are doing and understand the pitfalls of macro side-effects and lack of type-safety.  Just google "macros are bad" and you'll find plenty of explanations.

As for the sleep, it guarantees the interrupt latency has no jitter.  When not sleeping, interrupt response time is delayed when the interrupt triggers during the first cycle of a multi-cycle instruction.  See my blog post on interrupt latency for more details.

MCUdude, 2020-05-13:
> The Arduino core does lots of stupid things; many of the contributors weren't experts in embedded software. Macro functions should only be used when you REALLY know what you are doing and understand [...]

Yeah, I'm aware that macro functions can be "dangerous". But is there any reason why they had to undefine the standard avrlibc one and create their own. The new one obviously has its flaws...

> As for the sleep, it guarantees the interrupt latency has no jitter. When not sleeping, interrupt response time is delayed when the interrupt triggers during the first cycle of a multi-cycle instruction. See my blog post on interrupt latency for more details.

Thanks, will have a good read tonight!

nerdralph, 2020-05-13:
> Yeah, I'm aware that macro functions can be "dangerous". But is there any reason why they had to undefine the standard avrlibc one and create their own. The new one obviously has its flaws...

Don't assume that the people writing the Arduino were smart.  For the heck of it I did some searching, and the issue has been known for over 10 years:
https://forum.arduino.cc/index.php?topic=46274.0

Some fool claimed the abs() macro "just works", when the built-in abs() function is what works properly, and with all types (int, long, float, double).

MCUdude, 2020-05-13:
Thanks for the link. I've just removed the abs macro.

I have to say, performance-wise the picoUART library is just amazing! I'm able to send *and receive* at 500kbaud when running the ATtiny13 at 4.8 MHz. That's a HUGE improvement compared to the old BasicSerial library. And the speed goes all the way down to 300 baud if you like.

Do you have a table or a formula for figuring out what baud rates different clock frequencies support? It seems like 500kbaud works great @ 4.8 MHz, for 460800 does not for some reason.
I'm trying to create a table where all working baud rates for all clocks are listed.

nerdralph, 2020-05-13:
> Thanks for the link. I've just removed the abs macro.
>
> I have to say, performance-wise the picoUART library is just amazing! I'm able to send _and receive_ at 500kbaud when running the ATtiny13 at [...]

The maximum possible speed is 1/7th of the clock rate, though steps between baud rates will be coarse.  For example at 8 cycles/bit at 8M, the speed will be 1mbps.  At 9 cycles/bit, the speed will be 888.9kbps, which might be close enough for a UART running at 921.6.  Once you get above 15-20 cycles/bit, picoUART will get close enough to the target baud rate for error-free communication.

For the interrupt-driven rx, the overhead of the ISR significantly reduces the maximum error-free baud rate to about 1/60th of the clock rate.  I could probably rewrite the ISR to allow for receiving at 4x that speed, but it's already good enough for 115.2kbps @ 8Mhz.

When you go above 115.2kbps, you need to make sure your UART adapter works at your chosen speed, especially under Windows where the drivers usually are not as good as Linux for setting arbitrary baud rates.  For example the Windows driver for my PL2303HX modules does not support 500kbps, but it does support 460.8 and 921.6kbps.  Under Linux I can use 460.8, 500, 921.6, and many baud rates in between.  With your CH330, the Windows driver may support 500kbps, but not 460.8.

commit 6adf381c (2020-05-13): Use @nerdralph's picoUART for serial communication
Closes #93
closed 2020-05-13 by MCUdude