MCUdude/MicroCore #94: Transparent millis support?

MCUdude, 2019-12-23:
To quote @nerdralph from #89:

> ..... Just before you added the issue for HalfDuplexSerial, I had almost finished transparent millis() support. When you call millis() in your sketch, millis.S get linked in, and when you don't call it, it gets left out. I also optimized the millis() function using inline asm. The last thing I need to do is move the millis init code (WDTCR & sei) out of wiring.c into millis.S.

Transparent millis (and maybe micros in the future?) support would be awesome. This means that there wouldn't any need for a dedicated menu option to enable/disable millis, it happens automatically if either is called in the user program. Fewer options for users to deal with equals more user-friendliness IMO.

nerdralph, 2019-12-24:
It works, but I've run into some merge conflicts with your latest push.  I should have it sorted out by tomorrow, so I can send you a pull request as an early Xmas present. :-)

nerdralph, 2019-12-24:
As promised.
https://github.com/MCUdude/MicroCore/pull/95

MCUdude, 2019-12-24:
Thanks! I'll look into it soon. For obvious reasons, I have limited time for this stuff today 😉

nerdralph, 2019-12-24:
BTW, here's the blink-millis sketch I used for testing the millis changes.
https://gist.github.com/nerdralph/8af5ade84839f4cc3256faa470434db8

commit c7b986ec (2019-12-25): Automatically link in millis when used
 This means we no longer need a separate menu option to enable or disable millis. It's handled automatically now. Thanks to @nerdralph for providing this. Closes #94 and #95
closed 2019-12-25 by MCUdude