MCUdude/MicroCore #96: put Serial on a diet

nerdralph, 2019-12-24:
The standard Arduino Serial & Print classes are resource pigs.  Adding a single Serial.println() to a sketch in MicroCore increases flash use by over 200 bytes, and uses 12 bytes of RAM.  One of the easiest ways to strip them down is to make the classes fully static.  MicroCore is not going to have Serial, Serial1, & Serial2, so creating a non-static instance wastes flash & RAM.
A few years ago I started writing a minimal Arduino core, and I got most of it working for the mX8 parts.  One of the things I did was make Serial static.  You could look at my picoWiring for ideas.
https://github.com/nerdralph/picoWiring

MCUdude, 2019-12-24:
I had a look at some picroWiring code, but I'm not sure how this would be applied to MicroCore. It seems like you're tweaking the Stream class, but MicroCore's HalfDuplexSerial inherits Print, and Stream isn't even available.

What would happen if other libraries (like LiquidCrystal) inherited Print when it's static?

nerdralph, 2019-12-24:
I started with Wiring's core (wiring.org.co), since it had improved the digitalWrite() performance after Arduino forked from it's base.  The same idea should work with Serial + Print.  You are correct that there could be limits to what can change without breaking other subclasses of Print.

nerdralph, 2019-12-25:
I played with gcc options for a while to try to get it to devirtualize the Print class virtual functions implemented by Serial.  -fdevirtualize is included with -Os, but I couldn't get the compiler to optimize them away so I changed the functions to be no longer virtual.  That saves 228 bytes of flash and 10 bytes of RAM, and doesn't impact the ability for other classes (i.e. LiquidCrystal) to inherit from Print.

MCUdude, 2019-12-25:
Thanks! Your instructions aren't 100% clear to me though. I tried to remove `virtual` from print.h, and the initializer (= 0), but I'm just getting this error:

`/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T//cclSd7Xl.ltrans0.ltrans.o: In function `write':
/Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/Print.cpp:49: undefined reference to `Print::write(unsigned char)'
/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T//cclSd7Xl.ltrans0.ltrans.o: In function `print':
/Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/Print.cpp:111: undefined reference to `Print::write(unsigned char)'
/Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/Print.cpp:111: undefined reference to `Print::write(unsigned char)'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ATtiny13.
`

Do I still have to add the `-fdevirtualize ` flag?

MCUdude, 2019-12-25:
In other words, would you like to show me how this can be done?

Merry Christmas! 🎄

nerdralph, 2019-12-25:
Here's the diff for Print.h:
```
diff --git a/avr/cores/microcore/Print.h b/avr/cores/microcore/Print.h
index db06635..0774f04 100644
--- a/avr/cores/microcore/Print.h
+++ b/avr/cores/microcore/Print.h
@@ -160,21 +160,19 @@
 class Print
 {
   private:
-    int write_error;
-
     size_t printNumber(UNSIGNED_PRINT_INT_TYPE, uint8_t);
     size_t printFloat(double, uint8_t);
   protected:
-    void setWriteError(int err = 1) { write_error = err; }
+    void setWriteError(int err = 1) { }
   public:
-    Print() : write_error(0) {}
+    Print() {}

-    int getWriteError() { return write_error; }
+    int getWriteError() { return 0; }
     void clearWriteError() { setWriteError(0); }

-    virtual size_t write(uint8_t) = 0;
+    size_t write(uint8_t) {};
     size_t write(const char *str) { return write((const uint8_t *)str, strlen(s
tr)); }
-    virtual size_t write(const uint8_t *buffer, size_t size);
+    size_t write(const uint8_t *buffer, size_t size);
```

god jul

MCUdude, 2019-12-25:
It does compile, but I'm getting some compiler warning I'd prefer to get rid of. If the functions work like expected and these are only "empty warnings", is there a way to simply mute them? Or could the code be tweaked with, so the compiler doesn't throw a warning in the first place?

I did try this fix with the ASCIITable example. It seems like Serial.write() is the only function that works now. But the compiled size is greatly reduced though!

> god jul

Tusen takk / Thank you!

```
In file included from /Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/HalfDuplexSerial.h:24:0,
                 from /Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/Arduino.h:119,
                 from /Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/Tone.cpp:25:
/Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/Print.h: In member function 'void Print::setWriteError(int)':
/Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/Print.h:169:34: warning: unused parameter 'err' [-Wunused-parameter]
     void setWriteError(int err = 1) { }
                                  ^
/Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/Print.h: In member function 'size_t Print::write(uint8_t)':
/Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/Print.h:179:28: warning: no return statement in function returning non-void [-Wreturn-type]
     size_t write(uint8_t) {};
                            ^
In file included from /Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/HalfDuplexSerial.h:24:0,
                 from /Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/Arduino.h:119,
                 from /Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/HalfDuplexSerial.cpp:24:
/Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/Print.h: In member function 'void Print::setWriteError(int)':
/Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/Print.h:169:34: warning: unused parameter 'err' [-Wunused-parameter]
     void setWriteError(int err = 1) { }
                                  ^
/Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/Print.h: In member function 'size_t Print::write(uint8_t)':
/Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/Print.h:179:28: warning: no return statement in function returning non-void [-Wreturn-type]
     size_t write(uint8_t) {};
                            ^
In file included from /Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/HalfDuplexSerial.h:24:0,
                 from /Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/Arduino.h:119,
                 from /Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/Print.cpp:30:
/Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/Print.h: In member function 'void Print::setWriteError(int)':
/Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/Print.h:169:34: warning: unused parameter 'err' [-Wunused-parameter]
     void setWriteError(int err = 1) { }
                                  ^
/Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/Print.h: In member function 'size_t Print::write(uint8_t)':
/Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/Print.h:179:28: warning: no return statement in function returning non-void [-Wreturn-type]
     size_t write(uint8_t) {};
                            ^
In file included from /Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/HalfDuplexSerial.h:24:0,
                 from /Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/Arduino.h:119,
                 from /Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/main.cpp:12:
/Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/Print.h: In member function 'void Print::setWriteError(int)':
/Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/Print.h:169:34: warning: unused parameter 'err' [-Wunused-parameter]
     void setWriteError(int err = 1) { }
                                  ^
/Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/Print.h: In member function 'size_t Print::write(uint8_t)':
/Users/hans/Documents/Arduino/hardware/MicroCore/avr/cores/microcore/Print.h:179:28: warning: no return statement in function returning non-void [-Wreturn-type]
     size_t write(uint8_t) {};
                            ^
```

MCUdude, 2019-12-25:
It could easily be "fixed" again by changing the write instance back to  
`virtual size_t write(uint8_t) = 0;`.  
However, the compiled sized grew by ~120 bytes.

nerdralph, 2019-12-25:
I didn't notice them since I didn't have warnings on.
try:
`    size_t write(uint8_t) {return 0;}`

MCUdude, 2019-12-25:
Great: But this is the output (compiled size 620 bytes):

```
!
"
#
$
%
&
'
(
)
*
+
,
-
.
/
0
1
2
3
4
5
6
7
8
9
:
;
<
=
>
?
@
A
B
C
D
E
F
G
H
I
J
K
L
M
N
O
P
Q
R
S
T
U
V
W
X
Y
Z
[
\
]
^
_
`
a
b
c
d
e
f
g
h
i
j
k
l
m
n
o
p
q
r
s
t
u
v
w
x
y
z
{
|
}
~
```

However, this is what it's supposed to look like (compiled size 754 bytes):

```
Char: !, dec: 33, hex: 21, bin: 100001
Char: ", dec: 34, hex: 22, bin: 100010
Char: #, dec: 35, hex: 23, bin: 100011
Char: $, dec: 36, hex: 24, bin: 100100
Char: %, dec: 37, hex: 25, bin: 100101
Char: &, dec: 38, hex: 26, bin: 100110
Char: ', dec: 39, hex: 27, bin: 100111
Char: (, dec: 40, hex: 28, bin: 101000
Char: ), dec: 41, hex: 29, bin: 101001
Char: *, dec: 42, hex: 2A, bin: 101010
Char: +, dec: 43, hex: 2B, bin: 101011
Char: ,, dec: 44, hex: 2C, bin: 101100
Char: -, dec: 45, hex: 2D, bin: 101101
Char: ., dec: 46, hex: 2E, bin: 101110
Char: /, dec: 47, hex: 2F, bin: 101111
Char: 0, dec: 48, hex: 30, bin: 110000
Char: 1, dec: 49, hex: 31, bin: 110001
Char: 2, dec: 50, hex: 32, bin: 110010
Char: 3, dec: 51, hex: 33, bin: 110011
Char: 4, dec: 52, hex: 34, bin: 110100
Char: 5, dec: 53, hex: 35, bin: 110101
Char: 6, dec: 54, hex: 36, bin: 110110
Char: 7, dec: 55, hex: 37, bin: 110111
Char: 8, dec: 56, hex: 38, bin: 111000
Char: 9, dec: 57, hex: 39, bin: 111001
Char: :, dec: 58, hex: 3A, bin: 111010
Char: ;, dec: 59, hex: 3B, bin: 111011
Char: <, dec: 60, hex: 3C, bin: 111100
Char: =, dec: 61, hex: 3D, bin: 111101
Char: >, dec: 62, hex: 3E, bin: 111110
Char: ?, dec: 63, hex: 3F, bin: 111111
Char: @, dec: 64, hex: 40, bin: 1000000
Char: A, dec: 65, hex: 41, bin: 1000001
Char: B, dec: 66, hex: 42, bin: 1000010
Char: C, dec: 67, hex: 43, bin: 1000011
Char: D, dec: 68, hex: 44, bin: 1000100
Char: E, dec: 69, hex: 45, bin: 1000101
Char: F, dec: 70, hex: 46, bin: 1000110
Char: G, dec: 71, hex: 47, bin: 1000111
Char: H, dec: 72, hex: 48, bin: 1001000
Char: I, dec: 73, hex: 49, bin: 1001001
Char: J, dec: 74, hex: 4A, bin: 1001010
Char: K, dec: 75, hex: 4B, bin: 1001011
Char: L, dec: 76, hex: 4C, bin: 1001100
Char: M, dec: 77, hex: 4D, bin: 1001101
Char: N, dec: 78, hex: 4E, bin: 1001110
Char: O, dec: 79, hex: 4F, bin: 1001111
Char: P, dec: 80, hex: 50, bin: 1010000
Char: Q, dec: 81, hex: 51, bin: 1010001
Char: R, dec: 82, hex: 52, bin: 1010010
Char: S, dec: 83, hex: 53, bin: 1010011
Char: T, dec: 84, hex: 54, bin: 1010100
Char: U, dec: 85, hex: 55, bin: 1010101
Char: V, dec: 86, hex: 56, bin: 1010110
Char: W, dec: 87, hex: 57, bin: 1010111
Char: X, dec: 88, hex: 58, bin: 1011000
Char: Y, dec: 89, hex: 59, bin: 1011001
Char: Z, dec: 90, hex: 5A, bin: 1011010
Char: [, dec: 91, hex: 5B, bin: 1011011
Char: \, dec: 92, hex: 5C, bin: 1011100
Char: ], dec: 93, hex: 5D, bin: 1011101
Char: ^, dec: 94, hex: 5E, bin: 1011110
Char: _, dec: 95, hex: 5F, bin: 1011111
Char: `, dec: 96, hex: 60, bin: 1100000
Char: a, dec: 97, hex: 61, bin: 1100001
Char: b, dec: 98, hex: 62, bin: 1100010
Char: c, dec: 99, hex: 63, bin: 1100011
Char: d, dec: 100, hex: 64, bin: 1100100
Char: e, dec: 101, hex: 65, bin: 1100101
Char: f, dec: 102, hex: 66, bin: 1100110
Char: g, dec: 103, hex: 67, bin: 1100111
Char: h, dec: 104, hex: 68, bin: 1101000
Char: i, dec: 105, hex: 69, bin: 1101001
Char: j, dec: 106, hex: 6A, bin: 1101010
Char: k, dec: 107, hex: 6B, bin: 1101011
Char: l, dec: 108, hex: 6C, bin: 1101100
Char: m, dec: 109, hex: 6D, bin: 1101101
Char: n, dec: 110, hex: 6E, bin: 1101110
Char: o, dec: 111, hex: 6F, bin: 1101111
Char: p, dec: 112, hex: 70, bin: 1110000
Char: q, dec: 113, hex: 71, bin: 1110001
Char: r, dec: 114, hex: 72, bin: 1110010
Char: s, dec: 115, hex: 73, bin: 1110011
Char: t, dec: 116, hex: 74, bin: 1110100
Char: u, dec: 117, hex: 75, bin: 1110101
Char: v, dec: 118, hex: 76, bin: 1110110
Char: w, dec: 119, hex: 77, bin: 1110111
Char: x, dec: 120, hex: 78, bin: 1111000
Char: y, dec: 121, hex: 79, bin: 1111001
Char: z, dec: 122, hex: 7A, bin: 1111010
Char: {, dec: 123, hex: 7B, bin: 1111011
Char: |, dec: 124, hex: 7C, bin: 1111100
Char: }, dec: 125, hex: 7D, bin: 1111101
Char: ~, dec: 126, hex: 7E, bin: 1111110
```

nerdralph, 2019-12-25:
I'm seeing similar issues.  It doesn't seem to impact printNumber, but printing strings seems to get optimized away; the compiler inlines the Print::write(uint8_t) method instead of using the Serial class implementation.
I'll have to do more debugging to figure it out.

MCUdude, 2019-12-26:
I thought `virtual` was mandatory if a class that inherits Print modifies a function, in this case, write(uint8_t)? I've tried to work around it, but I don't see another solution than to copy all the contents of print.h and print.cpp into HalfDuplexSerial.h/cpp. Do you know of a better solution?

Even though we learned some C# OOP some years ago, I'm definitely not a pro at this.

MCUdude, 2019-12-26:
Just for fun, I tried to copy the contents of Print.h/cpp into HalfDupexSerial.h/cpp to see how much can be freed up by not relying on virtual functions. Instead of the previous compiled size of 754 bytes, the ASCIITable example compiles to 660 bytes instead. That's a great improvement!

Even though this generates a smaller compiled program, the implementation isn't very pretty. A workaround that would still let us use print.h/cpp would be preferable. However, compiled size is more important.

nerdralph, 2019-12-26:
gcc can devirtualize with LTO, I just haven't figured out why it doesn't in
this case.  Simply making HalfDuplexSerial as final isn't enough.  I'm
still looking at gcc and ld options for something thay will get it to
deirtualize at the Print methods without hacking up the code.

MCUdude, 2019-12-27:
I tried to add `final`, but it didn't do much. I also tried to search Google for similar cases with avr-gcc, but no luck. Have you had any progress?

nerdralph, 2019-12-27:
> 
> 
> I tried to add `final`, but it didn't do much. I also tried to search Google for similar cases with avr-gcc, but no luck. Have you had any progress?

I didn't have time for hacking today.  Probably Sunday.

nerdralph, 2019-12-29:
I did a bit of testing, and confirmed avr-gcc 5.4.0 can devirtualize methods when lto is enabled, even without the final specifier.
```
// devirtualization test
// avr-gcc -std=c++11 -mmcu=attiny13 -Os -flto devirt.cpp -o devirt
// 86 bytes w/o lto, obj contains derivedfoo impl
// 48 bytes w/lto, Derived final makes no difference

class Base
{
public:
    virtual int foo(int count) = 0;
    void bar(int count) {foo(count);}
};

class Derived : public Base
{
public:
    int foo(int count) {return count/2;}
};

Derived myD;

int main(int argc, char* argv[])
{
    return myD.foo(argc);
}
```

I think the way the Arduino core is being built and linked is making it harder for lto to devirtualize.  In the past I've seen lto fail to fully optimize depending on how the compile and link is done.
http://nerdralph.blogspot.com/2014/07/gcc-lto-call-graph-generation.html

I don't think MicroCore can't control the build process, so I've tried a few different compiler flags.  "-fdevirtualize-speculatively" didn't make any difference.  "-fwhole-program" on the link line may be worth a try.

I've also thought about changing Print.h to inline all the methods and remove Print.cpp.  I'll probably do some experiments with manually linking the core .o files with the sketch .o first, to see if lto still has trouble optimizing archived libraries the way it did back in 2014 when I made the blog post above.

On a separate issue related to the size of Serial, I've noticed Serial.print does a lot of saving and restoring registers before/after calling TxByte, even though it only uses 3 registers.  I can eliminate most of that by using an inline asm function that calls TxByte, specifying to the compiler the specific registers used.  It's a technique I invented a few years ago, and was something I used with the latest millis optimizations.
http://nerdralph.blogspot.com/2015/12/lightweight-avr-assembler-functions.html

MCUdude, 2019-12-29:
> I don't think MicroCore can't control the build process, so I've tried a few different compiler flags. "-fdevirtualize-speculatively" didn't make any difference. "-fwhole-program" on the link line [...]

If it turns out to be difficult it's absolutely possible to just copy most of the stuff from print.h/cpp into HalfDuplexSerial.h/cpp. It's not ideal from a maintenance point of view, but the compiler doesn't care really.

> On a separate issue related to the size of Serial, I've noticed Serial.print does a lot of saving and restoring registers before/after calling TxByte, even though it only uses 3 registers. I can [...]

You mean [this print definition](https://github.com/MCUdude/MicroCore/blob/c7b986ec97a2f4e851ba2f1933c218e30517020a/avr/cores/microcore/Print.cpp#L109-L112)? Print() calls write(), which again calls TxByte().

nerdralph, 2019-12-30:
I did a bunch of testing today, and still haven't been able to figure out why write isn't being devirtualized.  I created a test base class hierarchy with a virtual and non-virtual method, and a derived class that implements the virtual method.  Main calls the non-virtual method on a global instance of dervied, which calls the virtual method.
I compile with "avr-gcc -std=gnu++11 -Os -flto -ffunction-sections *.cpp -c", create an archive with the base & derived class implementation, and link with "avr-gcc -std=gnu++11 -Os -flto -Wl,--gc-sections".  I didn't learn much, other than finding that "-std=c++11 or -std=gnu++11" seems to be required in order to get gcc to fully devirtualize, but the Arduino build already uses "-std=c++11".

I could keep adding to the test base & derived classes to more closely mirror the Print hierarchy, but at least for now I'm going to put it on the back burner so I can work on other things.

MCUdude, 2019-12-30:
Thanks for all the research you've done. I've concluded that it's just not worth spending more time on this. I'll copy some of the contents from print.h/cpp into HalfDuplexSerial.h/cpp and call it a day. I'll close this issue when I'm finished. Again, thank you!

MCUdude, 2019-12-30:
Fixed in 8b96cfa

closed 2019-12-30 by MCUdude